espressif/esp-va-sdk #96: Amazon_aia not responding

GregTheTexan, 2020-06-08:
Hello

I followed carefully all your instructions for windows
I updated cmake to 3.17.3 version and ninja to 1.10.0
I modified the project.cmake file from the \esp-idf\tools\cmake folder
The project example amazon_aia built successfully without error
I successfully programmed the amazon_aia on my board (I can connect it to WiFi and link it to my Amazon account, but without certificate of course Alexa doesn't answer)
I registered for temporary certificates created in Espressif's AWS account, copied the files and modified the mfg_config.csv with the file location, generated the mfg.bin and successfully programmed them on my board
As instructed, I reprogrammed the amazon_aia application on my board
But it still doesn't respond to the Alexa wake-up word...
The five blue LEDs are ON and a green LED is shifting from left to right
Below is the log from the boards from the startup

Any idea what I could have missed? 
It seems like application programming and registration went well, but I can't wake up Alexa :(

C:\ESPRESSIF\esp-idf\repos\esp-va-sdk\examples\amazon_aia>idf.py monitor
Checking Python dependencies...
Python requirements from C:\ESPRESSIF\esp-idf\repos\esp-idf\requirements.txt are satisfied.
Executing action: monitor
Choosing default port b'COM3' (use '-p PORT' option to set a specific serial port)
Running idf_monitor in directory c:\espressif\esp-idf\repos\esp-va-sdk\examples\amazon_aia
Executing "C:\ESPRESSIF\esp-idf\tools\.espressif\python_env\idf4.0_py3.7_env\Scripts\python.exe C:\ESPRESSIF\esp-idf\repos\esp-idf\tools/idf_monitor.py -p COM3 -b 115200 c:\espressif\esp-idf\repos\esp-va-sdk\examples\amazon_aia\build\aia.elf -m 'C:\ESPRESSIF\esp-idf\tools\.espressif\python_env\idf4.0_py3.7_env\Scripts\python.exe' 'C:\ESPRESSIF\esp-idf\repos\esp-idf\tools\idf.py'"...
--- idf_monitor on COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6704
ho 0 tail 12 room 4
load:0x40078000,len:13892
load:0x40080400,len:4072
entry 0x40080684
I (68) boot: Chip Revision: 3
I (71) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot: ESP-IDF -128-NOTFOUND 2nd stage bootloader
I (42) boot: compile time 22:54:20
I (42) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 80MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 8MB
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (92) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (100) boot:  4 ota_0            OTA app          00 10 00020000 003f0000
I (107) boot:  5 ota_1            OTA app          00 11 00410000 003f0000
I (115) boot: End of partition table
I (119) boot_comm: chip revision: 3, min. application chip revision: 0
I (126) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1bd71c (1824540) map
I (773) esp_image: segment 1: paddr=0x001dd744 vaddr=0x3ffbdb60 size=0x028cc ( 10444) load
I (777) esp_image: segment 2: paddr=0x001e0018 vaddr=0x400d0018 size=0x126e64 (1207908) map
0x400d0018: _stext at ??:?

I (1202) esp_image: segment 3: paddr=0x00306e84 vaddr=0x3ffc042c size=0x01d18 (  7448) load
I (1205) esp_image: segment 4: paddr=0x00308ba4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/ESPRESSIF/esp-idf/repos/esp-idf/components/freertos/xtensa_vectors.S:1778

I (1209) esp_image: segment 5: paddr=0x00308fac vaddr=0x40080400 size=0x1bfc4 (114628) load
I (1282) boot: Loaded app from partition at offset 0x20000
I (1282) boot: Disabling RNG early entropy source...
W (1788) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (1870) wifi:wifi driver task: 3ffd66fc, prio:23, stack:6656, core=0
I (1880) wifi:wifi firmware version: aa5336b
[... 228 lines cut ...]
[http_transport]: Got response on topic fromservice: {"header":{"name":"Disconnect","messageId":"b8a18db0-6124-42a7-9360-89e500afe357"},"payload":{"code":"GOING_OFFLINE","description":"Disconnecting due to a fatal exception."}}
[http_transport]: AIS Disconnected.
[dialog]: Listen end
[dialog]: Event LISTEN_END unsupported in IDLE state
[dialog]: Speech done
[dialog]: Event SPEECH_DONE unsupported in IDLE state
[dialog]: Stream finished
[dialog]: Event STREAM_CLOSED unsupported in IDLE state
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/943bcd8d-cf42-41d5-aac5-201945746b02/connection/fromclient, {"header":{"name":"Connect","messageId":"a779b2db-17c0-ddf8-ad19-2b76055a6adf"},"payload":{"awsAccountId":"282183071650","clientId":"943bcd8d-cf42-41d5-aac5-201945746b02"}}
^C
C:\ESPRESSIF\esp-idf\repos\esp-va-sdk\examples\amazon_aia>
Traceback (most recent call last):
  File "C:\ESPRESSIF\esp-idf\repos\esp-idf\tools/idf_monitor.py", line 865, in <module>
    main()
  File "C:\ESPRESSIF\esp-idf\repos\esp-idf\tools/idf_monitor.py", line 777, in main
    monitor.main_loop()
  File "C:\ESPRESSIF\esp-idf\repos\esp-idf\tools/idf_monitor.py", line 375, in main_loop
    (event_tag, data) = self.event_queue.get()
  File "C:\Users\GregC\AppData\Local\Programs\Python\Python37\lib\queue.py", line 170, in get
    self.not_empty.wait()
  File "C:\Users\GregC\AppData\Local\Programs\Python\Python37\lib\threading.py", line 296, in wait
    waiter.acquire()
KeyboardInterrupt

Aborted!

chiragatal, 2020-06-08:
@GregTheTexan , This appears to be a server side issue.

GregTheTexan, 2020-06-08:
> @GregTheTexan , This appears to be a server side issue.

How do we fix this server issue?
I really need to get it up and running for my company to evaluate this voice solution

Alexlizx, 2020-06-08:
I am having the same issue as well. Compilation and provisioning went smoothly, but I've been stuck here since. @GregTheTexan I share your worry, I planned to demo this to my company as well today

albanoandrea, 2020-06-08:
I have this problem from today: on friday was working correctly. 
Today it works only the first time I do the provisioning, then if I reboot, I need to erase the flash and do everything from 0.
I suspect an Amazon change

Alexlizx, 2020-06-08:
> > @GregTheTexan , This appears to be a server side issue.
> 
> How do we fix this server issue?
> I really need to get it up and running for my company to evaluate this voice solution

Hi @GregTheTexan, I just want to let you know that I just solved this issue by erasing the flash and then redoing everything from step 1.

albanoandrea, 2020-06-08:
> Hi @GregTheTexan, I just want to let you know that I just solved this issue by erasing the flash and then redoing everything from step 1.

Ok but if you reboot you will have the issue again.

chiragatal, 2020-06-08:
> I have this problem from today: on friday was working correctly.
> Today it works only the first time I do the provisioning, then if I reboot, I need to erase the flash and do everything from 0.
> I suspect an Amazon change

@albanoandrea Thanks for the info. This is exactly what we have noticed too. Whenever the "capabilities" of the device are reported (this happens only after a factory reset), the device connects to the server.

Alexlizx, 2020-06-08:
> Ok but if you reboot you will have the issue again.

You are right, just resetting the device will bring back the issue. This is not going to boat well for a demo.

chiragatal, 2020-06-08:
As a temporary work-around, could you try adding this before `ais_mqtt_init()` in app_main.c:
```
unsigned char temp[32];
size_t temp_len = sizeof(temp);
va_nvs_set_blob("cap_version_2", temp, temp_len);
```
You will also need to add `#include <va_nvs_utils.h>`.

Alexlizx, 2020-06-08:
> As a temporary work-around, could you try adding this before `ais_mqtt_init()` in app_main.c:
> 
> ```
> unsigned char temp[32];
> size_t temp_len = sizeof(temp);
> va_nvs_set_blob("cap_version_2", temp, temp_len);
> ```
> 
> You will also need to add `#include <va_nvs_utils.h>`.

Thanks! This worked for me. Can you please explain how va_nvs_set_blob() was able to fix the issue?

GregTheTexan, 2020-06-08:
> As a temporary work-around, could you try adding this before `ais_mqtt_init()` in app_main.c:
> 
> ```
> unsigned char temp[32];
> size_t temp_len = sizeof(temp);
> va_nvs_set_blob("cap_version_2", temp, temp_len);
> ```
> 
> You will also need to add `#include <va_nvs_utils.h>`.

Thank you chiragatal for the instructions
Those solved my problem and I can now run Amazon AIA demo (even after a power-cycle)

closed 2020-06-08 by GregTheTexan