espressif/esp-va-sdk #103: Espressif's AWS Account doesn't work

klisevich, 2020-07-13:
Hi!
I had no luck with provisioning data by the Android app and tried temporary Espressif's AWS account. Got an email with creds, compiled them, flashed, "Waiting for authentication tokens" message kept printing. Please, help!

```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6800
ho 0 tail 12 room 4
load:0x40078000,len:13892
load:0x40080400,len:4072
entry 0x40080684
I (68) boot: Chip Revision: 3
I (71) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.0.1-197-g94d97499a-dirty 2nd stage bootloader
I (42) boot: compile time 22:26:39
I (43) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 80MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 8MB
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (94) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (101) boot:  4 ota_0            OTA app          00 10 00020000 003f0000
I (109) boot:  5 ota_1            OTA app          00 11 00410000 003f0000
I (116) boot: End of partition table
I (121) boot: No factory image, trying OTA 0
I (125) boot_comm: chip revision: 3, min. application chip revision: 0
I (133) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1be084 (1826948) map
I (780) esp_image: segment 1: paddr=0x001de0ac vaddr=0x3ffbdb60 size=0x01f64 (  8036) load
I (783) esp_image: segment 2: paddr=0x001e0018 vaddr=0x400d0018 size=0x126ba4 (1207204) map
0x400d0018: _stext at ??:?

I (1208) esp_image: segment 3: paddr=0x00306bc4 vaddr=0x3ffbfac4 size=0x026a8 (  9896) load
I (1212) esp_image: segment 4: paddr=0x00309274 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/klisevich/Documents/repo/esp-sdk-test/esp-idf/components/freertos/xtensa_vectors.S:1778

I (1216) esp_image: segment 5: paddr=0x0030967c vaddr=0x40080400 size=0x1c058 (114776) load
I (1289) boot: Loaded app from partition at offset 0x20000
I (1305) boot: Set actual ota_seq=1 in otadata[0]
I (1305) boot: Disabling RNG early entropy source...
I (1891) wifi:wifi driver task: 3ffd6724, prio:23, stack:6656, core=0
I (1901) wifi:wifi firmware version: d74ef33
I (1901) wifi:config NVS flash: enabled
I (1901) wifi:config nano formating: disabled
I (1901) wifi:Init dynamic tx buffer num: 32
I (1901) wifi:Init data frame dynamic rx buffer num: 32
I (1911) wifi:Init management frame dynamic rx buffer num: 32
I (1921) wifi:Init management short buffer num: 32
I (1921) wifi:Init static tx buffer num: 32
I (1921) wifi:Init static rx buffer size: 1600
I (1931) wifi:Init static rx buffer num: 10
I (1931) wifi:Init dynamic rx buffer num: 32
I (2091) wifi:mode : sta (c4:4f:33:40:ee:35)
I (2091) wifi:Set ps type: 0

I (2211) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2211) wifi:state: init -> auth (b0)
I (2211) wifi:state: auth -> assoc (0)
I (2221) wifi:state: assoc -> run (10)
I (2971) wifi:connected with bel, aid = 5, channel 11, BW20, bssid = 08:e8:4f:2a:29:a0
I (2971) wifi:security: WPA2-PSK, phy: bgn, rssi: -50
I (2971) wifi:pm start, type: 0

I (3001) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[app_main]: Connected with IP Address: 192.168.100.6
[alexa]: Waiting for time to be updated
[alexa]: Done getting current time: 1594671931
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
```

avsheth, 2020-07-14:
Hi @klisevich 
Looks like since device already had wifi credentials, it was looking for Alexa tokens next. There are 2 ways to approach it.
1. In the app, you can go to Manage devices, select your device, and perform Alexa sign 
2. You can reset the device (by erasing the flash and reflashing all the artifacts) and start over with provisioning.

klisevich, 2020-07-14:
Thank you @avsheth for the response.

> 1. In the app, you can go to Manage devices, select your device, and perform Alexa sign

I can't see my device in the Manage list. My phone and device are in the same WiFi network.

>2. You can reset the device (by erasing the flash and reflashing all the artifacts) and start over with provisioning.

I did this 3 times, still no luck, and as I said provisioning not working, the Android mobile app crashing whenever I push the Allow button on the Alexa site.

By the way, should I add a product to the Alexa account before provision starts?

khushbushah2302, 2020-07-14:
Hi @klisevich , 
Can you install Amazon Shopping app in Android mobile and try again for provisioning ?

klisevich, 2020-07-14:
> Hi @klisevich ,
> Can you install Amazon Shopping app in Android mobile and try again for provisioning ?

Thank you, @khushbushah2302 It helped! Is it obligatory to have the Amazon Shopping App installed? Would it be fixed in the future?

khushbushah2302, 2020-07-14:
Hi @klisevich ,
If Amazon shopping app is not installed, app will open Amazon login option in browser. It is working fine but there may be issue with some browser. That is why I asked to try with Amazon app.
Let me know which browser you are using in your phone. I will check the issue.

klisevich, 2020-07-14:
> Hi @klisevich ,
> If Amazon shopping app is not installed, app will open Amazon login option in browser. It is working fine but there may be issue with some browser. That is why I asked to try with Amazon app.
> Let me know which browser you are using in your phone. I will check the issue.

Sure! Samsung Internet version 12.0.1.47

closed 2020-07-15 by klisevich