espressif/esp-va-sdk #104: [http_transport]: Error subscribing : -28

klisevich, 2020-07-14:
Hi! I'm trying to make Alexa's voice assistant work. After ESP-Vaquita booted and connected, I uttered "Alexa" and after some state changes `error [http_transport]: Error subscribing : -28`  floated up and the board crashed.
The same error arose if the board left untouched in silence for about a minute.
Could you help me find a solution?
```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6800
ho 0 tail 12 room 4
load:0x40078000,len:13892
load:0x40080400,len:4072
entry 0x40080684
I (68) boot: Chip Revision: 3
I (71) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.0.1-197-g94d97499a-dirty 2nd stage bootloader
I (42) boot: compile time 15:38:04
I (43) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 80MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 8MB
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (94) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (101) boot:  4 ota_0            OTA app          00 10 00020000 003f0000
I (109) boot:  5 ota_1            OTA app          00 11 00410000 003f0000
I (116) boot: End of partition table
I (121) boot_comm: chip revision: 3, min. application chip revision: 0
I (128) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1bdf7c (1826684) map
I (775) esp_image: segment 1: paddr=0x001ddfa4 vaddr=0x3ffbdb60 size=0x0206c (  8300) load
I (778) esp_image: segment 2: paddr=0x001e0018 vaddr=0x400d0018 size=0x126ad0 (1206992) map
0x400d0018: _stext at ??:?

I (1203) esp_image: segment 3: paddr=0x00306af0 vaddr=0x3ffbfbcc size=0x025a0 (  9632) load
I (1207) esp_image: segment 4: paddr=0x00309098 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/klisevich/Documents/repo/esp-sdk-test/esp-idf/components/freertos/xtensa_vectors.S:1778

I (1211) esp_image: segment 5: paddr=0x003094a0 vaddr=0x40080400 size=0x1c058 (114776) load
I (1283) boot: Loaded app from partition at offset 0x20000
I (1284) boot: Disabling RNG early entropy source...
I (1870) wifi:wifi driver task: 3ffd67a8, prio:23, stack:6656, core=0
I (1880) wifi:wifi firmware version: d74ef33
I (1880) wifi:config NVS flash: enabled
I (1880) wifi:config nano formating: disabled
I (1880) wifi:Init dynamic tx buffer num: 32
I (1880) wifi:Init data frame dynamic rx buffer num: 32
I (1890) wifi:Init management frame dynamic rx buffer num: 32
I (1900) wifi:Init management short buffer num: 32
I (1900) wifi:Init static tx buffer num: 32
I (1900) wifi:Init static rx buffer size: 1600
I (1910) wifi:Init static rx buffer num: 10
I (1910) wifi:Init dynamic rx buffer num: 32
I (2070) wifi:mode : sta (c4:4f:33:40:ee:35)
I (2070) wifi:Set ps type: 0

I (2190) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2190) wifi:state: init -> auth (b0)
I (2200) wifi:state: auth -> assoc (0)
I (2200) wifi:state: assoc -> run (10)
I (2530) wifi:connected with bel, aid = 4, channel 11, BW20, bssid = 08:e8:4f:2a:29:a0
I (2530) wifi:security: WPA2-PSK, phy: bgn, rssi: -47
I (2540) wifi:pm start, type: 0

I (2580) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[app_main]: Connected with IP Address: 192.168.100.6
[alexa]: Waiting for time to be updated
[alexa]: Done getting current time: 1594731455
[alexa]: Authentication done
[dialog]: Entering VA_IDLE
[speaker]: Volume changed to 40
[attachment]: New with content_id: ais_attachment
[http_transport]: AWS Account ID: 282183071650
[http_transport]: Client ID: 1053fa12-d0ca-49f2-a8bb-ffa400452b1c
[http_transport]: Current time in London: Tue Jul 14 12:57:35 2020
[auth-delegate]: Token will be refreshed after 3000 seconds.
[alexa_auth_delegate]: Registration Request Buffer: {"authentication":{"token":"Atzr|IwEBII7vW2JqZBg-HPQ3ya5A-X3MQe5jpRUcfrZbpMnyTZsq77OiGKCpy_hcgzWwkCY2Hw4yCt0yPSlJuqBlLDAMNQdxsCStm3Q9iY0MXRiKwHY7kD3NUqonHdAw62td02nql_4t9HWdS7XllnP8AZol9AUCsrCU5gtMkkqlT7mb3gytu8lLm0IPxJLSz56ZkegDneIoZ7_hIy-VzQWtgxz42fWEvt0FW1PF6UHMtxJtwBTMREtj-lQ7wSYAhjr2CKsSV6wkz3Mkzajt7L83bVv687Gn34fXzH0Q7sDLXID0wwUu_JKZySk54VLa-AvtFOXWSKMXWAyvgfEBlVmuboRS1RGM-ySrKe2nqKtgvWzTmQVQwqmtfBkEmgWwS0AuvUNoDmAEpIpfiCiXq7pwIk8jV67B", "clientId":"amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9"},"encryption":{"algorithm":"ECDH_CURVE_25519_32_BYTE", "publicKey":"DBUDS59OJcDYcvdUPLRORFDc9S828XgwBGdfA1BDCQ8="},"iot":{"awsAccountId":"282183071650", "clientId":"1053fa12-d0ca-49f2-a8bb-ffa400452b1c","endpoint":"aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com","topicRoot":"$aws/alexa"}}
[alexa_auth_delegate]: Registration Response Buffer: {"encryption": {"publicKey": "hFnohdiLSIx2dM6rwcN5hlh3L76RR+DxOVYVeymy9zs="}, "iot": {"topicRoot": "$aws/alexa"}}
Transport Shared Secret::
0x2b 0x4f 0x25 0x51 0x38 0x35 0xab 0xe5 0x6d 0x16 0x1c 0x15 0xb1 0x02 0x28 0x6b 0x15 
0xf0 0x8b 0x16 0xfd 0x00 0x27 0x88 0xe1 0xd5 0x6f 0xea 0x0d 0x0b 0x06 0x48 
n'.��kM�c�
                 MCm%�N��Y}�ƨ!D 7��g�V�L�݄Y�bQ��Ӎ|����{��8@o��'�4nab�l�+�?n�ihi*�fW13.20: Neural network detection triggered output 1.
[speech_recognizer]: New recognize request: WW length 0
[dialog]: Dialog new
[speech_recognizer]: On Focus: Foreground
[dialog]: Entering VA_LISTENING
E (25320) [http_transport]: Ignoring http_enqueue since device is not active.
[dialog]: Stream finished
[dialog]: Entering VA_IDLE
E (32190) [http_transport]: Error subscribing : -28 
abort() was called at PC 0x400dbe01 on core 0
0x400dbe01: http_transport_speech_stop at ??:?

ELF file SHA256: 09841dd0bd66480d

Backtrace: 0x40093158:0x3ffe81e0 0x40093545:0x3ffe8200 0x400dbe01:0x3ffe8220 0x400dc9f1:0x3ffe8260
0x40093158: invoke_abort at /home/klisevich/Documents/repo/esp-sdk-test/esp-idf/components/esp32/panic.c:157

0x40093545: abort at /home/klisevich/Documents/repo/esp-sdk-test/esp-idf/components/esp32/panic.c:172

0x400dbe01: http_transport_speech_stop at ??:?

0x400dc9f1: http_task_mqtt_connect at ??:?

```

chiragatal, 2020-07-14:
Hi @klisevich , This is because of a recent update in esp-aws-iot repository. We are working on it. Till then could you go to esp-aws-iot/aws-iot-device-sdk-embedded-C and change the branch to 'v3.0.1' and try again?

klisevich, 2020-07-15:
> Hi @klisevich , This is because of a recent update in esp-aws-iot repository. We are working on it. Till then could you go to esp-aws-iot/aws-iot-device-sdk-embedded-C and change the branch to 'v3.0.1' and try again?

Thank you, @chiragatal It helped!

closed 2020-07-15 by klisevich