espressif/esp-va-sdk #105: The board continuously disconnects.

klisevich, 2020-07-15:
Hi!
Finally I have made ESP-Vaquita work with AIA!. But it didn't last long. After a couple of hours the board started continuously disconnecting and connecting to the service. Please, help.

```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6800
ho 0 tail 12 room 4
load:0x40078000,len:13892
load:0x40080400,len:4072
entry 0x40080684
I (68) boot: Chip Revision: 3
I (71) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.0.1-197-g94d97499a-dirty 2nd stage bootloader
I (42) boot: compile time 18:42:42
I (43) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 80MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 8MB
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (94) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (101) boot:  4 ota_0            OTA app          00 10 00020000 003f0000
I (109) boot:  5 ota_1            OTA app          00 11 00410000 003f0000
I (116) boot: End of partition table
I (121) boot: No factory image, trying OTA 0
I (125) boot_comm: chip revision: 3, min. application chip revision: 0
I (133) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1bdf1c (1826588) map
I (780) esp_image: segment 1: paddr=0x001ddf44 vaddr=0x3ffbdb60 size=0x020cc (  8396) load
I (783) esp_image: segment 2: paddr=0x001e0018 vaddr=0x400d0018 size=0x126a20 (1206816) map
0x400d0018: _stext at ??:?

I (1208) esp_image: segment 3: paddr=0x00306a40 vaddr=0x3ffbfc2c size=0x02540 (  9536) load
I (1212) esp_image: segment 4: paddr=0x00308f88 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/klisevich/Documents/repo/esp-sdk-test/esp-idf/components/freertos/xtensa_vectors.S:1778

I (1215) esp_image: segment 5: paddr=0x00309390 vaddr=0x40080400 size=0x1c058 (114776) load
I (1288) boot: Loaded app from partition at offset 0x20000
I (1305) boot: Set actual ota_seq=1 in otadata[0]
I (1305) boot: Disabling RNG early entropy source...
I (1891) wifi:wifi driver task: 3ffd67a8, prio:23, stack:6656, core=0
I (1901) wifi:wifi firmware version: d74ef33
I (1901) wifi:config NVS flash: enabled
I (1901) wifi:config nano formating: disabled
I (1901) wifi:Init dynamic tx buffer num: 32
I (1901) wifi:Init data frame dynamic rx buffer num: 32
I (1911) wifi:Init management frame dynamic rx buffer num: 32
I (1921) wifi:Init management short buffer num: 32
I (1921) wifi:Init static tx buffer num: 32
I (1921) wifi:Init static rx buffer size: 1600
I (1931) wifi:Init static rx buffer num: 10
I (1931) wifi:Init dynamic rx buffer num: 32
I (2081) wifi:mode : sta (c4:4f:33:40:ee:35)
I (2091) wifi:Set ps type: 0

I (2211) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2211) wifi:state: init -> auth (b0)
I (2211) wifi:state: auth -> assoc (0)
I (2221) wifi:state: assoc -> run (10)
I (2481) wifi:connected with bel, aid = 1, channel 11, BW20, bssid = 08:e8:4f:2a:29:a0
I (2481) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
I (2481) wifi:pm start, type: 0

I (2481) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[... 131 lines cut ...]
[http_transport]: Got response on topic fromservice: {"header":{"name":"Acknowledge","messageId":"d4db2a87-c618-4fba-bfc6-564fcfa3d7e7"},"payload":{"connectMessageId":"69edd9bb-f82f-ceee-cd49-ea11c56b1e71","code":"CONNECTION_ESTABLISHED"}}
[26 seconds]: [http_transport]: Event data: {"events":[{"header":{"name":"SynchronizeState","messageId":"d99b604c-60bd-fde0-e5ee-1dc419644198"},"payload":{"speaker":{"volume":60},"alerts":{"allAlerts":[]}}}]}
[http_transport]: Got response on topic fromservice: {"header":{"name":"Disconnect","messageId":"ae8ab88c-5cea-4e7d-9207-4035e09272f7"},"payload":{"code":"GOING_OFFLINE","description":"Disconnecting due to a fatal exception."}}
[http_transport]: AIS Disconnected.
[dialog]: Listen end
[dialog]: Event LISTEN_END unsupported in IDLE state
[dialog]: Speech done
[dialog]: Event SPEECH_DONE unsupported in IDLE state
[dialog]: Stream finished
[dialog]: Event STREAM_CLOSED unsupported in IDLE state
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/1053fa12-d0ca-49f2-a8bb-ffa400452b1c/connection/fromclient, {"header":{"name":"Connect","messageId":"532f5610-ed9b-68d9-e1a0-9f6309ad9bf4"},"payload":{"awsAccountId":"282183071650","clientId":"1053fa12-d0ca-49f2-a8bb-ffa400452b1c"}}
[http_transport]: Got response on topic fromservice: {"header":{"name":"Acknowledge","messageId":"9ef221d8-241a-43db-aa57-f23b627f75f4"},"payload":{"connectMessageId":"532f5610-ed9b-68d9-e1a0-9f6309ad9bf4","code":"CONNECTION_ESTABLISHED"}}
[27 seconds]: [http_transport]: Event data: {"events":[{"header":{"name":"SynchronizeState","messageId":"2378435d-6688-dbac-c506-c63e01b89f74"},"payload":{"speaker":{"volume":60},"alerts":{"allAlerts":[]}}}]}
[28 seconds]: [http_transport]: Free Memory Internal: 77416, External: 2838396
E (29871) [http_transport]: Cannot get data from pool for topic: 5
[http_transport]: Got response on topic fromservice: {"header":{"name":"Disconnect","messageId":"0e1a424d-62bf-49a7-90e6-adafed00c6ff"},"payload":{"code":"GOING_OFFLINE","description":"Disconnecting due to a fatal exception."}}
[http_transport]: AIS Disconnected.
[dialog]: Listen end
[dialog]: Event LISTEN_END unsupported in IDLE state
[dialog]: Speech done
[dialog]: Event SPEECH_DONE unsupported in IDLE state
[dialog]: Stream finished
[dialog]: Event STREAM_CLOSED unsupported in IDLE state
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/1053fa12-d0ca-49f2-a8bb-ffa400452b1c/connection/fromclient, {"header":{"name":"Connect","messageId":"2f4a2a6e-1247-6b6c-9d48-1ff92b93ff73"},"payload":{"awsAccountId":"282183071650","clientId":"1053fa12-d0ca-49f2-a8bb-ffa400452b1c"}}
```

chiragatal, 2020-07-15:
Hi @klisevich , Please check this: https://github.com/espressif/esp-va-sdk/issues/96#issuecomment-640663213

klisevich, 2020-07-15:
> Hi @klisevich , Please check this: [#96 (comment)](https://github.com/espressif/esp-va-sdk/issues/96#issuecomment-640663213)

Thank you! It works!

closed 2020-07-15 by klisevich