espressif/esp-va-sdk #106: MQTT over WebSocket

klisevich, 2020-07-15:
Hi!
We want to use ESP-VA-SDK as a code reference for developing a demo for ESP-Vaquita board appliance. In our project we want to utilize [AWS Signature version 4](https://docs.aws.amazon.com/AmazonS3/latest/API/sig-v4-authenticating-requests.html) and [MQTT over WebSocket](https://docs.aws.amazon.com/iot/latest/developerguide/mqtt-ws.html) functionality for connecting to AWS IoT Core. Are there possible ways for us to substitute ais_mqtt_init() with custom ais_mqtt_over ws_init() function with preserving voice assistant functionality?

chiragatal, 2020-07-15:
Can you explain your use-case a bit more? Is it something that can't be supported with the existing implementation?

avsheth, 2020-07-15:
Hi @klisevich Any reason for using MQTT over Websocket? SDK could send MQTT packets over TCP sockets directly to IoT core.

klisevich, 2020-07-15:
> Can you explain your use-case a bit more? Is it something that can't be supported with the existing implementation?

> Hi @klisevich Any reason for using MQTT over Websocket? SDK could send MQTT packets over TCP sockets directly to IoT core.

Sure! The point of this demo is to replace MQTT protocol with MQTT over WebSocket protocol. As AIA works over the MQTT protocol, it requires managing security certificates when used over TLS. This can create considerable friction in the provisioning procedure. MQTT can use the WebSocket protocol as a transport and it allows for a significantly simpler authentication procedure. We going to utilize pre-signed URL scheme used by AWS for opening the connection over WebSocket. It would be cool if you add to the SDK an option to work with MQTT over WebSocket protocol. Unfortunately, we couldn't find any information regarding MQTT over TCP support from AWS.