espressif/esp-va-sdk #107: Registration with a custom AWS account.

klisevich, 2020-07-17:
Hi!
I evaluated voice assistant with Espressif AWS account and now want it run on my account. I created an AVS product and security profile but couldn't  get my device registered. I called  `menuconfig` and set product ID to 'esp_vaq_va' and when I was pushing "Allow" button in amazon shopping app the following error floated up:
[image]
After rolling back the Product ID to "esp_alexa_open" the console outputs 
```
[avs_config]: App Got: ANANryPHlMfYQNXbhIkA, amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9, amzn://com.espressif.provbleavs, abcd1234
```
and application wasn't crashing. What should I do to run with a custom product ID?

chiragatal, 2020-07-17:
Hi @klisevich , As mentioned in the Readme(https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta#32-using-your-aws-account), if you want to use your custom AVS Product ID, you will have to generate the phone apps with your AVS Product.
However you can use your own AWS Account without creating a new AVS Product and continue using the existing phone apps.

klisevich, 2020-07-17:
> Hi @klisevich , As mentioned in the Readme(https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta#32-using-your-aws-account), if you want to use your custom AVS Product ID, you will have to [...]

Thank you for responding! For me, it is important to work with a custom AVS Product. I'm not an experienced Android developer, so could you kindly provide example files with changes(what and where) that I should make in the Android project.