espressif/esp-va-sdk #109: amazon_aia project compilation failed on Ubuntu 16.04LTS

wws666, 2020-07-21:
When I fellow this guide(https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta) to compile amazon_aia project, I met follow issue. I also tried to build the same environment on my Windows. I did not find this issue on my Windows.

idf.py flash monitorcmake: /usr/local/lib/libcurl.so.4: no version information available (required by cmake)
Checking Python dependencies...
Python requirements from /home/wensheng/source_code/ESP32-WROVER-E/esp-idf/requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory /home/wensheng/source_code/ESP32-WROVER-E/esp-va-sdk/examples/amazon_aia/build
Executing "ninja all"...
[2/935] cd /home/wensheng/source_code/ESP32-WROVER-E/esp-va-...***********************************************************"
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
Partition table binary generated. Contents:
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
fctry_aia,data,nvs,0x10000,24K,
fctry,data,nvs,0x16000,24K,
otadata,data,ota,0x1c000,8K,
ota_0,app,ota_0,0x20000,4032K,
ota_1,app,ota_1,0x410000,4032K,
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
*******************************************************************************
[3/935] Building C object esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth.c.obj
FAILED: /home/wensheng/source_code/ESP32-WROVER-E/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -Iconfig -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/platform_include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/freertos/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/heap/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/log/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/soc/esp32/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/soc/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_rom/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_common/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/xtensa/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/xtensa/esp32/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp32/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/driver/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_ringbuf/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_event/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/tcpip_adapter/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/lwip/include/apps -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/lwip/include/apps/sntp -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/lwip/lwip/src/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/lwip/port/esp32/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/lwip/port/esp32/include/arch -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/vfs/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_wifi/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_wifi/esp32/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/efuse/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/efuse/esp32/include -I/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -DVOICE_ASSISTANT_AIA -Os -freorder-blocks -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1-316-g48ea44f3d-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -std=gnu11 -MMD -MT esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth.c.obj -MF esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth.c.obj.d -o esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth.c.obj   -c /home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c
In file included from /home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h: In function 'atomic_flag_test_and_set_explicit':
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:271:31: error: request for member '__val' in something not a structure or union
  __atomic_exchange_n(&(object)->__val, desired, order)
                               ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:387:10: note: in expansion of macro 'atomic_exchange_explicit'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
          ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:387:2: error: incompatible type for argument 1 of '__atomic_exchange_n'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
  ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h: In function 'atomic_flag_clear_explicit':
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:285:28: error: request for member '__val' in something not a structure or union
  __atomic_store_n(&(object)->__val, desired, order)
                            ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:394:2: note: in expansion of macro 'atomic_store_explicit'
  atomic_store_explicit(&__object->__flag, 0, __order);
  ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:394:2: error: incompatible type for argument 1 of '__atomic_store_n'
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_install':
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:91:47: error: request for member '__val' in something not a structure or union
 #define atomic_init(obj, value)  ((void)((obj)->__val = (value)))
                                               ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c:178:5: note: in expansion of macro 'atomic_init'
     atomic_init(&eth_driver->ref_count, 1);
     ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:91:47: error: request for member '__val' in something not a structure or union
 #define atomic_init(obj, value)  ((void)((obj)->__val = (value)))
                                               ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c:179:5: note: in expansion of macro 'atomic_init'
     atomic_init(&eth_driver->fsm, ESP_ETH_FSM_STOP);
     ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_uninstall':
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:264:39: error: request for member '__val' in something not a structure or union
  __atomic_compare_exchange_n(&(object)->__val, expected,  \
                                       ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/newlib/include/stdatomic.h:347:2: note: in expansion of macro 'atomic_compare_exchange_strong_explicit'
  atomic_compare_exchange_strong_explicit(object, expected, \
  ^
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c:221:10: note: in expansion of macro 'atomic_compare_exchange_strong'
[... cut ...]
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c:373:5: error: incompatible type for argument 1 of '__atomic_fetch_sub'
/home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp_eth/src/esp_eth.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
[3/935] Performing build step for 'bootloader'
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
ninja: no work to do.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

wws666, 2020-07-21:
My SOC module is : ESP32-WROVER-E
Kit name is: ESP32-Voquita-DSPG_v1.0

chiragatal, 2020-07-21:
Hi @wws666 , Can you tell your:
1. cmake version
2. esp-idf branch and commit ID
3. toolchain version: xtensa-esp32-elf-gcc -v

wws666, 2020-07-21:
cmake --version
cmake: /usr/local/lib/libcurl.so.4: no version information available (required by cmake)
cmake version 3.5.1

Branch info(release/V4.0)
commit 48ea44f3d1e4b5763a2d0b65bc5c56a84690e910 (HEAD -> test, origin/release/v4.0)
Merge: 009492ca9 5da359c25
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Wed Jul 15 20:08:55 2020 +0800

    Merge branch 'feature/light_sleep_reject_v4.0' into 'release/v4.0'
    
    sleep: enable sleep reject when entering light sleep (v4.0)
    
    See merge request espressif/esp-idf!9501

Toolchain version:
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/wensheng/source_code/ESP32-WROVER-E/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-80-g6c4433a' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-cloog=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio
Thread model: posix
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a)

chiragatal, 2020-07-21:
@wws666 , 
1. Can you update the cmake version to latest? I think it is 3.18.0
2. The branch looks alright
3. The toolchain version does not look updated. The version on release/v4.0 branch of IDF should be esp-2020r1. I think you will need to run the `./install.sh` command again.

wws666, 2020-07-21:
Hi @chiragatal 
How to check which version of IDF is used for branch release/v4.0? I cloned release/v4.0 branch into my local.  And esp-2019r2-8.2.0 is used there. 
./install.sh 
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Installing Python environment and packages
Installing Python packages from /home/wensheng/source_code/ESP32-WROVER-E/esp-idf/requirements.txt

I'm concerned that the incompatibility of GCC version  may be the issue's root cause.

wws666, 2020-07-22:
After re-iinstalling toolchain to esp-2019r2-8.2.0, this issue is fixed.
re-installing steps:
1. delete previous installation information
  ---> rm -rf  ~/.espressif
2. Install  esp-2019r2-8.2.0
--> ./install.sh

Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=~/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2019r2' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2019r2)

closed 2020-07-22 by wws666