espressif/esp-va-sdk #110: Waiting for voice_assistant initialisation infinitely

wws666, 2020-07-22:
I'm following this guide https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta to run amazon_aia project.
After I flashed all images and certificates. I met below issue.  

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6792
load:0x40078000,len:12612
load:0x40080400,len:3788
entry 0x40080684
I (61) boot: Chip Revision: 3
I (33) boot: ESP-IDF v4.0.1-316-g48ea44f3d-dirty 2nd stage bootloader
I (33) boot: compile time 13:46:41
I (34) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 80MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 8MB
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 fctry_aia        WiFi data        01 02 00010000 00006000
I (85) boot:  3 fctry            WiFi data        01 02 00016000 00006000
I (92) boot:  4 otadata          OTA data         01 00 0001c000 00002000
I (100) boot:  5 ota_0            OTA app          00 10 00020000 003f0000
I (107) boot:  6 ota_1            OTA app          00 11 00410000 003f0000
I (115) boot: End of partition table
I (119) boot: No factory image, trying OTA 0
I (124) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1c5ea0 (1859232) map
I (736) esp_image: segment 1: paddr=0x001e5ec8 vaddr=0x3ffbdb60 size=0x041f8 ( 16888) load
I (742) esp_image: segment 2: paddr=0x001ea0c8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/freertos/xtensa_vectors.S:1778

I (743) esp_image: segment 3: paddr=0x001ea4d0 vaddr=0x40080400 size=0x05b40 ( 23360) load
I (760) esp_image: segment 4: paddr=0x001f0018 vaddr=0x400d0018 size=0x117914 (1145108) map
0x400d0018: _stext at ??:?

I (1131) esp_image: segment 5: paddr=0x00307934 vaddr=0x40085f40 size=0x1164c ( 71244) load
0x40085f40: esp_flash_erase_region at /home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/spi_flash/esp_flash_api.c:374

I (1173) boot: Loaded app from partition at offset 0x20000
I (1189) boot: Set actual ota_seq=1 in otadata[0]
I (1190) boot: Disabling RNG early entropy source...
I (1190) psram: This chip is ESP32-D0WD
I (1194) spiram: Found 64MBit SPI RAM device
I (1199) spiram: SPI RAM mode: flash 80m sram 80m
I (1204) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1211) cpu_start: Pro cpu up.
I (1215) cpu_start: Application information:
I (1220) cpu_start: Project name:     aia
I (1225) cpu_start: App version:      1.0
I (1229) cpu_start: Compile time:     Jul 22 2020 13:46:44
I (1236) cpu_start: ELF file SHA256:  20fe5111a267f78a...
I (1242) cpu_start: ESP-IDF:          v4.0.1-316-g48ea44f3d-dirty
I (1248) cpu_start: Starting app cpu, entry point is 0x40081630
0x40081630: call_start_cpu1 at /home/wensheng/source_code/ESP32-WROVER-E/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (1748) spiram: SPI SRAM memory test OK
I (1752) heap_init: Initializing. RAM available for dynamic allocation:
I (1752) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1755) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1761) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1767) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1773) heap_init: At 3FFC7D38 len 000182C8 (96 KiB): DRAM
I (1780) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1786) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1793) heap_init: At 4009758C len 00008A74 (34 KiB): IRAM
[... 132 lines cut ...]
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assi

chiragatal, 2020-07-22:
@wws666 , When ais_mqtt_init() is called, the device waits indefinitely and tries to get the current_time from SNTP server. In your case, it looks like the device was not able to get the updated time.
- Is this happening every time?
- In the device's CLI, could you type `sock-dump` and share the output?
- Could you erase_flash and try again with a different network?

wws666, 2020-07-22:
Yes. When I use esp32 App to register device, it will come here, waiting for voice_assistant initialization forever.

sock-dump:
sock-dump
sock_fd	protocol	local_addr		peer_addr
55	1:tcp	         0.0.0.0:80
56	2:udp	       127.0.0.1:32768
57	2:udp	         0.0.0.0:0
58	2:udp	         0.0.0.0:1900
59	1:tcp	         0.0.0.0:80	         0.0.0.0:40218
Remaining sockets: 4

Will find another AP for testing, updating test result soon...

wws666, 2020-07-22:
After changing a new WIFI AP, this issue disappears.

closed 2020-07-22 by wws666