espressif/esp-va-sdk #114: Use my AWS account

pengrenlai, 2020-09-23:
Hi!
I follow [https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta#32-using-your-aws-account] to use my own AWS account. For a quick evaluation, I skip the AVS product creation step. But when I connect to AWS account, I get this error.
[image]
How do I resolve the registration problem? Thanks.

chiragatal, 2020-09-23:
The AWS account ID needs to be linked with the AVS product for registration to work. I can temporarily link your AWS account ID with Espressif's product ID. Could you share it over email (sales@espressif.com)?

pengrenlai, 2020-09-23:
I've already sent my AWS account ID to [sales@espressif.com](sales@espressif.com). Could you let me know if you link my AWS account ID with Espressif's product ID? Thanks a lot!