espressif/esp-va-sdk #119: help with esp-va-sdk and esp32-a1s

jazzvivi2, 2020-11-03:
I am trying to run the alexa example with my Ai Thinker A1S board. I managed to run the example play mp3. but in the example of alexa I think that more changes have to be made to be compatible with my board. any help with this?
Thank you!

vikramdattu, 2020-11-04:
@jazzvivi2 which example have you tried exactly? And with which branch?
Did you follow steps here to run the amazon_alexa example?: https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md

jazzvivi2, 2020-11-04:
hello, thanks for replying. use the main branch and follow the steps. use this https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/tree/master/components to get the sound driver but I'm not sure how to modify or create the audio_board.cmake. audio_board.mk

chiragatal, 2020-11-06:
@jazzvivi2 , You will have to add this audio_board in the board_support_pkgs like this: https://github.com/espressif/esp-va-sdk/tree/master/board_support_pkgs/lyrat

I think you can start by copy-pasting the lyrat board, then edit it according to your board.
For example, the codec used on your board is AC101. You will need to replace the existing es8388 with ac101 and then modify the glue layer (media_hal_codec_init.c) accordingly.

The other pin_configs will also be need to changed in your audio_board.

Then there would be corresponding changes in the audio_board.mk to point to the new components.