espressif/esp-va-sdk #121: Error(-23) connecting to xxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com

lorenzomerici, 2020-12-01:
I'm trying to configure an ESP32-lyrat, firmware of aia-beta branch, with our AWS cloud account. 

Provisioning, with our modified app, goes well, but after we get:

`E (109853) aws_iot: failed! mbedtls_net_connect returned -0x52
E (109853) [http_transport]: Error(-23) connecting to xxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com
`
My idea is that is something related with device.info file. In your email, sent by your tool, it contains an UUID, but we cannot understand from the readme where you took it in the AWS dashboard. Where can I get it after creating a new thing on the AWS dashboard? 

And If this is not the case, and that UUID may be a random string as I thought, what can cause this issue? 
Tell me if you need further log. 

Thanks

chiragatal, 2020-12-01:
The device.info is just a random string. But the 'thing' name in AWS console is same as the device.info. Can you make both of them same and try?

lorenzomerici, 2020-12-01:
I tried but I've got the same result. Do you have any suggestions?

With the same endpoint and with those device certs, I can connect and publish with MQTT mosquitto client.

chiragatal, 2020-12-02:
Have you assigned the policy correctly? Have you 'activated' the device?

lorenzomerici, 2020-12-02:
Yes policy attached and device activated. I followed the entire ReadMe two times but same result. Do you think that something related to the APP may cause this issue? 

I've enabled silly log, but nothing intresting appeared. 

> D (110899) aws_iot: Seeding the random number generator...                     
D (110909) aws_iot: Loading embedded CA root certificate ...                   
D (110909) aws_iot: ok (0 skipped)     
D (110919) aws_iot: Loading embedded client certificate...                     
D (110919) aws_iot: Loading embedded client private key...                     
D (110929) aws_iot: ok                 
D (110929) aws_iot: Connecting to a1bun9vsnz7i7j-ats.iot.eu-west-1.amazonaws.com                                                                              
/8883...                               
E (111199) aws_iot: failed! mbedtls_net_connect returned -0x52

chiragatal, 2020-12-02:
I don't think this is problem on the App side.

I think the problem might be, that there are 'new_lines' (blank lines) at the end of your device.info, account.info and endpoint.info files. If this is the case, can you remove the new_lines and try?

If even this does not work, can you share the complete device log?

lorenzomerici, 2020-12-02:
You are right. I didn't know that my editor was appending new line on save. 

Now it works, I get `E (32553) [directive_proc]: No handler found for :Exception, ignoring... ` but that's another error, I will investigate. Thanks for your help

closed 2020-12-02 by lorenzomerici

chiragatal, 2020-12-02:
Great. We will add something related to new_line in the documentation too.

I think I might already know what the other error would be. By default, smart-home is enabled in the application (app_smart_home_init()), but your account might not be white-listed for smart-home yet. That is why you are getting the error. You can comment that line and you won't get the error.

However, you might get another error after that related to 'yield'. We had pushed a new commit yesterday for fixing that. If you use the latest branch, your device should work.

lorenzomerici, 2020-12-02:
Yes I know about the error 'yield', but I've seen the issue on gitlab and so I knew you were working on it. I'll try to remove the line app_smart_home_init() and give a feedback to you. 

Yes I think that a few lines of comment about the 5 files to flash on the device would be very useful :+1:

lorenzomerici, 2020-12-02:
Yes you are right. Commenting that line it works.