espressif/esp-va-sdk #122: Include components in idf project

nixmeer, 2020-12-04:
hello :)

I am working on a project using the idf and flash it using the `idf.py`. I would like to include some components but not all of the sdk.

avsheth, 2020-12-05:
Hi @nixmeer 
Sorry I couldn't exactly get the query. You want to only compile and use specific components of IDF or esp-va-sdk ?

nixmeer, 2020-12-05:
I have an IDF project that I compile using `idf.py`. I would like to use some of the components of the va-sdk in this project. But since this one uses make and the idf cmake, I could not yet find a way to include the components.

NikhilPatale, 2021-01-12:
Hello!
I am having somewhat the same issue, I am trying to execute this same project in windows where you require CMakeLists.txt in order to flash the code through idf.py in MinGW or ESP-IDF Command Prompt. I am unable to flash the code because there is no CMakeList.txt in the project directory esp-va-sdk. So is there any way by which you can provide a CMakeList.txt file for this so that the project can be flashed through a windows machine?

avsheth, 2021-01-14:
Hi @NikhilPatale @nixmeer We are working to release newer AVS/AIA versions with latest IDF and with CMake support. Release is planned for early next month. I'll notify once the release is available.
Thanks.

nixmeer, 2021-01-14:
Hello @avsheth
My college project might be over by then, but I think it’s a step in the right direction :) 
Thank you

closed 2021-01-14 by nixmeer

avsheth, 2021-01-15:
Hi @nixmeer ,
when's your project deadline ?
I could try to cook up not-so-clean CMake integration with existing master.

NikhilPatale, 2021-01-15:
Hi @avsheth ,
Could you please provide a CMake integration file with the existing master as you mentioned above? I am running on a deadline too and I have to integrate Alexa with a lot of other functionalities. So if you could provide the CMake file along with the master, that could save me from a lot of hassle right from using a virtual box for Linux to installing esp-idf on it.

avsheth, 2021-01-15:
Hi @NikhilPatale 
Sure. Please give me some time. I'll try to squeeze some time to get to this.

nixmeer, 2021-01-15:
Hello @avsheth :) 
My project is basically done, I'm just writing and I could somehow solve my problem by just initializing the ES8388 audio codec chip on my LyraT manually instead of using the board init provided here. So there's absolutely no hurry to get it done for my project.
Best
Jan

NikhilPatale, 2021-01-15:
> Hi @NikhilPatale
> Sure. Please give me some time. I'll try to squeeze some time to get to this.

Hi @avsheth,
Thanks for working on this! This will definitely save a lot of time.

NikhilPatale, 2021-02-27:
> > Hi @NikhilPatale
> > Sure. Please give me some time. I'll try to squeeze some time to get to this.
> 
> Hi @avsheth,
> Thanks for working on this! This will definitely save a lot of time.

Hi @avsheth,
Any projected timeline for this or the updated version of sdk?

priyansh-shankhdhar, 2021-06-18:
do you all know any project repository for LyraT v4.3, where i can use Tensorflow lite model to process custom wake-word, because in skainet repository they use their own wake word engine so how can use our own model?

vikramdattu, 2021-06-18:
Hi @priyansh-shankhdhar,

We are planning a release of `esp-va-sdk` which will have IDF v4.2 and cmake support.

You can easily integrate TFLite project into this to have a custom WW.