espressif/esp-va-sdk #123: Makefile doesn't let you to move the example

lorenzomerici, 2020-12-07:
Hello, 
just a consideration. For whome who wants to edit the examples would be better to copy the folder and edit it somewhere else, creating a git repository. So you can pull esp-va-sdk without problems. 

Wouldn't be better to edit make file with something like this: 

> ifeq ($(ESP_VA_PATH),)
>   $(error Please specify esp_va_sdk path via ESP_VA_PATH)
> endif
> 
> EXTRA_COMPONENT_DIRS += $(AUDIO_BOARD_PATH)/../../../ $(AWS_IOT_PATH)/
> EXTRA_COMPONENT_DIRS += $(ESP_VA_PATH)/components $(ESP_VA_PATH)/examples/common $(ESP_VA_PATH)/examples/additional_components

In order to let this possibility? Or maybe it's already present and i missed it...
Just a suggestion, have a nice day ;)

chiragatal, 2020-12-11:
Sorry, I did not really understand.

If you are copying the example to a different location, you can always add the `ESP_VA_PATH` as you have mentioned above. I think that should work.

lorenzomerici, 2020-12-11:
In esp-idf you can just copy an example folder, move somewhere else, export IDF_PATH and compile. In esp-va-sdk you can't, since there is not VA_PATH variable... and you have to edit makefile or you get a linking error.

closed 2020-12-11 by lorenzomerici