espressif/esp-va-sdk #125: Using Espressif's AWS Account Error

cmm711, 2020-12-11:
Dear Sir,
I got a problem when opening the URL of Certificate Generation on using Espressif's AWS Account.
That  URL seems like 404 already. Please check.
Thanks.

chiragatal, 2020-12-11:
The Certificate Generation URL seems to be working fine: https://espressif.github.io/esp-va-sdk/#/
Can you try again? Maybe try with a different network?

cmm711, 2020-12-11:
Well, looks like it was not working on mainland China network.

It works fine when I change to HK network.

Thank you very much.

 

Best Regards.

说明: 说明: 说明: hswlogo5

陈铭铭 CMM

Mobile: +86-13798286987

Email  :  <mailto:cmm@hsunwealth.com> cmm@hsunwealth.com

Office <tel:+86>  Tel: +86-755-2373-2403

地址: 深圳市宝安区31区前进一路186号东方明珠商务大厦603室

closed 2020-12-11 by cmm711