espressif/esp-va-sdk #126: Compiling and flashing succesful but no leds and no wakeword reaction

Rappelbox, 2020-12-17:
Hi guys,

Pretty new to all this stuff but got it so far that i can compile and flash to the vaquita board.
AWS, Wifi and genaral functionality is given through the esp and alexa apps
leds or wakewords/buttons are not working
audio DOES
Although, there are some errors with unused/deprecated warnings and errors regarding the following folder and file:
 esp-va-sdk/components/audio_hal/esp_codec/es8311/components/codec_es8311/media_hal_codec_init.c 

additionally the versioning of both cmake and xtensa doesn't seem to match to what is described in various guides (which are also not entirely consistent)

I'd like to learn where and how i have to change things

chiragatal, 2020-12-18:
Could you share your device log?

Using the release/v4.0 branch of esp-idf, and running the export.sh command should point to the correct toolchain version (esp-2020r3).

If you have provisioned the device successfully, and in the logs you can see that the device is in `IDLE` state (and it says "Your Alexa device is ready"), then maybe it could just be a problem with the connecting cable.

Rappelbox, 2020-12-18:
sometimes in life it's easy things... 👍 
The flat cable was in wrong direction. Thanks for pointing to it.

closed 2020-12-18 by Rappelbox