espressif/esp-va-sdk #128: Cannot get past [alexa]: Waiting for authentication tokens.

SamMcCormick61, 2021-02-08:
Have tried 20-30 times on different phones android and apple.
Unable to get past the  [alexa]: Waiting for authentication tokens message. Lights remains blue
Android ESP Alexa App allows to provision every time.
Alexa remains offline.

Some times on rebooting if I press button 1 it allows to reset

here is a monitor log.

C:\Users\Sam\Documents\GitHub\esp-va-sdk\examples\amazon_aia>idf.py -pCOM4 monitor
Executing action: monitor
Running idf_monitor in directory c:\users\sam\documents\github\esp-va-sdk\examples\amazon_aia
Executing "C:\Users\Sam\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe C:\Users\Sam\esp\esp-idf\tools/idf_monitor.py -p COM4 -b 115200 --toolchain-prefix xtensa-esp32-elf- c:\users\sam\documents\github\esp-va-sdk\examples\amazon_aia\build\aia.elf -m 'C:\Users\Sam\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe' 'C:\Users\Sam\esp\esp-idf\tools\idf.py' '-pCOM4'"...
?[0;33m--- WARNING: GDB cannot open serial ports accessed as COMx?[0m
?[0;33m--- Using \\.\COM4 instead...?[0m
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7084
load:0x40078000,len:13240
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (29) boot: compile time 00:12:12
I (29) boot: chip revision: 3
I (32) boot.esp32: SPI Speed      : 80MHz
I (37) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 16MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 fctry_aia        WiFi data        01 02 00010000 00006000
I (85) boot:  3 fctry            WiFi data        01 02 00016000 00006000
I (92) boot:  4 otadata          OTA data         01 00 0001c000 00002000
I (100) boot:  5 ota_0            OTA app          00 10 00020000 003f0000
I (107) boot:  6 ota_1            OTA app          00 11 00410000 003f0000
I (115) boot: End of partition table
I (119) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1f9dac (2071980) map
I (793) esp_image: segment 1: paddr=0x00219dd4 vaddr=0x3ffbdb60 size=0x04540 ( 17728) load
I (800) esp_image: segment 2: paddr=0x0021e31c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/Sam/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (801) esp_image: segment 3: paddr=0x0021e728 vaddr=0x40080404 size=0x018f0 (  6384) load
I (812) esp_image: segment 4: paddr=0x00220020 vaddr=0x400d0020 size=0x11d300 (1168128) map
0x400d0020: _stext at ??:?

I (1193) esp_image: segment 5: paddr=0x0033d328 vaddr=0x40081cf4 size=0x17670 ( 95856) load
0x40081cf4: start_cpu0_default at C:/Users/Sam/esp/esp-idf/components/esp32/cpu_start.c:361

I (1243) boot: Loaded app from partition at offset 0x20000
I (1243) boot: Disabling RNG early entropy source...
I (1244) psram: This chip is ESP32-D0WD
I (1249) spiram: Found 64MBit SPI RAM device
I (1253) spiram: SPI RAM mode: flash 80m sram 80m
I (1259) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1266) cpu_start: Pro cpu up.
I (1270) cpu_start: Application information:
I (1275) cpu_start: Project name:     aia
I (1279) cpu_start: App version:      1.0
I (1284) cpu_start: Compile time:     Feb  8 2021 00:11:59
[... 191 lines cut ...]
[auth-delegate]: Waiting for voice_assistant initialisation.
I (63538) wifi_prov_mgr: Provisioning stopped
I (63538) [app_prov]: Provisioning stopped
I (63538) wifi_prov_scheme_ble: BTDM memory released
I (63548) [va_nvs_utils]: No value set for: friendlyname
I (63548) [va_nvs_utils]: No value set for: udn
I (63558) SSDP: Network Interface List (1):
I (63558) SSDP: 1. wifi  : 192.168.1.213
[alexa]: Waiting for time to be updated
I (63568) SSDP:

I (63578) LSSDP: create SSDP socket 58
[auth-delegate]: Waiting for voice_assistant initialisation.
[auth-delegate]: Waiting for voice_assistant initialisation.
[alexa]: Done getting current time: 1612808751
I (66998) [tls_certification]: Done setting global CA store
I (66998) [va_nvs_utils]: No value set for: clientId
I (66998) [va_nvs_utils]: No value set for: refreshToken
I (67008) [alexa]: Checking for companion app based authentication...

[alexa]: Waiting for authentication tokens.
W (67608) httpd_uri: httpd_uri: URI '/rootDesc.xml' not found
W (67608) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
W (67618) httpd_uri: httpd_uri: URI '/rootDesc.xml' not found
W (67618) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist

chiragatal, 2021-02-09:
Hi @SamMcCormick61 ,
1. Could you check your esp-idf branch? I think you are using v4.2 but it should be v4.0.
2. Can you tell which Android phone you are using? Could you try installing the Amazon Shopping app, sign-in into that with your account and try provisioning with ESP Alexa app again?
3. Also, make sure you are not 'skipping' the sign-in part and that it is successful in the ESP Alexa app.

SamMcCormick61, 2021-02-09:
Thanks,
I will clear everything down and get esp-idf 4.0

I used a Samsung Note 8
Amazon shopping was installed,
I also have the Alexa app for my other echo dot and devices
Yes I believe the app is showing success as it gets to the screen asking you to ask Alexa something

I have also ordered another board from Diki-key for comparison.

I post here if I am successful or not.

SamMcCormick61, 2021-02-09:
It is now working.

I  assumed my existing installation of esp-idf would work. I have learnt  my lesson.

Thank you for your prompt reply.

Now to try with my account

Sam

SamMcCormick61, 2021-02-09:
All fixed

closed 2021-02-09 by SamMcCormick61