espressif/esp-va-sdk #129: My Account certs, keys do not work - provisioned ok but always of line. Espressif Certs and key work

SamMcCormick61, 2021-02-17:
Can connect with ESPRESSIF credentials (mfg.bin). Alexa responds very well

However Only get so far with My AWS Accounts certificate and private key.  Alexa says having trouble - ( the device is offline.)
--------------------------------------------------------------------------
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/connection/fromclient, {"header":{"name":"Connect","messageId":"cd8e27e9-1ba2-7294-81a6-c8a5519a760b"},"payload":{"awsAccountId":"407723792719","clientId":"3caadd4c-b63e-4bbb-9b99-561572e14bb5"}}
----------------------------------------------------------
My certificates  and endpoints seem to work (tested via openssl) and are all liked to my the IOT thing I created in AWS IOT CORE

            "certificateArn": "arn:aws:iot:us-east-1:407723792719:cert/8b76d21430393c954ec7aa390bf47705224685cacf6c2113e7d9e259017c3ee6",
            "certificateId": "8b76d21430393c954ec7aa390bf47705224685cacf6c2113e7d9e259017c3ee6",
            "status": "ACTIVE",
            "creationDate": "2021-02-06T23:33:46.231000+00:00"

ESP-Alexa-Policy is linked to the certificate and the certificate is active

device private key,device certificate ,server.key  are in the  mfg.bin .

Tried to debug on the Amazon side  -- not sure where to look for any logs giving me a hint as to what is failing or not connecting

Any help would be appreciated

---------------------------------------
I (2306) wifi:mode : sta (24:0a:c4:e6:d9:b4)
I (2316) wifi:Set ps type: 1

I (4356) wifi:new:<3,0>, old:<1,0>, ap:<255,255>, sta:<3,0>, prof:1
I (4996) wifi:state: init -> auth (b0)
I (5006) wifi:state: auth -> assoc (0)
I (5006) wifi:state: assoc -> run (10)
I (5026) wifi:connected with Jacob, aid = 2, channel 3, BW20, bssid = 9c:80:df:bc:e3:9d
I (5026) wifi:security: WPA2-PSK, phy: bgn, rssi: -36
I (5036) wifi:pm start, type: 1

I (5126) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[app_wifi]: Connected with IP Address: 192.168.1.213
I (6446) tcpip_adapter: sta ip: 192.168.1.213, mask: 255.255.255.0, gw: 192.168.1.254
I (6446) [alexa_local_config]: Using saved device friendly name: ESP-Alexa-D9B4
I (6456) SSDP: Network Interface List (1):
I (6456) SSDP: 1. wifi  : 192.168.1.213
[alexa]: Waiting for time to be updated
I (6466) SSDP:

I (6476) LSSDP: create SSDP socket 58
[alexa]: Done getting current time: 1613521789
I (9996) [tls_certification]: Done setting global CA store
I (10006) auth-delegate-config: Client ID or Refresh token found in NVS
I (10006) [va_nvs_utils]: No value set for: clientSecret
I (10006) auth-delegate-config: Returning auth delegate subsequent auth configuration
[alexa]: Authentication done
I (10036) [app_va_cb]: Dialog state is: 8
[dialog]: Entering VA_IDLE
I (10076) [audio_codec]: Starting opus_decoder codec
I (10106) [va_nvs_utils]: No value set for: alerts_vol
I (10116) audio_stream: Starting http_stream stream
I (10116) [audio_codec]: Starting mp3_decoder codec
I (10116) [va_nvs_utils]: No value set for: alert_0_data
I (10116) [va_nvs_utils]: No value set for: alert_1_data
I (10126) [va_nvs_utils]: No value set for: alert_2_data
I (10126) [va_nvs_utils]: No value set for: alert_3_data
I (10136) [va_nvs_utils]: No value set for: alert_4_data
I (10146) [va_nvs_utils]: No value set for: alert_5_data
I (10146) [va_nvs_utils]: No value set for: alert_6_data
I (10156) [va_nvs_utils]: No value set for: alert_7_data
I (10156) [va_nvs_utils]: No value set for: alert_8_data
I (10166) [va_nvs_utils]: No value set for: alert_9_data
I (10176) [va_nvs_utils]: No value set for: alert_10_data
I (10176) [alerts_nvs]: Alerts loaded from NVS
W (11316) httpd_uri: httpd_uri: URI '/rootDesc.xml' not found
W (11316) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
W (11326) httpd_uri: httpd_uri: URI '/rootDesc.xml' not found
W (11326) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
[auth-delegate]: Token will be refreshed after 3000 seconds.
[http_transport]: AWS Account ID: 407723792719
[http_transport]: Client ID: 3caadd4c-b63e-4bbb-9b99-561572e14bb5
[http_transport]: Current time in London: Wed Feb 17 00:29:51 2021
I (11926) [http_transport]: String: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/connection/fromclient
I (11936) [http_transport]: String: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/connection/fromservice
I (11946) [http_transport]: String: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/capabilities/publish
I (11956) [http_transport]: String: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/capabilities/acknowledge
I (11976) [http_transport]: String: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/event
I (11976) [http_transport]: String: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/directive
I (11986) [http_transport]: String: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/microphone
I (11996) [http_transport]: String: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/speaker
I (12006) [http_transport]: Connecting to AWS...
I (12096) [va_nvs_utils]: No value set for: d2_fw_debug
I (12096) [va_nvs_utils]: No value set for: d4p_fw_debug
I (12096) [va_nvs_utils]: No value set for: d2_audio_debug
I (12096) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (12106) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (12116) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (12126) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (14696) [dspg_utility]: Successful switch to TDM0 clock
I (14706) [dspg_utility]: 5a 02 18 06 00 00 00 00 00 00
Transport Shared Secret found in NVS::
0x69 0xf3 0x99 0x79 0x27 0x67 0x44 0xb0 0xdc 0x58 0xbb 0x02 0xfa 0x87 0x2f 0x80 0xd2
0xfd 0x62 0xda 0x81 0xb1 0xa8 0x41 0x04 0x53 0x0d 0xe1 0x85 0x93 0x14 0x45
I (14766) [http_transport]: Subscribing /capabilities/acknowledge...
I (14976) [http_transport]: Subscribing /connection/fromservice...
I (15186) [http_transport]: Subscribing /directive...
I (15376) [http_transport]: Subscribing /speaker...
I (16366) [dspg_driver]: Calling start capture
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/connection/fromclient, {"header":{"name":"Connect","messageId":"cd8e27e9-1ba2-7294-81a6-c8a5519a760b"},"payload":{"awsAccountId":"407723792719","clientId":"3caadd4c-b63e-4bbb-9b99-561572e14bb5"}}
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/connection/fromclient, {"header":{"name":"Connect","messageId":"927820a0-a1ef-64bc-9a6c-5fabbb74b7c0"},"payload":{"awsAccountId":"407723792719","clientId":"3caadd4c-b63e-4bbb-9b99-561572e14bb5"}}
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/connection/fromclient, {"header":{"name":"Connect","messageId":"c1f07181-7995-d299-96fc-51911076d038"},"payload":{"awsAccountId":"407723792719","clientId":"3caadd4c-b63e-4bbb-9b99-561572e14bb5"}}

chiragatal, 2021-02-17:
Hi @SamMcCormick61 , Can you factory reset and try again?

Once provisioned again, the device will most likely get errors in AIA device registration, if you are not using your own phone app. If this happens, I could temporarily link your AWS account with Espressif's AVS product and it should work.

SamMcCormick61, 2021-02-17:
Chiragatal

I have tried with My Version of the phone app and it provisions ok after factory reset .  
I will try again from scratch (clear flash) Then using my app. (I will record the monitor logs whilst provisioning)

If this does not work I will ask you for a temporary link with Espressifs AVS  your product.

Many thanks
Sam

SamMcCormick61, 2021-02-17:
Hi @chiragatal 

I have setup the Android Provision app as per the
 (https://github.com/espressif/esp-idf-provisioning-android/tree/versions/avs-ble)
espressif/esp-idf-provisioning-android  readme.md and generated key via Andriod studio and set up the app on amazon downloading the api_key.txt  and install on my phone.
I do not get any
I (248606) auth-delegate-config: 
 [alexa_auth_delegate]: Registration Request Buffer: {"authentication":
mesages

It is if the App is not reading the amazon shopping credentials from my phone (as the espressif app does)
Also the apps are very different on how they look 

When I provision my provisioning app says ok .   On monitoring I only get this far - 
....Checking for companion app based authentication...

---------------------------------------------------------------
I (49446) [app_prov]: Provisioning successful
I (53136) wifi_prov_mgr: Provisioning stopped
I (53136) [app_prov]: Provisioning stopped
I (53136) wifi_prov_scheme_ble: BTDM memory released
I (53146) [va_nvs_utils]: No value set for: friendlyname
I (53146) [va_nvs_utils]: No value set for: udn
I (53156) SSDP: Network Interface List (1):
I (53156) SSDP: 1. wifi  : 192.168.1.213
[alexa]: Waiting for time to be updated
I (53166) SSDP:

I (53176) LSSDP: create SSDP socket 58
[alexa]: Done getting current time: 1613554952
I (54516) [tls_certification]: Done setting global CA store
I (54516) [va_nvs_utils]: No value set for: clientId
I (54516) [va_nvs_utils]: No value set for: refreshToken

I (54526) [alexa]: Checking for companion app based authentication...

[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.

-----------------------------------------------------------------
When I use Espressifs certificate/key and provisiong app

I (245076) [va_nvs_utils]: No value set for: refreshToken

I (245086) [alexa]: Checking for companion app based authentication...

[alexa]: Waiting for authentication tokens.
W (245606) httpd_uri: httpd_uri: URI '/rootDesc.xml' not found
W (245606) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
W (245616) httpd_uri: httpd_uri: URI '/rootDesc.xml' not found
W (245616) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
[alexa]: Waiting for authentication tokens.
I (248606) auth-delegate-config: Set client ID amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9 and refresh token Atzr|IwEBIIr4fmGqKuMz0MJ1MSh3exrGUyg5SxraJOBFhZ4l5_fkkNdA742bH12K3EiXNpkTgVTseTPOoCwf44OKo8ae2GPFnxQXKYDq_VuDNyQga-P9fBuY6_FCPHlzW-L-KdWZEEo3gSxNnj9DoBRHRoIHCkmNvifEX-YcEeJm3P2CGZ-acft6WGqayUPCIoA2EZP-LD8vm2ZhxcY9aEo5bMXSDCDFNxpo8EHrA3vVDjJxyBB7FAYdD7oIM-KwpvVquiBsfwygBRIq8dUOhcNIjyPojMLlfxQwofyaajSst9yMu8oiLCpkVotP80AH2v8hYL-qvue4T5BQEKq162DVcfqD7BAs in NVS successfully
[alexa]: Authentication done
I (249106) [app_va_cb]: Dialog state is: 8

---
and goes on to

[alexa_auth_delegate]: Registration Request Buffer: {"authentication":{"token":"Atzr|IwEBIIr4fmGqKuMz0MJ1MSh3exrGUyg5SxraJOBFhZ4l5_fkkNdA742bH12K3EiXNpkTgVTseTPOoCwf44OKo8ae2GPFnxQXKYDq_VuDNyQga-P9fBuY6_FCPHlzW-L-KdWZEEo3gSxNnj9DoBRHRoIHCkmNvifEX-YcEeJm3P2CGZ-acft6WGqayUPCIoA2EZP-LD8vm2ZhxcY9aEo5bMXSDCDFNxpo8EHrA3vVDjJxyBB7FAYdD7oIM-KwpvVquiBsfwygBRIq8dUOhcNIjyPojMLlfxQwofyaajSst9yMu8oiLCpkVotP80AH2v8hYL-qvue4T5BQEKq162DVcfqD7BAs", "clientId":"amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9"},"encryption":{"algorithm":"ECDH_CURVE_25519_32_BYTE", "publicKey":"P6YIitk8/K04EYfnllyyRWOxuytp5S+NpBy60YztM08="},"iot":{"awsAccountId":"282183071650", "clientId":"69fd3fb9-deb5-4c4c-895c-1e2668a91208","endpoint":"aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com","topicRoot":"$aws/alexa"}}

------------------------

SamMcCormick61, 2021-02-17:
Hi  @chiragatal

I have been using Provisioning App version avs-ble
will try buld version versions/avs

khushbushah2302, 2021-02-18:
Hi @SamMcCormick61 ,

I have some query related to phone app.
Which Espressif app you are using ? Is this [ESP Alexa](https://play.google.com/store/apps/details?id=com.espressif.provbleavs) ? 
I would like to understand how apps are looking different. 
The setup you have done from this (https://github.com/espressif/esp-idf-provisioning-android/tree/versions/avs-ble) and ESP Alexa are same.
Can you give some more details like which build variant you are using from your setup in Android studio ? (For example, bleSec1AvsRelease, bleSec0AvsRelease etc)

SamMcCormick61, 2021-02-18:
Hi @khushbushah2302

I think I download an early version first time . I have now used https://github.com/espressif/esp-idf-provisioning-android/tree/versions/avs-ble   as you have suggested
build variant bleSec1AvsDebug
installed on my phone

Amazon does not let me provision via LWA log on with Amazon see an image of the message when i press 'Sign in with Amazon'

[image]
 --------------------------
Is this an amazon message from Amazon (not the app?)

1. Keystore certificate information 
done
2. API key from LWA
done

    signingConfigs {
        config {
            keyAlias 'upload4'
            storeFile file('C:\\Users\\Sam\\AndroidStudioProjects\\keystore\\upload4.jks')
            storePassword 'xxxx'
            keyPassword 'xxxx'
        }
    }

        applicationId "com.lhd.provbleavs"

I have change this somy app has a different name
    <string name="app_name">AA Alexa SAM4</string>

Amazon security profile
API Key Name	
upload4
Identifies the app you will use with this API key.

Package	
com.lhd.provbleavs
The package name of your Android project. For example, com.mycompany.bestapp.

MD5 Signature	
49:92:21:D5:16:F3:22:6D:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
The MD5 signature of the certificate used to sign your app. Get instructions on obtaining this value.

SHA256 Signature	
76:93:62:02:95:0F:F2:30:2B:83:8A:73:3A:76:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
The SHA256 signature of the certificate used to sign your app. Get instructions on obtaining this value.

API key	
eyJhbGciOiJSU0EtU0hBMjU2IiwidmVyIjoiMSJ9.eyJ2ZXIiOiIzIiwiZW5kcG9pbnRzIjp7ImF1dGh6IjoiaHR0cHM6Ly93d3cuYW1hem9uLmNvbS9hcC9vYSIsInRva2VuRXhjaGFuZ2UiOiJodHRwczovL2FwaS5hbWF6b24uY29tL2F1dGgvbzIvdG9rZW4ifSwiY2xpZW50SWQiOiJhbXpuMS5hcHBsaWNhdGlvbi1vYTItY2xpZW50LmZjNWZlZmFmMjJjODQyZjhiZGU5NWU0ZjZjMjFiOTA0IiwiYXBwRmFtaWx5SW=
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

chiragatal, 2021-02-18:
This error in the app also shows up if the product id in the app and the device are different.
On the device side, can you change menuconfig -> voice assistant -> Device Product ID from the default esp_alexa_open to the product ID of the AVS product that you have used to set up the phone app?

SamMcCormick61, 2021-02-18:
Great we got further it now links up.

he is my monitor log now.
I (66756) [alexa_auth_delegate]: Base 64 encode response: 0
[alexa_auth_delegate]: Registration Request Buffer: {"authentication":{"token":"Atzr|IwEBIDG_q53j0lwW4Hwgq6Um4PRJCoOEWwaxsmjSE5OcWdmBp5kRHW15hjq7LbBZp94WkioJJc_vNiAfJmlRz7aEMCu-yfC9m-tE-3z1MKCuvS6J_DLhUnUnDV4bO2EaQeuhlCIj9FxZEJFYoOS8T50TpEu9pJXrfJeATZaeVaMyAWhqmiVHtrgW3EIEI4sj_04UB3j_lcGmEIx4qjd7FAZyems1Z046-DQFXfFFy538rP_JKKpqh4bIStl3a9odzJVd87GYJFRxXHr7KCfk-eZgsbj9Ecfz8hmk4y6esEc6oYhYz0zyqI6r39wEUFdNX98YeGPXa72J_AE7Eg1bw5AzIHJQ", "clientId":"amzn1.application-oa2-client.fc5fefaf22c842f8bde95e4f6c21b904"},"encryption":{"algorithm":"ECDH_CURVE_25519_32_BYTE", "publicKey":"iPs1QXmQqtPMeuTb4OKBjPNc6JC31vKCi1kejwdyBjI="},"iot":{"awsAccountId":"407723792719", "clientId":"3caadd4c-b63e-4bbb-9b99-561572e14bb5","endpoint":"ai95ozdzdx2s0-ats.iot.us-east-1.amazonaws.com","topicRoot":"$aws/alexa"}}
I (67296) [dspg_utility]: Successful switch to TDM0 clock
I (67306) [dspg_utility]: 5a 02 18 06 00 00 00 00 00 00
[alexa_auth_delegate]: Registration Response Buffer: {"encryption": {"publicKey": "yCCtE4U2pEt92iPuAmCb6GkpUy0LXokRva7A/2bwqAE="}, "iot": {"topicRoot": "$aws/alexa"}}
I (68676) [alexa_auth_delegate]: Cloud public key is yCCtE4U2pEt92iPuAmCb6GkpUy0LXokRva7A/2bwqAE=
Transport Shared Secret::
0x04 0x2b 0x34 0x98 0x0f 0xe7 0xee 0xb9 0xb2 0xf3 0xb2 0xf2 0xfa 0x6c 0xbd 0x03 0xc5
0xa8 0x8c 0x0c 0x7e 0xad 0x93 0x81 0x5c 0xff 0xfa 0x08 0xba 0x1c 0x59 0x78
I (68706) [http_transport]: Subscribing /capabilities/acknowledge...
I (68876) [http_transport]: Subscribing /connection/fromservice...
I (69016) [http_transport]: Subscribing /directive...
I (69166) [dspg_driver]: Calling start capture
I (69186) [http_transport]: Subscribing /speaker...
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/connection/fromclient, {"header":{"name":"Connect","messageId":"f2191ff1-4297-5fa5-e946-75ac346f3659"},"payload":{"awsAccountId":"407723792719","clientId":"3caadd4c-b63e-4bbb-9b99-561572e14bb5"}}
[http_transport]: Got response on topic fromservice: {"header":{"name":"Acknowledge","messageId":"49270f2e-457a-4c88-a1de-2f4ddbdd802d"},"payload":{"connectMessageId":"f2191ff1-4297-5fa5-e946-75ac346f3659","code":"CONNECTION_ESTABLISHED"}}
[capabilities]: Capabilities version not found in NVS or is different
I (71936) [capabilities]: Capabilities: {"header":{"name":"Publish","messageId":"a6ae6c43-34e1-55ef-f6b3-6cf8a933fe5a"},"payload":{"capabilities":[{"type":"AisInterface","interface":"System","version":"1.0","configurations":{"mqtt":{"message":{"maxSizeInBytes":8015}},"firmwareVersion":"42","locale":"en-US"}},{"type":"AisInterface","interface":"Microphone","version":"1.0","configurations":{"audioEncoder":{"format":"AUDIO_L16_RATE_16000_CHANNELS_1"}}},{"type":"AisInterface","interface":"Speaker","version":"1.0","configurations":{"audioBuffer":{"sizeInBytes":100000,"reporting":{"overrunWarningThreshold":90000,"underrunWarningThreshold":5000}},"audioDecoder":{"format":"OPUS","bitrate":{"type":"CONSTANT","bitsPerSecond":64000},"numberOfChannels":2}}},{"type":"AisInterface","interface":"Alerts","version":"1.0","configurations":{"maxAlertCount":10}},{"type":"AisInterface","interface":"EndpointForwarding","version":"1.0"}]}}
I (72016) [http_transport]: Sending to topic: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/capabilities/publish
I (72266) [http_transport]: Got capabilities acknowledge
I (72266) [http_transport]: Capabilities decrypted data: {"header":{"name":"Acknowledge","messageId":"70ed5569-e9ae-423f-9b26-26c2ab38aa23"},"payload":{"publishMessageId":"a6ae6c43-34e1-55ef-f6b3-6cf8a933fe5a","code":"CAPABILITIES_ACCEPTED","description":""}}
I (72276) [capabilities]: Setting capabilities digest in NVS
[alexa_discovery]: Digest has been changed
[alexa_discovery]: Smarthome capabilities version not found in NVS or is different
[70 seconds]: [http_transport]: Event data: {"events":[{"header":{"name":"EndpointForwarding","messageId":"809a262d-7829-4af4-cb14-26797a650b6f"},"payload":{"event": {"header": {"namespace": "Alexa.Discovery","name": "AddOrUpdateReport","payloadVersion": "3","messageId": "809a262d-7829-4af4-cb14-26797a650b6f","eventCorrelationToken": "809a262d-7829-4af4-cb14-26797a650b6f"},"payload": {"endpoints":[{"endpointId": "amzn1.application-oa2-client.fc5fefaf22c842f8bde95e4f6c21b904::LHD1::240ac4e6d9b4_Light","registration": {"productId": "LHD1","deviceSerialNumber": "d4b1f92d-318f-d24b-deef-bac39615c1aa"},"manufacturerName": "Espressif","description": "Alexa Device","friendlyName": "Light","displayCategories": ["LIGHT"],"additionalAttributes": {"manufacturer": "Espressif","model": "ESP32","serialNumber": "d4b1f92d-318f-d24b-deef-bac39615c1aa","firmwareVersion": "v1.2","softwareVersion": "v1.2","customIdentifier": "d4b1f92d-318f-d24b-deef-bac39615c1aa"},"capabilities": [{"type":"AlexaInterface","interface":"Alexa.PowerController","version":"3","properties": {"supported": [{"name": "powerState"}],"proactivelyReported": true,"retrievable": true}},{"type":"AlexaInterface","interface":"Alexa.RangeController","version":"3","instance": "Brightness","properties": {"supported": [{"name": "rangeValue"}],"proactivelyReported": true,"retrievable": true,"nonControllable": false},"configuration": {"supportedRange": {"minimumValue": 0,"maximumValue": 100,"precision": 1}},"capabilityResources":{"friendlyNames":[{"@type":"text","value":{"text":"Brightnes [... cut ...]
I (72436) [http_transport]: Sending to topic: $aws/alexa/ais/v1/3caadd4c-b63e-4bbb-9b99-561572e14bb5/event
[70 seconds]: [http_transport]: Free Memory Internal: 77324, External: 2569604
[directive_proc]: Json data: {"directives":[{"header":{"name":"Exception","messageId":"20331f44-8681-488c-9a3d-ceabc2022c29"},"payload":{"code":"INVALID_REQUEST","description":"Invalid first event, event: EndpointForwarding, desired: SynchronizeState"}}]}
I (72896) [directive_proc]: Name: Exception

E (72906) [directive_proc]: No handler found for :Exception, ignoring...

SamMcCormick61, 2021-02-18:
INVALID_REQUEST
[70 seconds]: [http_transport]: Free Memory Internal: 77324, External: 2569604
[directive_proc]: Json data: {"directives":[{"header":{"name":"Exception","messageId":"20331f44-8681-488c-9a3d-ceabc2022c29"},"payload":{"code":"INVALID_REQUEST","description":"Invalid first event, event: EndpointForwarding, desired: SynchronizeState"}}]}

khushbushah2302, 2021-02-19:
Hi @SamMcCormick61 ,

The error in Login With Amazon screen seems product id mismatch. 
After changing product id on device as suggested by @chiragatal ,  it should work.
I need some logs from app side to debug the issue.
Can you share phone app logs after doing below changes ?
Make debuggable flag to true at [this line](https://github.com/espressif/esp-idf-provisioning-android/blob/872bc36042a17b2a2062425fd8f38ef9dabd209d/app/build.gradle#L26) and use `bleSec1AvsRelease` build variant.
You can see logs in "Logcat" (in bottom toolbar of Android studio).

actiontang, 2021-02-19:
Hi  @khushbushah2302 & @chiragatal , 

I meet same @SamMcCormick61  issue

Our question is not in certification , Android APP or  product ID. 

We need to know how to get "Event data: {"events":[{"header":{"name":"EndpointForwarding",..." response as Espressif Demo ?

1. As above @SamMcCormick61 report , we always get as below response :
  **[directive_proc]: Json data: {"directives":[{"header":{"name":"Exception","messageId":"20331f44-8681-488c-9a3d-ceabc2022c29"},"payload":{"code":"INVALID_REQUEST","description":"Invalid first event, event: EndpointForwarding, desired: SynchronizeState"}}]}**

2. But , in Espressif Demo cert.. can get as below OK message:
**[directive_proc]: Json data: {"directives":[{"header":{"name":"EndpointForwarding","messageId":"0cb0a8aa-19b2-4435-ae3a-5167f4b8a7bb"},"payload":{"directive":{"header":{"namespace":"Alexa","name":"EventProcessed","messageId":"dba3b5b0-9619-48f9-ae3d-e355bf1ca8ea","eventCorrelationToken":"d65bda9e-a5f1-ed89-d725-c86d06abfd00"},"payload":"{}"}}}]}
I (155447) [directive_proc]: Name: EndpointForwarding**

And how to get "EndpointForwarding" ok response ?

SamMcCormick61, 2021-02-19:
@khushbushah2302 

I changed the product and I believe now it registered correctly wilt amazon -- My account Amazon show 1 registration against my device (LHD1). (does amazon have any logs to look at?)

It just this exception something to do with expecting SynchronizeState which is different to Espressifs (working) logs

[directive_proc]: Json data: {"directives":[{"header":{"name":"Exception","messageId":"20331f44-8681-488c-9a3d-ceabc2022c29"},"payload":{"code":"INVALID_REQUEST","description":"Invalid first event, event: EndpointForwarding, desired: SynchronizeState"}}]}

-----------------------------------------------------------------------------------
I any case .

I will set the debug flag
    buildTypes {
        release {
            debuggable true

I will use bleSec1AvsRelease

I will send back the logs to you.

Thanks
Sam

chiragatal, 2021-02-19:
@SamMcCormick61 
For the above exception, can you comment the app_smart_home_init() in app_main.c and try?
smart-home support needs to be white-listed for your AWS account for it to work. Since your account is not white-listed yet, you are getting the exception. (We will be adding this to the documentation soon.)

SamMcCormick61, 2021-02-19:
@chiragatal 
Ok thanks
I will comment out the  app_smart_home_init() in app_main.c 

I will look into having my account white-listed  (although I am only a small player at the moment)
If i am only testing why do i need white listing -- I shall ask amazon.

Another question where are the debug logs stored on my phone if I turn on debugging as suggested before

I the documentation can you add aline to make the ProductId need to be set in menuconfig . Mention thei in the documentation for both in the APP project and in the esp-idf project.

SamMcCormick61, 2021-02-19:
@chiragatal @khushbushah2302 
Success.
It now works.
Road to my success

1. Make sure you are using ESP-IDF 4.0
2. Request and get Espressif keys and cerificats to test devise using Espressifs. make sure the private ley is used
3. When you set up you amazon product make a note of your Product id and add to the ProductId via menuconfig
4. Make sure you are using the correct version of the Provisioning app APP Version 2.1.1 (same versionas ESP Aleax espressif uses in play store)
5. I you get a INVALID_REQUEST Invalid first event error in the Monitor logs (UART serial) then  your amazon account is not WHITE LISTED.  You need to  comment out the app_smart_home_init() line in the app_main.c  of the amazon_aia esp project.

-----------------------
    /* app_cloud_init() and alexa_provisioning_init() should be called after wifi_init(), but before prov_init() */
#ifdef CONFIG_ALEXA_ENABLE_CLOUD
    app_cloud_init();
#else /* !CONFIG_ALEXA_ENABLE_CLOUD */
 /*   app_smart_home_init(); */
#endif /* CONFIG_ALEXA_ENABLE_CLOUD */
    alexa_provisioning_init(amazon_cfg);
---------------------------
 
I am not sure how to be come white listed.?

Hope this helps someone else.

chiragatal, 2021-02-19:
Great!
For whitelisting your account for smart-home, you'll have to contact Amazon. If you do not have any point of contact in Amazon yet, then maybe you can check the 'contact us' section on the developer portal.

SamMcCormick61, 2021-02-19:
Thanks,
 I will contact Amazon.
Now that I all work I will set up an extra device as I have another  Vaquita board.
This can be closed now.

closed 2021-02-19 by SamMcCormick61