espressif/esp-va-sdk #130: Can't see "Light" smart home skill in the Alexa APP after connected AVS

actiontang, 2021-02-22:
HI, 

I use ESP32 Demo project and  Espressif's AWS Account/ Generate Certificate.
As below URL 9.2.1  said " In the Alexa app, this device will show up as 'Light' and the Power and Brightness can be controlled."

But I can't see "Light" smart home sill in the Alexa APP after connected AVS
Only can see "my's ESP Alexa" AVS Product device  in the Echo & Alexa Page of the Alexa APP. 

As blew my log:

[alexa_auth_delegate]: Registration Request Buffer: {"authentication":{"token":"Atzr|IwEBIEd3ySHxsOEKeK3lvBfP8ce878nW-22ahp5qeaSEPYcPwnHk3GwzIrccAEF0u9avwYVsmN2QFVRQkmrsoq2QmzhM4kvM4xoME-hyjqsbDwuggrVvqf5rQAojQUCn3dRWlEfL7ZzvwCos8_ol1jOstp3G-2ykkgiFuVohcP3uSIaiVXDLTVC2jqCvmaG60LJpWRmgzCtJwbZEryi3OZq7HRDvSKTI3VVhvNo7tXD34SfG7NgWjZPWOEaRkroEpK4y_MqN5DibP1cNZplxDuH6c0hJ2GAOSW25uvRc2UIY9-YKcaMS1_TVfpauW8nZhJOXFqlX5dUfGYsHKFISJw20o1fWZQT7qDej8KWuJMDwofywpA", "clientId":"amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9"},"encryption":{"algorithm":"ECDH_CURVE_25519_32_BYTE", "publicKey":"kDb7e2veZTnv0/GwaNlUunR9LZ7oGqz7wIKhbIqJv1k="},"iot":{"awsAccountId":"282183071650", "clientId":"482fa2bd-116d-4780-bb37-a75415875881","endpoint":"aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com","topicRoot":"$aws/alexa"}}
[alexa_auth_delegate]: Registration Response Buffer: {"encryption": {"publicKey": "hcVThYfUEGU7GkiSExm7knyBlm/e2uy22xDFyByBWns="}, "iot": {"topicRoot": "$aws/alexa"}}
I (22057) [alexa_auth_delegate]: Cloud public key is hcVThYfUEGU7GkiSExm7knyBlm/e2uy22xDFyByBWns=
Transport Shared Secret::
0x8c 0xab 0x2d 0xc6 0x27 0x1b 0x94 0x1f 0x1f 0xfd 0xbb 0xe3 0xe7 0x73 0xcd 0x57 0x2d
0x52 0xef 0x77 0xc7 0x0f 0x72 0x9d 0xde 0x01 0x8e 0xa3 0x7f 0x63 0x94 0x4b
I (22087) [http_transport]: Subscribing /capabilities/acknowledge...
I (22767) [http_transport]: Subscribing /connection/fromservice...
I (23287) [http_transport]: Subscribing /directive...
I (24407) [http_transport]: Subscribing /speaker...
[http_transport]: Sending unencrypted to topic: $aws/alexa/ais/v1/482fa2bd-116d-4780-bb37-a75415875881/connection/fromclient, {"header":{"name":"Connect","messageId":"d126b60d-a2ba-74ef-fa9f-e1b100167635"},"payload":{"awsAccountId":"282183071650","clientId":"482fa2bd-116d-4780-bb37-a75415875881"}}
[http_transport]: Got response on topic fromservice: {"header":{"name":"Acknowledge","messageId":"1c98b839-37bd-489d-beca-72c6d105d550"},"payload":{"connectMessageId":"d126b60d-a2ba-74ef-fa9f-e1b100167635","code":"CONNECTION_ESTABLISHED"}}
[capabilities]: Capabilities version not found in NVS or is different
I (28497) [capabilities]: Capabilities: {"header":{"name":"Publish","messageId":"dd54f479-e140-c8ce-b32f-ac81697ccdc2"},"payload":{"capabilities":[{"type":"AisInterface","interface":"System","version":"1.0","configurations":{"mqtt":{"message":{"maxSizeInBytes":8015}},"firmwareVersion":"42","locale":"en-US"}},{"type":"AisInterface","interface":"Microphone","version":"1.0","configurations":{"audioEncoder":{"format":"AUDIO_L16_RATE_16000_CHANNELS_1"}}},{"type":"AisInterface","interface":"Speaker","version":"1.0","configurations":{"audioBuffer":{"sizeInBytes":100000,"reporting":{"overrunWarningThreshold":90000,"underrunWarningThreshold":5000}},"audioDecoder":{"format":"OPUS","bitrate":{"type":"CONSTANT","bitsPerSecond":64000},"numberOfChannels":2}}},{"type":"AisInterface","interface":"Alerts","version":"1.0","configurations":{"maxAlertCount":10}},{"type":"AisInterface","interface":"EndpointForwarding","version":"1.0"}]}}
I (28567) [http_transport]: Sending to topic: $aws/alexa/ais/v1/482fa2bd-116d-4780-bb37-a75415875881/capabilities/publish
I (29237) [http_transport]: Got capabilities acknowledge
I (29237) [http_transport]: Capabilities decrypted data: {"header":{"name":"Acknowledge","messageId":"60560fe7-ed6d-49f5-afa0-87ef29d414e4"},"payload":{"publishMessageId":"dd54f479-e140-c8ce-b32f-ac81697ccdc2","code":"CAPABILITIES_ACCEPTED","description":""}}
I (29247) [capabilities]: Setting capabilities digest in NVS
[alexa_discovery]: Digest has been changed
[alexa_discovery]: Smarthome capabilities version not found in NVS or is different
[27 seconds]: [http_transport]: Event data: {"events":[{"header":{"name":"EndpointForwarding","messageId":"86bf8ba5-d410-53e8-988d-e9dcab3ab738"},"payload":{"event": {"header": {"namespace": "Alexa.Discovery","name": "AddOrUpdateReport","payloadVersion": "3","messageId": "86bf8ba5-d410-53e8-988d-e9dcab3ab738","eventCorrelationToken": "86bf8ba5-d410-53e8-988d-e9dcab3ab738"},"payload": {"endpoints":[{"endpointId": "amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9::esp_alexa_open::7c9ebdcf10fc_Light","registration": {"productId": "esp_alexa_open","deviceSerialNumber": "00d73716-7d54-5670-cd8c-876f79e963b0"},"manufacturerName": "Espressif","description": "Alexa Device","friendlyName": "Light","displayCategories": ["LIGHT"],"additionalAttributes": {"manufacturer": "Espressif","model": "ESP32","serialNumber": "00d73716-7d54-5670-cd8c-876f79e963b0","firmwareVersion": "v1.2","softwareVersion": "v1.2","customIdentifier": "00d73716-7d54-5670-cd8c-876f79e963b0"},"capabilities": [{"type":"AlexaInterface","interface":"Alexa.PowerController","version":"3","properties": {"supported": [{"name": "powerState"}],"proactivelyReported": true,"retrievable": true}},{"type":"AlexaInterface","interface":"Alexa.RangeController","version":"3","instance": "Brightness","properties": {"supported": [{"name": "rangeValue"}],"proactivelyReported": true,"retrievable": true,"nonControllable": false},"configuration": {"supportedRange": {"minimumValue": 0,"maximumValue": 100,"precision": 1}},"capabilityResources":{"friendlyNames":[{"@type":"text","value":{"text":"Brightness","locale":"en-US"}}]}}],"connections":[]}]}}}}]}
I (29407) [http_transport]: Sending to topic: $aws/alexa/ais/v1/482fa2bd-116d-4780-bb37-a75415875881/event
[28 seconds]: [http_transport]: Free Memory Internal: 77784, External: 2568500
[directive_proc]: Json data: {"directives":[{"header":{"name":"EndpointForwarding","messageId":"dbc879f6-4664-4157-bd79-f02a498f75fe"},"payload":{"directive":{"header":{"namespace":"Alexa","name":"EventProcessed","messageId":"75718e66-65f5-429f-ada0-6b986b61edbb","eventCorrelationToken":"86bf8ba5-d410-53e8-988d-e9dcab3ab738"},"payload":"{}"}}}]}
I (30277) [directive_proc]: Name: EndpointForwarding

I (30287) [alexa_smart_home]: Namespace: Alexa, Name: EventProcessed
I (30287) [alexa_smart_home]: Event with token 86bf8ba5-d410-53e8-988d-e9dcab3ab738 has been processed by server
I (30307) [alexa_discovery]: Setting alexa discovery digest in NVS
[28 seconds]: [http_transport]: Event data: {"events":[{"header":{"name":"SynchronizeState","messageId":"b49387e0-c38a-db9c-cf4f-81abafdb7079"},"payload":{"speaker":{"volume":80},"alerts":{"allAlerts":[]}}}]}
I (30327) [http_transport]: Sending to topic: $aws/alexa/ais/v1/482fa2bd-116d-4780-bb37-a75415875881/event
[29 seconds]: [http_transport]: Free Memory Internal: 77784, External: 2568452
[directive_proc]: Json data: {"directives":[{"header":{"name":"SetAttentionState","messageId":"0b97d377-d7bb-41f8-b189-943aa18a237f"},"payload":{"state":"IDLE"}}]}
I (31087) [directive_proc]: Name: SetAttentionState

I (31087) [va_nvs_utils]: No value set for: ready_prompt
I (31097) [prompt]: Processing request ===> Play
I (31097) [focus_manager]: Acquire channel: Prompt Prompt

[dialog]: Stream finished
[dialog]: Event STREAM_CLOSED unsupported in IDLE state
[prompt]: On Focus: Foreground
I (31117) [sys_playback]: Acquire
I (31117) [focus_manager]: Now active channel: Prompt Prompt

I (31127) [prompt]: Processing request ===> Foreground
[audio_codec]: Starting audio codec mp3_decoder
I (31137) [audio_codec]: Codec mp3_decoder Event Started
[basic_player]: Codec event: Codec Started
I (31147) MP3Decoder: Found ID3 tag, skipping 45 bytes
I (31157) MP3Decoder: MP3 sync operation...
I (31157) MP3Decoder: MP3 sync success
I (31177) [audio_codec]: Codec mp3_decoder Event Set Freq
[basic_player]: Codec event: Codec Set Frequency
I (31177) [basic_player]: Set Freq event: 24000, 1, 16
I (31177) [prompt]: Processing request ===> Pipeline Started
I (31767) MP3Decoder: Codec reader finished
I (32077) MP3Decoder: Done with last chunk
I (32077) MP3Decoder: Closed
I (32077) [audio_codec]: Codec mp3_decoder Event Stopped
[basic_player]: Codec event: Codec Stopped
I (32937) [prompt]: Processing request ===> Pipeline Stopped
I (32937) [focus_manager]: Release channel: Prompt Prompt
I (32937) [focus_manager]: Releasing Prompt
[... cut ...]
[35 seconds]: [http_transport]: Event data: {"events":[{"header":{"name":"SynchronizeState","messageId":"4b56dc44-ad8d-e298-97c0-e142a6fbf264"},"payload":{"speaker":{"volume":80},"alerts":{"allAlerts":[]}}}]}
I (37237) [http_transport]: Sending to topic: $aws/alexa/ais/v1/482fa2bd-116d-4780-bb37-a75415875881/event
[35 seconds]: [http_transport]: Free Memory Internal: 76132, External: 2568452
[directive_proc]: Json data: {"directives":[{"header":{"name":"SetAttentionState","messageId":"100bb1b2-ef90-4503-b68c-d1845e59a03e"},"payload":{"state":"IDLE"}}]}
I (37657) [directive_proc]: Name: SetAttentionState

[dialog]: Stream finished
[dialog]: Event STREAM_CLOSED unsupported in IDLE state

chiragatal, 2021-02-22:
Hi @actiontang , This is most likely a server side issue and we have already reported that to Amazon.

The device does not show up in the Alexa app because of the 'disconnect' that the device gets from the server after connecting successfully and going into the idle state.

actiontang, 2021-02-28:
Hi @chiragatal, 
Still can't see "Light" smart home skill when I tested again!
 
Can you check again ?

Thanks

omerio, 2021-11-12:
Any fix for this, I'm getting the same issue, the Light doesn't show up and is not recognised by Alexa

chiragatal, 2021-11-12:
With the latest release, the device shows up as something like 'Chirag's ESP Alexa' instead of 'Light'. There would be 2 devices in the Amazon Alexa app with the same name, one would be the AIA device and the other would be the light device.

Also, make sure you have enabled smart home in the AIA example.