espressif/esp-va-sdk #131: Problems with Google Assistant

KalashQK, 2021-03-24:
Hello!
I plan to use the ESP32-LyraT board as a smart speaker. After following all the instructions for installing the ESP-IDF, the SDK with Amazon Alexa was first installed, everything works well. When installing SDK to support Google Voice Assistant, there were many questions and problems, but in the end I managed to install the software on the board. Unfortunately, the device does not respond to speech commands in any way, whether it be with a wake-word or when the REC button is pressed. The device is not found in ESP-Alexa and Google Home applications. I tried to find information on forums and websites, but I could not find anything useful.  

Everything was done exactly according to the instructions (https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md).
>>cd esp-va-sdk/examples/google_voice_assistant
>>export ESPPORT=COM4
>>export IDF_PATH=C:/esp/msys32/home/Okalash/esp/esp-idf
>>export AUDIO_BOARD_PATH=C:/esp/msys32/home/Okalash/esp/esp-idf/esp-va-sdk/board_support_pkgs/lyrat/audio_board/audio_board_lyrat/
>>make -j 8 flash monitor [ALEXA_BT=1]
After Google Project customization:
>>python -m pip install --upgrade google-auth-oauthlib[tool]
>>google-oauthlib-tool --scope https://www.googleapis.com/auth/assistant-sdk-prototype \
--save --headless --client-secrets /C:/path/to/client_secret_XXXX.json
4/XXXX
>> nvs-set refreshToken string <1//XXXX>
>> nvs-set clientId string <XXXX.apps.googleusercontent.com>
>> nvs-set clientSecret string <XXXX>
>>wifi-set XXXX XXXX

After completing all the points of the instructions (https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md), the RST button was pressed on the board, then when the REC button was pressed, the commands were voiced. Monitor shows next information:

# make monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/esp/msys32/home/Okalash/esp/esp-idf/requirements.txt            are satisfied.
MONITOR
--- idf_monitor on COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6376
load:0x40078000,len:9484
load:0x40080400,len:6996
entry 0x40080744
I (28) boot: ESP-IDF v3.2.2 2nd stage bootloader
I (28) boot: compile time 13:49:05
I (28) boot: Enabling RNG early entropy source...
I (33) qio_mode: Enabling default flash chip QIO
I (38) boot: SPI Speed      : 80MHz
I (42) boot: SPI Mode       : QIO
I (46) boot: SPI Flash Size : 4MB
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00390000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0xa0b2c (658
220) map
I (268) esp_image: segment 1: paddr=0x000b0b54 vaddr=0x3ffbdb60 size=0x04af0 ( 1
9184) load
I (274) esp_image: segment 2: paddr=0x000b564c vaddr=0x40080000 size=0x00400 (
1024) load
0x40080000: _WindowOverflow4 at C:/esp/msys32/home/Okalash/esp/esp-idf/component
s/freertos/xtensa_vectors.S:1779

I (275) esp_image: segment 3: paddr=0x000b5a54 vaddr=0x40080400 size=0x0a5bc ( 4
2428) load
I (297) esp_image: segment 4: paddr=0x000c0018 vaddr=0x400d0018 size=0xe9eac (95
8124) map
0x400d0018: _flash_cache_start at ??:?

I (547) esp_image: segment 5: paddr=0x001a9ecc vaddr=0x4008a9bc size=0x0d674 ( 5
4900) load
[... 150 lines cut ...]
I (40984) [http_stream]: [sid: 1] Response content type is application/grpc
I (40984) [grpc_http_transport]: [sid 1] stream close
[dialog]: Stream finished
[va_dsp]: Sending stop command
I (40994) [focus_manager]: Release channel: Dialog SpeechRecognizer
I (40994) [app_va_cb]: Dialog state is: 8
[va_dsp]: Event 3 unsupported in STOPPED state
I (41004) [focus_manager]: Releasing SpeechRecognizer
[dialog]: Entering VA_IDLE
I (41014) [sys_playback]: Release
I (41024) [sys_playback]: Got event stopped
[41 seconds]: [http_stream]: [stream_delete: 1] Internal: 67468, External: 28824
76, min ever internal: 57776, largest free block: 60136
[va_button]: button pressed: 36
I (74074) [va_button]: tap to talk pressed
[va_dsp]: Sending start for tap to talk command
[speech_recognizer]: New recognize request: WW length 0
[dialog]: Dialog new
I (74074) [focus_manager]: Acquire channel: Dialog SpeechRecognizer

The device does not respond to voice commands in any way, whether it be with a wake-word or when the REC button is pressed. The device is not detected in the applications of ESP-Alexa and Google Home applications.

I try to apply to Software Bug Report, but an error occurs when submitting. 
There is already one appeal on the Github with this problem (https://github.com/espressif/esp-va-sdk/issues/113), as far back as last year, but there is still no answer.
Please, help with the solution, this project is very important for me, it depends on this part to receive a grant for further development. The fact is that I have more experience with the hardware and using ESP32 with sensors, but not working with voice recognition and Google. Thank you!

chiragatal, 2021-03-26:
@KalashQK , We tried this from scratch multiple times yesterday by just following the documentation, and it worked fine.

I don't see any explicit problems with the log. It could maybe be a problem with the configuration of the project (maybe 'Enabling the API' step?). Could you also make sure the required permissions are set in the google account?

It could also be that something is missing from our documentation or Google's documentation.

KalashQK, 2021-03-30:
@chiragatal Thanks a lot for your answer!
I tried to check everything again, but, unfortunately, to no avail.
Yes, I think the most likely reason is in the project settings. The fact is that at some steps in the instructions everything is described well, in detail and understandable, but on others, during the first experiment without experience, I can miss something. Therefore, I want to attach screenshots of the instructions and the corresponding results, so that, for example, you can compare my results and yours and find the differences, if you don't mind.

1. Instructions
[image]

Results 
I create a new project
[image]

2. Instructions
[image]

Results 
[image]

[image]

I tried to fill out information in all sections.

The top indicates that Credentials needs to be added to use the API. When I click on create Credentials, a choice appears between three options. I created a service account.

[image]

[image]

I set up OAuth consent screen

[image]

There are sections for filling out the API key and the Create OAuth client ID, but the instructions do not say about this, I do not know how to fill them out.

I will not provide further screenshots for now, because with permission to send data and register the device, because, as far as I understand, there were no problems with this item, I received the secret ID, token and other data.

I read different forums, someone wrote that there was a problem with the service account, after filling it out, everything worked.

Most likely, the main problem was precisely in that section of the project settings. I will be very grateful if you look at the screenshots, compare with the projection settings that you have, and write what the problem is with the settings of my project. Perhaps this will somehow still be able to help with improving the instructions on the SDK page, so that it is a little clearer and easier for users to use the SDK, because sometimes they wrote on the forums about small misunderstandings. The problems are connected not so much with the SDK itself as with the instructions from Google, but if everything works fine for you, I would like to understand what exactly needs to be done for this.

Many thanks!

chiragatal, 2021-03-31:
@KalashQK , I can see 1-2 things different than what we had done.
- On the OAuth consent screen, I just filled whatever was 'required'. I think it was only the email ID.
- Then added the user via 'Add Users'.
- I did not use 'Create Credentials'.
- Then once the device model has been registered and the client_secret.json has been downloaded, to get the refresh_token, etc, I followed the first 3 steps from here https://developers.google.com/assistant/sdk/guides/service/python/embed/config-dev-project-and-account

KalashQK, 2021-04-01:
@chiragatal I have previously added test users through the "Add Users", as well as registered the device, and, according to the instructions, set up the token parameters and others.
I tried several more times to do everything again, from the very beginning, clearly according to the instructions. Previously, I took several erroneous unnecessary steps, now without them, but, unfortunately, nothing helps anyway.
I tried to create 3 projects with different languages, on different Google accounts. But now there is also an error with the token.

To make everything clearer, I fully recorded all the steps in the document. Please check the correctness.

I apologize for asking too much, but for more than 2 months I have been trying to make the program work, I have looked for articles, I have already asked familiar programmers, nothing helps. I have been using Espressif boards for several years now, everything has always worked well, but this time I have no other options for what else to do.

[Description and results.pdf](https://github.com/espressif/esp-va-sdk/files/6243525/Description.and.results.pdf)

chiragatal, 2021-04-01:
@KalashQK , a few things:
- The project setup part on the console looks exactly like what I have done. You have not mentioned the 'activity permissions' part in the google account, but I hope all the permissions are given.
- When running the CLI tool, after you get the URL, I did not end the command. I opened the URL, signed in, got the code and pasted it without running the command again. I think you did this in 2 steps. (Not sure if this will affect anything)
- For the nvs-set command, you should not be including the <> in the command. example: nvs-set key string value

KalashQK, 2021-04-02:
@chiragatal Great thank you!
As far as I remember, earlier when using nvs-set of commands without <>, there was an error, but, probably, it was connected with something else, and not with <>. I checked all the permissions in my Google account once again, checked everything thoroughly, finally now it works! Unfortunately, the device is not displayed in applications, but this is not the most important thing. And it only works when a button is pressed. Do you have any plans to add wakeword support in the future, like in Alexa's SDK? Thanks a lot again!

closed 2021-08-19 by KalashQK