espressif/esp-va-sdk #132: Failed idf.py build for the branch feature/aia-beta

jassi00713, 2021-03-25:
Trying to implement Alexa on ESP32 Vaquita DSPG and using branch feature/aia-beta

Followed instructions on the below link:-
https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta#2-development-setup

Build is failing. 
Error caused due to -> multiple definition of `set_xpd_sar'
Kindly help with the soltion to this.

Below are logs on running idf.py build.
LOGS-

idf.py build
Checking Python dependencies...
Python requirements from /home/javivek/espaia2/esp-idf/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory /home/javivek/espaia2/esp-va-sdk/examples/amazon_aia/build
Executing "ninja all"...
[211/1168] cd /home/javivek/espaia2/esp-va-sdk/examples/amazon_aia/build/esp-...****************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
fctry_aia,data,nvs,0x10000,24K,
fctry,data,nvs,0x16000,24K,
otadata,data,ota,0x1c000,8K,
ota_0,app,ota_0,0x20000,4032K,
ota_1,app,ota_1,0x410000,4032K,
*******************************************************************************
[304/1168] Performing configure step for 'bootloader'
-- Component directory /home/javivek/espaia2/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/javivek/espaia2/esp-idf/components/cmock does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/javivek/espaia2/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- Project version: v4.0.2-442-g41efdb0b3
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/javivek/espaia2/esp-va-sdk/examples/amazon_aia/sdkconfig
-- Adding linker script /home/javivek/espaia2/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/javivek/espaia2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/javivek/espaia2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/javivek/espaia2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/javivek/espaia2/esp-idf/components/bootloader/subproject/main/esp32.bootloader.ld
-- Adding linker script /home/javivek/espaia2/esp-idf/components/bootloader/subproject/main/esp32.bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_rom esptool_py log main micro-ecc partition_table soc spi_flash xtensa
-- Component paths: /home/javivek/espaia2/esp-idf/components/bootloader /home/javivek/espaia2/esp-idf/components/bootloader_support /home/javivek/espaia2/esp-idf/components/efuse /home/javivek/espaia2/esp-idf/components/esp32 /home/javivek/espaia2/esp-idf/components/esp_common /home/javivek/espaia2/esp-idf/components/esp_rom /home/javivek/espaia2/esp-idf/components/esptool_py /home/javivek/espaia2/esp-idf/components/log /home/javivek/espaia2/esp-idf/components/bootloader/subproject/main /home/javivek/espaia2/esp-idf/components/bootloader/subproject/components/micro-ecc /home/javivek/espaia2/esp-idf/components/partition_table /home/javivek/espaia2/esp-idf/components/soc /home/javivek/espaia2/esp-idf/components/spi_flash /home/javivek/espaia2/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/javivek/espaia2/esp-va-sdk/examples/amazon_aia/build/bootloader
[331/1168] Performing build step for 'bootloader'
[1/2] Linking C executable bootloader.elf
[2/2] Generating binary image from built executable
esptool.py v2.9-dev
Generated /home/javivek/espaia2/esp-va-sdk/examples/amazon_aia/build/bootloader/bootloader.bin
[1067/1168] Building C object esp-idf/streams/CMakeFiles/__idf_streams.dir/i2s_stream/i2s_stream.c.obj
/home/javivek/espaia2/esp-va-sdk/components/streams/i2s_stream/i2s_stream.c: In function 'parse_i2s_config':
/home/javivek/espaia2/esp-va-sdk/components/streams/i2s_stream/i2s_stream.c:45:15: warning: unused variable 'ret' [-Wunused-variable]
     esp_err_t ret = ESP_OK;
               ^~~
[1078/1168] Building C object esp-idf/httpc/CMakeFiles/__idf_httpc.dir/httpc.c.obj
/home/javivek/espaia2/esp-va-sdk/components/httpc/httpc.c: In function 'http_connection_new':
/home/javivek/espaia2/esp-va-sdk/components/httpc/httpc.c:156:5: warning: 'esp_tls_conn_new' is deprecated [-Wdeprecated-declarations]
     tls = esp_tls_conn_new(&url[u->field_data[UF_HOST].off], u->field_data[UF_HOST].len,
     ^~~
In file included from /home/javivek/espaia2/esp-va-sdk/components/httpc/httpc.h:18,
                 from /home/javivek/espaia2/esp-va-sdk/components/httpc/httpc.c:20:
/home/javivek/espaia2/esp-idf/components/esp-tls/esp_tls.h:271:12: note: declared here
 esp_tls_t *esp_tls_conn_new(const char *hostname, int hostlen, int port, const esp_tls_cfg_t *cfg)  __attribute__ ((deprecated));
            ^~~~~~~~~~~~~~~~
/home/javivek/espaia2/esp-va-sdk/components/httpc/httpc.c: In function 'http_connection_new_async':
[... 56 lines cut ...]
/home/javivek/espaia2/esp-va-sdk/components/audio_hal/esp_codec/es8311/components/codec_es8311/media_hal_codec_init.c:47:39: warning: assignment to 'esp_err_t (*)(uint8_t *)' {aka 'int (*)(unsigned char *)'} from incompatible pointer type 'int (*)(int *)' [-Wincompatible-pointer-types]
     media_hal->audio_codec_get_volume = es8311_get_volume;
                                       ^
/home/javivek/espaia2/esp-va-sdk/components/audio_hal/esp_codec/es8311/components/codec_es8311/media_hal_codec_init.c:48:37: warning: assignment to 'esp_err_t (*)(_Bool)' {aka 'int (*)(_Bool)'} from incompatible pointer type 'int (*)(int)' [-Wincompatible-pointer-types]
     media_hal->audio_codec_set_mute = es8311_set_mute;
                                     ^
[1146/1168] Building C object esp-idf/codec_es8311/CMakeFiles/__idf_codec_es8311.dir/es8311.c.obj
/home/javivek/espaia2/esp-va-sdk/components/audio_hal/esp_codec/es8311/components/codec_es8311/es8311.c: In function 'es8311_init':
/home/javivek/espaia2/esp-va-sdk/components/audio_hal/esp_codec/es8311/components/codec_es8311/es8311.c:321:9: warning: unused variable 'port_num' [-Wunused-variable]
     int port_num = media_hal_conf->port_num;
         ^~~~~~~~
/home/javivek/espaia2/esp-va-sdk/components/audio_hal/esp_codec/es8311/components/codec_es8311/es8311.c:320:28: warning: unused variable 'es8388_dac_output' [-Wunused-variable]
     media_hal_dac_output_t es8388_dac_output = media_hal_conf->dac_output;
                            ^~~~~~~~~~~~~~~~~
/home/javivek/espaia2/esp-va-sdk/components/audio_hal/esp_codec/es8311/components/codec_es8311/es8311.c:319:27: warning: unused variable 'es8311_adc_input' [-Wunused-variable]
     media_hal_adc_input_t es8311_adc_input = media_hal_conf->adc_input;
                           ^~~~~~~~~~~~~~~~
[1167/1168] Linking CXX executable aia.elf
FAILED: aia.elf 
: && /home/javivek/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/aia.elf.dir/project_elf_src.c.obj  -o aia.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/bt/libbt.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp [... cut ...]
/home/javivek/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(wifi_init.c.obj): in function `set_xpd_sar':
/home/javivek/espaia2/esp-idf/components/esp_wifi/src/wifi_init.c:219: multiple definition of `set_xpd_sar'; esp-idf/common/libcommon.a(app_wifi.c.obj):/home/javivek/espaia2/esp-va-sdk/examples/common/app_wifi.c:185: first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

chiragatal, 2021-03-25:
I think this is because of a recent commit in esp-idf. Could you comment the `set_xpd_sar()` API in esp-va-sdk/examples/common/app_wifi.c:184 and try again?

jassi00713, 2021-03-25:
Thanks a lot. It worked.

If i could get one more help. Firmware flashes fine. I have generated the new ESP Alexa Credentials. 
mfg_config.csv file i have received in the mail has sme extra feilds as compared to the one already present in SDK.

mfg_config.csv in SDK :-
key,type,encoding,value
device,namespace,,
client_cert,file,binary,/path/to/esp-alexa/examples/amazon_aia/main/certs/device.crt
client_key,file,binary,/path/to/esp-alexa/examples/amazon_aia/main/certs/device.key
server_cert,file,binary,/path/to/esp-alexa/examples/amazon_aia/main/certs/server.crt

mfg_config.csv Received in mail:-
key,type,encoding,value
device,namespace,,
device_id,file,binary,/path/to/device.info
account_id,file,binary,/path/to/account.info
mqtt_host,file,binary,/path/to/endpoint.info
client_cert,file,binary,/path/to/device.cert
client_key,file,binary,/path/to/device.key
server_cert,file,binary,/path/to/server.cert

Thanks.

chiragatal, 2021-03-25:
All the fields in the downloaded one are required.
The one already present in the SDK only has some of those as the SDK provides other options (like menuconfig) to set the other fields.

jassi00713, 2021-03-28:
cool :+1: