espressif/esp-va-sdk #133: Getting : Error(-12) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com:8883

celron, 2021-03-25:
I'm using the vaquita-DSPG development board and using Espressif's AWS Account.
The board seems to be working but cannot connect to (I assume) Amazon's MQTT.  I've checked the certificates using openssl and they seem fine and am stuck as to what could be the problem.  I include the most relevant parts of the log
~~~
[[0;32mI (1981) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE^[[0m^M
^[[0;32mI (1981) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE^[[0m^M
I (2001) wifi:wifi firmware version: 82f9f1b^M
I (2001) wifi:config NVS flash: enabled^M
I (2001) wifi:config nano formating: disabled^M
I (2001) wifi:Init data frame dynamic rx buffer num: 32^M
I (2011) wifi:Init management frame dynamic rx buffer num: 32^M
I (2011) wifi:Init management short buffer num: 32^M
I (2021) wifi:Init static tx buffer num: 32^M
I (2021) wifi:Init tx cache buffer num: 32^M
I (2031) wifi:Init static rx buffer size: 1600^M
I (2031) wifi:Init static rx buffer num: 10^M
I (2031) wifi:Init dynamic rx buffer num: 32^M
^[[0;32mI (2041) wifi_init: rx ba win: 16^[[0m^M
^[[0;32mI (2041) wifi_init: tcpip mbox: 32^[[0m^M
^[[0;32mI (2051) wifi_init: udp mbox: 6^[[0m^M
^[[0;32mI (2051) wifi_init: tcp mbox: 12^[[0m^M
^[[0;32mI (2051) wifi_init: tcp tx win: 14360^[[0m^M
^[[0;32mI (2061) wifi_init: tcp rx win: 14360^[[0m^M
^[[0;32mI (2061) wifi_init: tcp mss: 1440^[[0m^M
^[[0;32mI (2071) wifi_init: WiFi/LWIP prefer SPIRAM^[[0m^M
^[[0;32mI (2071) [app_smart_home]: Registering command: smart-home^[[0m^M
[app_smart_home]: *************** Light's Power turned ON ***************^M
[app_smart_home]: *************** Light's Brightness changed to 100 ***************^M
^[[0;32mI (2091) wifi_prov_scheme_ble: BT memory released^[[0m^M
^[[0;32mI (2111) [app_main]: Error reading mqtt_host from nvs of size 0. Error code: 4354^[[0m^M
[app_main]: Using the non-NVS aws_endpoint: aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com^M
^[[0;32mI (2121) [app_main]: Error reading account_id from nvs of size 0. Error code: 4354^[[0m^M
[app_main]: Using the non-NVS aws_account_id: 282183071650^M
^[[0;32mI (2131) [app_main]: Error reading device_id from nvs of size 0. Error code: 4354^[[0m^M
[app_main]: Using the non-NVS client_id: 240ac4e6c764^M
^[[0;32mI (2151) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3^[[0m^M
^[[0;32mI (2151) I2S: PLL_D2: Req RATE: 48000, real rate: 48076.000, BITS: 16, CLKM: 13, BCK: 8, MCLK: 12292917.167, SCLK: 1538432.000000, diva: 64, divb: 1^[[0m^M
^[[0;32mI (2181) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ^[[0m^M
^[[0;32mI (2191) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ^[[0m^M
^[[0;32mI (2201) [diag_cli]: Registering command: up-time^[[0m^M
^[[0;32mI (2201) [scli]: Initialising UART on port 0^[[0m^M
^[[0;32mI (2201) uart: queue free spaces: 8^[[0m^M
>> ^[[0;32mI (2211) [diag_cli]: Registering command: mem-dump^[[0m^M
^[[0;32mI (2211) [diag_cli]: Registering command: task-dump^[[0m^M
^[[0;32mI (2221) [diag_cli]: Registering command: cpu-dump^[[0m^M
^[[0;32mI (2231) [diag_cli]: Registering command: sock-dump^[[0m^M
^[[0;32mI (2231) [diag_cli]: Registering command: heap-trace^[[0m^M
^[[0;32mI (2241) [va_diag_cli]: Registering command: nvs-get^[[0m^M
^[[0;32mI (2241) [va_diag_cli]: Registering command: nvs-set^[[0m^M
^[[0;32mI (2251) [va_diag_cli]: Registering command: nvs-erase^[[0m^M
^[[0;32mI (2251) [va_diag_cli]: Registering command: reboot^[[0m^M
^[[0;32mI (2261) [va_diag_cli]: Registering command: crash^[[0m^M
^[[0;32mI (2271) [wifi_cli]: Registering command: wifi-set^[[0m^M
^[[0;32mI (2271) [app-auth]: Registering command: sign-in^[[0m^M
^[[0;32mI (2281) [app-auth]: Registering command: sign-out^[[0m^M
^[[0;32mI (2281) [speaker]: Registering command: vol-up^[[0m^M
^[[0;32mI (2291) [speaker]: Registering command: vol-down^[[0m^M
^[[0;32mI (2291) [speaker]: Registering command: vol-set^[[0m^M
^M^[[0;32mI (2301) [va_nvs_utils]: No value set for: speaker_vol^[[0m^M
^[[0;32mI (2301) [speaker]: Using default volume^[[0m^M
[speaker]: Volume changed to 80^M
^[[0;32mI (2311) [va_nvs_utils]: No value set for: sor_tone^[[0m^M
^[[0;32mI (2321) [va_nvs_utils]: No value set for: eor_tone^[[0m^M
^[[0;32mI (2331) [audio_codec]: Starting mp3_decoder codec^[[0m^M
^[[0;32mI (2331) [va_nvs_utils]: No value set for: wifi_reset_bit^[[0m^M
^[[0;32mI (2331) [app_main]: Already provisioned, starting station^[[0m^M
^[[0;32mI (2341) wifi_prov_scheme_ble: BTDM memory released^[[0m^M
^[[0;32mI (2351) phy_init: phy_version 4660,0162888,Dec 23 2020^[[0m^M
I (2441) wifi:mode : sta (24:0a:c4:e6:c7:64)^M
I (2451) wifi:Set ps type: 1^M
^M
[... 49 lines cut ...]
^[[0;32mI (11031) [http_transport]: String: $aws/alexa/ais/v1/240ac4e6c764/capabilities/publish^[[0m^M
^[[0;32mI (11041) [http_transport]: String: $aws/alexa/ais/v1/240ac4e6c764/capabilities/acknowledge^[[0m^M
^[[0;32mI (11051) [http_transport]: String: $aws/alexa/ais/v1/240ac4e6c764/event^[[0m^M
^[[0;32mI (11051) [http_transport]: String: $aws/alexa/ais/v1/240ac4e6c764/directive^[[0m^M
^[[0;32mI (11061) [http_transport]: String: $aws/alexa/ais/v1/240ac4e6c764/microphone^[[0m^M
^[[0;32mI (11071) [http_transport]: String: $aws/alexa/ais/v1/240ac4e6c764/speaker^[[0m^M
^[[0;32mI (11081) [http_transport]: Connecting to AWS...^[[0m^M
^[[0;32mI (11161) [va_nvs_utils]: No value set for: d2_fw_debug^[[0m^M
^[[0;32mI (11161) [va_nvs_utils]: No value set for: d4p_fw_debug^[[0m^M
^[[0;32mI (11161) [va_nvs_utils]: No value set for: d2_audio_debug^[[0m^M
^[[0;32mI (11171) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ^[[0m^M
^[[0;32mI (11181) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ^[[0m^M
^[[0;32mI (11191) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ^[[0m^M
^[[0;32mI (11201) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ^[[0m^M
^[[0;31mE (12861) [http_transport]: Error(-12) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com:8883^[[0m^M
^[[0;32mI (13911) [dspg_utility]: Successful switch to TDM0 clock^[[0m^M
^[[0;32mI (13921) [dspg_utility]: 5a 02 18 06 00 00 00 00 00 00 ^[[0m^M
^[[0;31mE (15571) [http_transport]: Error(-12) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com:8883^[[0m^M
^[[0;32mI (16541) [dspg_driver]: Calling start capture^[[0m^M
^[[0;31mE (18301) [http_transport]: Error(-12) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com:8883^[[0m^M
^[[0;31mE (21091) [http_transport]: Error(-12) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com:8883^[[0m^M
^[[0;31mE (23851) [http_transport]: Error(-12) connecting to aw2zep8xn2fby-ats.iot.us-east-1.amazonaws.com:8883^[[0m^M
~~~

I'm at a loss as to what I can do as it seems that it cannot connect to Amazon, but I can't tell if it's a server issue or a client issue.  I've looked at `esp-aws-iot/port/network_mbedtls_wrapper.c` where I think the error occurs but that module doesn't provide much insight.  Any help would be appreciated.

chiragatal, 2021-03-26:
@celron , I could not see any explicit errors from the log. I suspect it could be because of some problem with the certificates.
Could you try with the mfg.bin that you might have generated with the certificates and other details from the email?

celron, 2021-03-26:
@chiragatal I did have problems using the mfg_config.csv as sent by espressif, and had to change it to 
~~~
key,type,encoding,value
device,namespace,,
client_cert,file,binary,/Users/telhar/smart_plug/esp-va-sdk/examples/amazon_aia/certs/device.cert
client_key,file,binary,/Users/telhar/smart_plug/esp-va-sdk/examples/amazon_aia/certs/device.key
server_cert,file,binary,/Users/telhar/smart_plug/esp-va-sdk/examples/amazon_aia/certs/server.cert
~~~
The original as provided in the zip file would generate certificate errors.

How would I test with just the mfg.bin?  I'm not familiar with that process

chiragatal, 2021-03-26:
@celron , what problems were you facing? It should have worked.

mfg.bin is the binary file that is generated and you flash on the device for aia_certificates.

celron, 2021-03-26:
Actually I got it working.
When I used the mfg_config.csv file that was provided to generate the mfg.bin file the board started working
There was a README.md file in the cert directory that said
~~~
Copy the Device Credentials to this folder.

Follow these steps:
- Copy the credentials in this folder and rename them according to mfg_config.csv.
    - device.crt: This is the device certificate.
    - device.key: This is the device private key.
    - server.crt: This is the AWS root CA certificate.
**- (Optional) You can modify mfg_config.csv add other details also in the NVS such as AWS account ID, AWS Endpoint URL, Device ID. You can take a look at app_main.c for reference.**
- Modify the file mfg_config.csv and add the relevant paths like: /Users/.../esp-alexa/examples/amazon_aia/main/certs/device.crt
- Run the command: python $IDF_PATH/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py mfg_config.csv mfg.bin 0x6000
- Now flash this mfg.bin on the device using the command: python $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 --port $ESPPORT write_flash 0x10000 mfg.bin
- Now flash the board normally as you would using: make flash OR make spiffs_flash
~~~
The Optional entries appear to be required.  I had tried to use my own credentials with the README before trying the files from Espressif and had copied the entries from the zip file.

I would assume that they (device_id, account_id, mqtt_host) are also required if I used my own credentials?

chiragatal, 2021-03-26:
If you are using your own credentials, then you need the account_id and the mqtt_host (endpoint_id) of your AWS account.

celron, 2021-03-26:
Thank you! then the README.md should probably be changed in **aia_beta** branch **/amazon_aia/main/certs**

closed 2021-03-26 by celron

chiragatal, 2021-03-26:
Yes, I think that readme should be deleted and instead just the top level readme needs to be followed. Thanks for pointing out!