espressif/esp-va-sdk #136: Newlib version

lorenzomerici, 2021-05-05:
Hi there, 
the Alexa built-in certification process performs a scan of the binaries looking for security issues. 
There are a lot of issues found in newlib < v4.0.0, but the firmware on AIA-beta has the version of newlib 2.2, so the firmware cannot be certified. 

Do you know how we can solve this? Do you plan to update it or it's possible to do it easily? 

Thanks, Lorenzo

avsheth, 2021-05-06:
Hi Lorenzo
Almost all of the newlib vulnerabilities do not apply to ESP32. And can be marked as NA in the certification process.

closed 2021-05-06 by lorenzomerici

avsheth, 2021-05-06:
If you are looking to go for certification of your product, please do get in touch with sales@espressif.com.