espressif/esp-va-sdk #140: No rule to make target building error

NikhilPatale, 2021-07-26:
Hello,
While trying to replicate the building and compile steps, I am facing with the following error log 

nikhil@nikhil-VirtualBox:~$ cd esp-va-sdk/examples/amazon_alexa
nikhil@nikhil-VirtualBox:~/esp-va-sdk/examples/amazon_alexa$ export ESPPORT=/dev/ttyUSB0
nikhil@nikhil-VirtualBox:~/esp-va-sdk/examples/amazon_alexa$ export IDF_PATH=/path/to/esp-idf
nikhil@nikhil-VirtualBox:~/esp-va-sdk/examples/amazon_alexa$ export AUDIO_BOARD_PATH=/path/to/esp-va-sdk/board_support_pkgs/lyrat/audio_board/audio_board_lyratd_msc/
nikhil@nikhil-VirtualBox:~/esp-va-sdk/examples/amazon_alexa$ make -j 8 flash monitor [ALEXA_BT=1]
Makefile:42: /path/to/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target '/path/to/esp-idf/make/project.mk'.  Stop.
 
Any kind of guidance or help regarding this issue would be highly appreciated

chiragatal, 2021-07-27:
In the export commands, replace the '/path/to/' with the actual path to the repository. example: `export IDF_PATH=~/esp-idf/`

NikhilPatale, 2021-07-27:
Hello Chirag, 
Many thanks for guiding through the above error, however I am facing a new issue after that. Attaching the Screenshot below. 
[image]
 
I have also Installed the extensa toolchain with the following version.
[image]

chiragatal, 2021-07-28:
1. What is your IDF branch?
2. Are the submodules updated in IDF?
3. `[ALEXA_BT=1]` is an optional argument and is supposed to be used without `[]`. Remove that for now.

closed 2021-08-11 by NikhilPatale