espressif/esp-va-sdk #141: Undefined Reference error

shaikhzahid06, 2021-08-10:
Getting the undefined reference to '<api_name>' but as mentioned in the instruction after adding **target_link_libraries(${project_elf} "-Wl,--start-group")**  in the idf project.cmake file 
getting the below error while building:
**cannot specify link libraries for target -Wl,--start-group which is not built by this project**.

Kindly assist me on this as i am really stuck.

vikramdattu, 2021-08-11:
Hi @shaikhzahid06 current release of project do not support `cmake` or latest 4.x IDF releases.

You need to build the project with IDF version `release/v3.2`, gcc 5.0 based toolchain and  make commands.

Please follow the procedure here: https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md and here: https://docs.espressif.com/projects/esp-idf/en/v3.2.2/get-started/index.html

We are working on the release which will work on IDF version release/v4.2.

shaikhzahid06, 2021-08-11:
> Hi @shaikhzahid06 current release of project do not support `cmake` or latest 4.x IDF releases.
>
> You need to build the project with IDF version `release/v3.2`, gcc 5.0 based toolchain and make [...]

Hi @vikramdattu thanks for the reply, I have tried building the project with IDF version release/v3.2 but still facing the same issue after **LD build alexa.elf**

LD build/alexa.elf
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x50): undefined reference to `esp_avrc_tg_init'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x54): undefined reference to `esp_avrc_tg_register_callback'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x58): undefined reference to `esp_avrc_psth_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x5c): undefined reference to `esp_avrc_tg_set_psth_cmd_filter'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x60): undefined reference to `esp_avrc_rn_evt_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x64): undefined reference to `esp_avrc_tg_set_rn_evt_cap'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x141): undefined reference to `esp_avrc_tg_init'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x175): undefined reference to `esp_avrc_tg_register_callback'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x18f): undefined reference to `esp_avrc_psth_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x19c): undefined reference to `esp_avrc_psth_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x1aa): undefined reference to `esp_avrc_psth_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x1b7): undefined reference to `esp_avrc_psth_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x1c4): undefined reference to `esp_avrc_psth_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.text.bluetooth_internal_start+0x1d2): more undefined references to `esp_avrc_psth_bit_mask_operation' follow
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x1eb): undefined reference to `esp_avrc_tg_set_psth_cmd_filter'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x207): undefined reference to `esp_avrc_rn_evt_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x210): undefined reference to `esp_avrc_tg_set_rn_evt_cap'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o):(.literal.bt_app_av_notify_vol_change+0xc): undefined reference to `esp_avrc_tg_send_rn_rsp'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o):(.literal.bt_app_rc_ct_cb+0x8): undefined reference to `esp_avrc_ct_send_get_rn_capabilities_cmd'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o):(.text.bt_av_new_track+0x13): undefined reference to `esp_avrc_rn_evt_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o):(.text.bt_av_playback_changed+0xc): undefined reference to `esp_avrc_rn_evt_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o):(.text.bt_av_play_pos_changed+0xc): undefined reference to `esp_avrc_rn_evt_bit_mask_operation'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o): In function `bt_app_av_notify_vol_change':
(.text.bt_app_av_notify_vol_change+0x3e): undefined reference to `esp_avrc_tg_send_rn_rsp'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o): In function `bt_app_rc_ct_cb':
(.text.bt_app_rc_ct_cb+0x2a): undefined reference to `esp_avrc_ct_send_get_rn_capabilities_cmd'
/home/luminox/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o): In function `bt_app_rc_tg_cb':
(.text.bt_app_rc_tg_cb+0x58): undefined reference to `esp_avrc_tg_send_rn_rsp'
collect2: error: ld returned 1 exit status
make: *** [/home/luminox/esp/esp-idf/make/project.mk:461: /home/luminox/esp/esp-va-sdk/examples/amazon_alexa/build/alexa.elf] Error 1

shaikhzahid06, 2021-08-11:
yes i missed to apply the patches.
The sdk has flashed successfully but now the espalexa app is not able to discover the device(ESP32 Lyratd msc).
While flashing  I have passed the Alexa_BT=1 argument but the device is not discoverable on bluetooth

vikramdattu, 2021-08-12:
Go ahead and close the issue. Device not shown discussion can be continued on other thread.

closed 2021-08-12 by shaikhzahid06