espressif/esp-va-sdk #142: Deivce Not Showing Up on ESPAlexa App

shaikhzahid06, 2021-08-10:
After successfully flashing the sdk when trying to view the device on the mobile app it doesn't show.
Any help would be highly appreciated

NikhilPatale, 2021-08-11:
Just adding to the above issue, since me and Zahid are working together, 
The following is the console output after flashing.

So has the build been flashed or is it exceeding the partition memory hence unable to work? and if yes, then is there any way wherein we can solve it? we did try specifying 4M in the partition table but it did not work. 

[image]

vikramdattu, 2021-08-12:
@NikhilPatale @shaikhzahid06 that means the image size doesn't fit the flash.

Please comment out following line from app_main.c and give it a try: `alexa_tone_enable_larger_tones();`

shaikhzahid06, 2021-08-12:
@vikramdattu Thanks for your prompt reply, commenting out the above line worked for us and the device is visible but now we are facing new errors:
E (1416531) esp-tls: couldn't get hostname for :avs-alexa-eu.amazon.com:
E (1416531) esp-tls: Failed to open new connection
E (1416531) [sh2lib]: [sh2-connect] esp-tls connection failed
E (1416531) [http_transport]: Failed to connect
E (1152941) CODEC_ES8388: es8388_read_reg error
E (1152941) CODEC_ES8388: es8388_write_reg error
E (1152941) CODEC_ES8388: es8388_read_reg error
E (1152941) CODEC_ES8388: es8388_write_reg error
E (1152951) CODEC_ES8388: es8388_write_reg error
E (1152951) CODEC_ES8388: es8388_write_reg error
E (1152961) CODEC_ES8388: es8388_write_reg error
E (1152961) CODEC_ES8388: es8388_write_reg error

vikramdattu, 2021-08-12:
```
E (1416531) esp-tls: couldn't get hostname for :avs-alexa-eu.amazon.com:
E (1416531) esp-tls: Failed to open new connection
E (1416531) [sh2lib]: [sh2-connect] esp-tls connection failed
```
This is due to internet issues. And should go away when internet is back.

```
E (1152941) CODEC_ES8388: es8388_read_reg error
E (1152941) CODEC_ES8388: es8388_write_reg error
E (1152941) CODEC_ES8388: es8388_read_reg error
E (1152941) CODEC_ES8388: es8388_write_reg error
E (1152951) CODEC_ES8388: es8388_write_reg error
E (1152951) CODEC_ES8388: es8388_write_reg error
E (1152961) CODEC_ES8388: es8388_write_reg error
E (1152961) CODEC_ES8388: es8388_write_reg error
```
The board is you are using is `lyrat` right?

If it is lyrat-mini, you will need to add support for that.

shaikhzahid06, 2021-08-12:
The above issue was due to selecting an inappropriate board.

shaikhzahid06, 2021-08-12:
The device provisioning has been done successfully after selecting the correct board; now while interacting with the device it doesn't seem to respond and following errors are being printed on the monitor.

E (4048) i2c: i2c command link malloc error
E (4058) i2c: i2c command link malloc error
E (4058) i2c: i2c command link malloc error
E (4068) i2c: i2c command link malloc error
E (4068) LED: is31fl3216_write_reg error
E (4068) i2c: i2c command link malloc error
E (4078) i2c: i2c command link malloc error
E (4078) i2c: i2c command link malloc error
E (4088) i2c: i2c command link malloc error
E (4088) i2c: i2c command link malloc error
E (4098) LED: is31fl3216_write_reg error
E (4098) i2c: i2c command link malloc error
E (4108) i2c: i2c command link malloc error
E (4108) i2c: i2c command link malloc error
E (4118) i2c: i2c command link malloc error
E (4118) i2c: i2c command link malloc error
E (4128) LED: is31fl3216_write_reg error

[bluetooth-internal]: Error reading paired device list from NVS

vikramdattu, 2021-08-12:
Internal memory is probably full.
If you do not need Bluetooth functionality, you may consider not enabling it.

Command line to use is: (not ALEXA_BT parameter)
```
make -j8
```
Give it a try with build generated with this.

You may club it as:

```
make -j8 flash monitor
```

shaikhzahid06, 2021-08-12:
Tried making the build without ALEXA_BT but still getting the same error. The board we are using is lyratd msc_A V2.2

shaikhzahid06, 2021-08-13:
The issue has been resolved and the device is working as expected.. Thanks for your help!!

closed 2021-08-13 by shaikhzahid06