espressif/esp-va-sdk #148: Linking CXX executable aia.elf
FAILED: aia.elf

Nalininalu, 2021-08-29:
Hai .. I'm getting this error .. I tried the solution mentioned on the docs (changes in project.cmake file) bt it didn't worked .. I'm using Ubuntu 20.4 version and esp-idf( used all version in 4)

chiragatal, 2021-08-30:
Hi @Nalininalu , which is the IDF branch you are using? Also can you share the full logs?

Nalininalu, 2021-08-30:
Thanks for ur rply..Im using exactly these steps from this doc ( bt in esp-idf v4.3)
https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta#12-the-software

And this is my log

1360/1361] Linking CXX executable aia.elf
FAILED: aia.elf
: && /home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address   CMakeFiles/aia.elf.dir/project_elf_src_esp32.c.obj  -o aia.elf  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/perfmon/libperfmon.a  esp-idf/esp32/libesp32.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/json_parser/libjson_parser.a  esp-idf/httpc/libhttpc.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/esp_httpd_ota/libesp_httpd_ota.a  esp-idf/multipart_parser/libmultipart_parser.a  esp-idf/qrcode/libqrcode.a  esp-idf/sh2lib/libsh2lib.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_cloud/libapp_cloud.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/esp-aws-iot/libesp-aws-iot.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/vvdn/esp/esp-va-sdk/examples/amazon_aia/build/aia.map  -fno-rtti  -fno-lto  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/es [... cut ...]
/home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/vvdn/esp/esp-va-sdk/components/basic_player/basic_player.c:577: undefined reference to `http_playback_stream_set_stack_size'
/home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/vvdn/esp/esp-va-sdk/components/basic_player/basic_player.c:580: undefined reference to `audio_stream_init'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Nalininalu, 2021-08-30:
> Hi @Nalininalu , which is the IDF branch you are using? Also can you share the full logs?

Thanks for ur rply..Im using exactly these steps from this doc ( bt in esp-idf v4.3)
https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta#12-the-software

And this is my log

1360/1361] Linking CXX executable aia.elf
FAILED: aia.elf
: && /home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address   CMakeFiles/aia.elf.dir/project_elf_src_esp32.c.obj  -o aia.elf  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/perfmon/libperfmon.a  esp-idf/esp32/libesp32.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/json_parser/libjson_parser.a  esp-idf/httpc/libhttpc.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/esp_httpd_ota/libesp_httpd_ota.a  esp-idf/multipart_parser/libmultipart_parser.a  esp-idf/qrcode/libqrcode.a  esp-idf/sh2lib/libsh2lib.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_cloud/libapp_cloud.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/esp-aws-iot/libesp-aws-iot.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/vvdn/esp/esp-va-sdk/examples/amazon_aia/build/aia.map  -fno-rtti  -fno-lto  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/c [... cut ...]
/home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/vvdn/esp/esp-va-sdk/components/basic_player/basic_player.c:577: undefined reference to `http_playback_stream_set_stack_size'
/home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/vvdn/esp/esp-va-sdk/components/basic_player/basic_player.c:580: undefined reference to `audio_stream_init'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

chiragatal, 2021-08-30:
For the first error regarding `set_xpd_sar`, can you try this: https://github.com/espressif/esp-va-sdk/issues/132#issuecomment-806482366

For the undefined reference errors, for the change in esp-idf/tools/cmake/project.cmake instead of line 394 can you try it on line 445 instead and see if it works?
```
..snip..
    add_custom_target(_project_elf_src DEPENDS "${project_elf_src}")
    add_executable(${project_elf} "${project_elf_src}")
    add_dependencies(${project_elf} _project_elf_src)
+  target_link_libraries(${project_elf} "-Wl,--start-group")

    if(test_components)
        target_link_libraries(${project_elf} "-Wl,--whole-archive")
..snip..
```

Nalininalu, 2021-08-30:
> For the first error regarding `set_xpd_sar`, can you try this: https://github.com/espressif/esp-va-sdk/issues/132#issuecomment-806482366
>
> For the undefined reference errors, for the change in [...]

i replaced "void set_xpd_sar(bool en)" with "void set_xpd_sar()" in app.wifi.c(line 184) and added that line as u mention.. bt im still getting this error.. Thanks in advance...

#warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
[1360/1361] Linking CXX executable aia.elf
FAILED: aia.elf
: && /home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address   CMakeFiles/aia.elf.dir/project_elf_src_esp32.c.obj  -o aia.elf  -Wl,--start-group  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/perfmon/libperfmon.a  esp-idf/esp32/libesp32.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/json_parser/libjson_parser.a  esp-idf/httpc/libhttpc.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/esp_httpd_ota/libesp_httpd_ota.a  esp-idf/multipart_parser/libmultipart_parser.a  esp-idf/qrcode/libqrcode.a  esp-idf/sh2lib/libsh2lib.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_cloud/libapp_cloud.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/esp-aws-iot/libesp-aws-iot.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/vvdn/esp/esp-va-sdk/examples/amazon_aia/build/aia.map  -fno-rtti  -fno-lto  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a   [... cut ...]
/home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a(dialog.o): in function `dialog_sm_thread':
/Users/ganesh/work/gitlab/esp-alexa/components/voice_assistant/common/src/dialog.c:324: undefined reference to `xTaskNotify'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

chiragatal, 2021-08-30:
For `set_xpd_sar`, You would need to comment it in app_main.c.

For the remaining undefined reference, the change in project.cmake should have fixed it. Let me check.

Nalininalu, 2021-08-30:
> For `set_xpd_sar`, You would need to comment it in app_main.c.
> 
> For the remaining undefined reference, the change in project.cmake should have fixed it. Let me check.
> 
> 

In which line I have to add that (esp-va-sdk/examples/amazon_aia/main/app_main.c)?

chiragatal, 2021-08-30:
Sorry I meant to comment the blank API in esp-va-sdk/examples/common/app_wifi.c: https://github.com/espressif/esp-va-sdk/blob/feature/aia-beta/examples/common/app_wifi.c#L184

Nalininalu, 2021-08-30:
> Sorry I meant to comment the blank API in esp-va-sdk/examples/common/app_wifi.c: https://github.com/espressif/esp-va-sdk/blob/feature/aia-beta/examples/common/app_wifi.c#L184

No.. I didn't worked .. I replaced void set_xpd_sar(bool en) with this blank.. am I doing that wrong ?

chiragatal, 2021-08-30:
In app_wifi.c: https://github.com/espressif/esp-va-sdk/blob/feature/aia-beta/examples/common/app_wifi.c#L184

Replace:
```
void set_xpd_sar(bool en)
{

}
```
With:
```
// void set_xpd_sar(bool en)
// {
// 
// }
```

Nalininalu, 2021-08-30:
> In app_wifi.c: https://github.com/espressif/esp-va-sdk/blob/feature/aia-beta/examples/common/app_wifi.c#L184
> 
> Replace:
> ```
> void set_xpd_sar(bool en)
> {
> 
> }
> ```
> With:
> ```
> // void set_xpd_sar(bool en)
> // {
> // 
> // }
> ```
> 

Still Im getting those error..

[5/6] Linking CXX executable aia.elf
FAILED: aia.elf
: && /home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address   CMakeFiles/aia.elf.dir/project_elf_src_esp32.c.obj  -o aia.elf  -Wl,--start-group  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/perfmon/libperfmon.a  esp-idf/esp32/libesp32.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/json_parser/libjson_parser.a  esp-idf/httpc/libhttpc.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/esp_httpd_ota/libesp_httpd_ota.a  esp-idf/multipart_parser/libmultipart_parser.a  esp-idf/qrcode/libqrcode.a  esp-idf/sh2lib/libsh2lib.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_cloud/libapp_cloud.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/esp-aws-iot/libesp-aws-iot.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/vvdn/esp/esp-va-sdk/examples/amazon_aia/build/aia.map  -fno-rtti  -fno-lto  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/l [... cut ...]
/home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a(dialog.o): in function `dialog_sm_thread':
/Users/ganesh/work/gitlab/esp-alexa/components/voice_assistant/common/src/dialog.c:324: undefined reference to `xTaskNotify'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

NikhilPatale, 2021-08-30:
Hello everybody,

I had started working with this same exact build an hour ago on my lyraT and I had encountered the same exact error which @Nalininalu had encountered, pasting the reference screenshot below. 
[image]

This seems to be the same exact issue encountered by her. I went across your steps @chiragatal and that indeed solved all the build issues. The only change that I had to make in your recommended steps was to paste the line no 394 code at line no 398 and not at 445. So I think she would be able to do the same and get away with her errors.
So what I can do for you @Nalininalu is I can paste reference screenshots so that you would be able to replicate the same. Primarily you have 2 make changes at 2 places in the codes as mentioned by @chiragatal 

Number 1:
Navigate to /esp-va-sdk/examples/common/app_wifi.c
Open it and navigate to line no 184 and comment the following method using //
[image]

Number 2:
Navigate to /esp-idf/tools/cmake/project.cmake
Make the following change and simply add this one line target_link_libraries(${project_elf} "-Wl,--start-group") at line no 398 instead of 445 (As mentioned by Chirag) and no 394 (As mentioned by the SDK) 
[image]

Save your code after making every change and then perform an idf.py flash monitor. It should now be able to build and flash.

Thank you @chiragatal for your prompt replies and guidance. I hope @Nalininalu this makes Chirag's instructions crystal clear. 

Thank you
Nikhil

Nalininalu, 2021-08-31:
> Hello everybody,
> 
> I had started working with this same exact build an hour ago on my lyraT and I had encountered the same exact error which @Nalininalu had encountered, pasting the reference screenshot below. 
> [image]
> 
> This seems to be the same exact issue encountered by her. I went across your steps @chiragatal and that indeed solved all the build issues. The only change that I had to make in your recommended steps was to paste the line no 394 code at line no 398 and not at 445. So I think she would be able to do the same and get away with her errors.
> So what I can do for you @Nalininalu is I can paste reference screenshots so that you would be able to replicate the same. Primarily you have to make changes at 2 places in the codes as mentioned by @chiragatal 
> 
> Number 1:
> Navigate to /esp-va-sdk/examples/common/app_wifi.c
> Open it and navigate to line no 184 and comment the following method using //
> [image]
> 
> Number 2:
> Navigate to /esp-idf/tools/cmake/project.cmake
> Make the following change and simply add this one line target_link_libraries(${project_elf} "-Wl,--start-group") at line no 398 instead of 445 (As mentioned by Chirag) and no 394 (As mentioned by the SDK) 
> [image]
> 
> Save your code after making every change and then perform an idf.py flash monitor. It should now be able to build and flash.
> 
> Thank you @chiragatal for your prompt replies and guidance. I hope @Nalininalu this makes Chirag's instructions crystal clear. 
> 
> Thank you
> Nikhil
>  

Hai @NikhilPatale .. I exactly did what u and @chiragatal said.. bt im again getting this error..[image][image]

Nalininalu, 2021-08-31:
@NikhilPatale and FYI I'm using vaquita board

NikhilPatale, 2021-08-31:
> Hai @NikhilPatale .. I exactly did what u and @chiragatal said.. bt im again getting this error..[image][image]

In the second screenshot, remove line number 444 and paste the same on line number 398, Try to compile it. If the error persists, send the screenshot of the code where you made this change along with the error log. 

Thank you,
Nikhil

Nalininalu, 2021-08-31:
> In the second screenshot, remove line number 444 and paste the same on line number 398, Try to compile it. If the error persists, send the screenshot of the code where you made this change along [...]

Hai @Where should I paste that line.. in b/w this ?[image]

Nalininalu, 2021-08-31:
@NikhilPatale 

This is my log even after changing those 2 things in project.cmake and app_wifi.c

[1346/1348] Linking CXX executable aia.elf
FAILED: aia.elf
: && /home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address   CMakeFiles/aia.elf.dir/project_elf_src_esp32.c.obj  -o aia.elf  -Wl,--start-group  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/perfmon/libperfmon.a  esp-idf/esp32/libesp32.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/json_parser/libjson_parser.a  esp-idf/httpc/libhttpc.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/esp_httpd_ota/libesp_httpd_ota.a  esp-idf/multipart_parser/libmultipart_parser.a  esp-idf/qrcode/libqrcode.a  esp-idf/sh2lib/libsh2lib.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_cloud/libapp_cloud.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/esp-aws-iot/libesp-aws-iot.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/vvdn/esp/esp-va-sdk/examples/amazon_aia/build/aia.map  -fno-rtti  -fno-lto  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/led_pattern/libled_pattern.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/common/libcommon.a  esp-idf/alexa_equalizer/libalexa_equalizer.a  esp-idf/app_smart_home/libapp_smart_home.a  esp-idf/custom_player/libcustom_player.a  esp-idf/audio_board_dspg_avs_kit_m/libaudio_board_dspg_avs_kit_m.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  esp-idf/main/libmain.a  /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /home/vvdn/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /home/vvdn/esp/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /home/vvdn/esp/esp-va-sdk/components/audio_hal/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/li [... cut ...]
/home/vvdn/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/vvdn/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a(dialog.o): in function `dialog_sm_thread':
/Users/ganesh/work/gitlab/esp-alexa/components/voice_assistant/common/src/dialog.c:324: undefined reference to `xTaskNotify'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Nalininalu, 2021-08-31:
> @NikhilPatale
>
> This is my log even after changing those 2 things in project.cmake and app_wifi.c
>
>
>
> [1346/1348] Linking CXX executable aia.elf
> FAILED: aia.elf
> : && [...]

Hai @NikhilPatale ... I tried those changes which is given by u and him in esp-idf v4 instead of esp-idf v4.3.. it's now working fine.. I have a doubt now .. shall I proceed?

NikhilPatale, 2021-08-31:
Hello,

Yes, you are supposed to use esp-idf version 4.0 strictly and not version 4.3 because the build is version-specific. Stick to esp-idf version 4.0 and you should be good to go ahead. 

Thanks,
Nikhil

Nalininalu, 2021-08-31:
> Hello,
> 
> Yes, you are supposed to use esp-idf version 4.0 strictly and not version 4.3 because the build is version-specific. Stick to esp-idf version 4.0 and you should be good to go ahead. 
> 
> Thanks,
> Nikhil 

While using flash monitor cmd .. at the end I'm getting this one.. I assume it's becz I didn't proceed the aws certificate steps..am I r8?[image]

NikhilPatale, 2021-08-31:
> While using flash monitor cmd .. at the end I'm getting this one.. I assume it's becz I didn't proceed the aws certificate steps..am I r8?[image]

Yes, that's right. You'll require that, if you don't have that then you can use espressif keys for the time being by providing your mail id by navigating to the hyperlink mentioned in the guide.

Nalininalu, 2021-08-31:
Hai...@NikhilPatale thank you so much for helping me..

closed 2021-08-31 by avsheth

Nalininalu, 2021-09-02:
@NikhilPatale  Hai Nikhil.. I have another doubt