espressif/esp-va-sdk #149: Extending the board support for Lyratd_Msc v 2.2?

NikhilPatale, 2021-08-30:
Hi Everyone,

Is there any possibility from the developers that this SDK could be build to support Lyratd_Msc v 2.2? I have a fair bit of idea that the previous version of the same SDK was built for it. We were able to flash that successfully into Lyratd_Msc, However we were not successful in establishing a connection to aws iot server. Alexa was working as intended but establishing a connection to the server was not successful, hence we stumbled upon the latest version of this SDK but unfortunately board support isn't provided. So what we did was, we took the board packages of Lyratd_Msc from the previous build and made some minor changes in the make files for it. We tried using those board packages for the latest version of SDK but weren't successful. Hence we have 2 fundamental problems which could be solved either ways.

1. If some guidance is provided in order to make a successful connection to aws iot using the previous SDK (which requires esp-idf version 3.2) we should be able to work with Lyratd_msc.
2. If for some unfortunate reason that doesn't work could you be providing Board support for Lyratd_Msc in the current version of SDK? 
 
Thanks all in advance,
Nikhil

chiragatal, 2021-09-01:
Since lyratd_msc is not certifiable, I would recommend to in general use the vaquita or the lyratd_dspg board.

If you really want to use lyratd_msc, I would suggest to port the board files from the v3.2 version (current master branch) onto the new major release we have planned soon.

If you are facing problems in porting the board files, we can help you with that.

NikhilPatale, 2021-09-01:
Hello,

We have currently built a Esp32 wrover-16MB custom design with Lyratd_msc as reference for audio architecture.

And we aimed to get it built-in Alexa certification

Can you clarify what do you mean by not certifiable? Because if it's the limitation in the esp flash size that doesn't allow it to be certified then we have already solved that issue. Our main aim is to get our custom board certified by Amazon at the end and Amazon doesn't seem to have any conditions to certification related to audio codecs as per our knowledge.

Nonetheless, it would be great if you could help us port the board files from previous version.

We have tried using the board package of LyraTd_Msc from the master branch to the new feature branch, but are not successful.

Could you give us a detailed instruction set on what changes are supposed to be made in the board cmake files or pin headers or Cmakelists ? That would really ease our work of porting.

chiragatal, 2021-09-01:
The wake word engine used on lyratd_msc by default is Espressif's, and it does not pass Amazon certification yet.

The WWE on lyratd_dspg or vaquita here is Espressif's, but we have a separate certifiable WWE for these boards which we can give you access to (please contact sales@espressif.com).

Regarding porting lyratd_msc from v3.2 version to the upcoming v4.2 version, the next release will have more documentation which will help you port the board files.

NikhilPatale, 2021-09-02:
Hello @chiragatal ,

We have currently developed a custom PCB using lyratd_msc as a reference in regards to the audio DSP zl38063. So we can comfortably say that both the boards are quite similar.

It would be great if you could give us a confirmation on whether we would be able to proceed with this design to get built-in Alexa certification from Amazon with the future master branch v4.2 if the board porting is successful in the next release as you have mentioned earlier. 

You also mentioned that Espressif provides production-ready and certifiable firmware for WWE.
Can that be made compatible for lyratd_msc DSP zl38063 in the future master release?

If there is no way forward with this design of ours (LyraTd_Msc) then we will instantly scrap this design and pivot to using vaquita_dspg as our audio architecture reference since Vaquita_dspg is already a certified amazon device and Espressif already has a production-ready Amazon certifiable WWE firmware for that.

NikhilPatale, 2021-09-06:
Hi @chiragatal 

Could you please address our above queries so that we are able to formulate a better solution for our product as soon as possible? 
Any assistance in this regard would be highly appreciated from your end. 

Thank you,
Nikhil

kedars, 2021-09-08:
@NikhilPatale From acoustic certification perspective, it would not be possible to be done on the 'msc' variant even after the board porting is done with IDFv4.2.  As Chirag advised above, for production scenarios, I'd advise to contact sales@espressif.com

NikhilPatale, 2021-09-09:
Hi @kedars 

As adviced from your end, we have drafted and sent a mail to the sales team at Espressif. In terms of additional assistance and end to end integration of avs into our product, could we reach out to your team for the same? 

Regards,
Nikhil

Serdcoff, 2023-04-08:
> @NikhilPatale From acoustic certification perspective, it would not be possible to be done on the 'msc' variant even after the board porting is done with IDFv4.2 https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta#0-important-note. As Chirag advised above, for production scenarios, I'd advise to contact [sales@espressif.com](mailto:sales@espressif.com)

Hi there. I've got LyraTD MSc board about 1 month and no chance to find get the answer from sales@espressif.com
Culd you please hep me with porting version of VA-SDK from 3.2 to new one? Aslo old certificates in 3.2 cannot work properly. As promised Espressif team must be upload a new one. But onthing happens. I just turned off the mbedit time section and it works.
For now I need your help in porting ver 3.2 to 4.2. Also very hard to get tha answer from Espressif team as right.
Thank you.