espressif/esp-va-sdk #150: ESP Alexa app is not recognizing my Esp32

Nalininalu, 2021-09-02:
I previously tried to configure WiFi and Bluetooth .. That worked pretty fine.. my esp32 (vaquita) board connected to my phone WiFi and Bluetooth.. when I try to run esp-va-sdk code ( by downloading espressif's certificate) the board is getting connected with my phone's hotspot but in ESP ALEXA APP.. I can't view any new device ...

chiragatal, 2021-09-02:
Are you talking about provisioning or managing? Can you share screenshots of the app?
Can you also share the device logs?

Nalininalu, 2021-09-02:
Hai.. thank you for your reply
 
I'm in provisioning stage ...[image]

[image]

And here is my log

[2/4] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Running esptool.py in directory /home/vvdn/esp/esp-va-sdk/examples/amazon_aia/build
Executing "/home/vvdn/.espressif/python_env/idf4.0_py3.8_env/bin/python /home/vvdn/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash @flash_project_args"...
esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB 0x8000 partition_table/partition-table.bin 0x1c000 ota_data_initial.bin 0x1000 bootloader/bootloader.bin 0x20000 aia.bin
esptool.py v2.9-dev
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:e5:af:e4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 149...
Wrote 3072 bytes (149 compressed) at 0x00008000 in 0.0 seconds (effective 3419.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0001c000 in 0.0 seconds (effective 14989.5 kbit/s)...
Hash of data verified.
Compressed 24016 bytes to 14904...
Wrote 24016 bytes (14904 compressed) at 0x00001000 in 0.3 seconds (effective 562.8 kbit/s)...
Hash of data verified.
Compressed 3337344 bytes to 2437722...
Wrote 3337344 bytes (2437722 compressed) at 0x00020000 in 56.5 seconds (effective 472.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /home/vvdn/esp/esp-va-sdk/examples/amazon_aia
Executing "/home/vvdn/.espressif/python_env/idf4.0_py3.8_env/bin/python /home/vvdn/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 /home/vvdn/esp/esp-va-sdk/examples/amazon_aia/build/aia.elf -m '/home/vvdn/.espressif/python_env/idf4.0_py3.8_env/bin/python' '/home/vvdn/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:12
load:0x3fff0028,len:6996
load:0x40078000,len:13520
ho 0 tail 12 room 4
load:0x40080400,len:3380
entry 0x40080650
I (68) boot: Chip Revision: 3
I (37) boot: ESP-IDF v4.0.3-144-gd64bf0dfbe-dirty 2nd stage bootloader
I (37) boot: compile time 11:23:48
I (38) boot: Enabling RNG early entropy source...
I (43) boot: SPI Speed      : 80MHz
I (47) boot: SPI Mode       : DIO
I (51) boot: SPI Flash Size : 8MB
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
[... 123 lines cut ...]
I (4465) wifi:state: assoc -> run (10)
I (4495) wifi:connected with Nalini, aid = 3, channel 12, BW20, bssid = 1e:76:04:30:28:88
I (4495) wifi:security: WPA2-PSK, phy: bgn, rssi: -30
I (4495) wifi:pm start, type: 1

I (4515) wifi:AP's beacon interval = 102400 us, DTIM period = 2
[app_wifi]: Connected with IP Address: 192.168.43.106
I (5475) tcpip_adapter: sta ip: 192.168.43.106, mask: 255.255.255.0, gw: 192.168.43.1
I (5475) [va_nvs_utils]: No value set for: friendlyname
I (5485) SSDP: Network Interface List (1):
[alexa]: Waiting for time to be updated
I (5485) SSDP: 1. wifi  : 192.168.43.106
I (5495) SSDP:

I (5495) LSSDP: create SSDP socket 58
[alexa]: Done getting current time: 1630573680
I (10245) [tls_certification]: Done setting global CA store
I (10245) [va_nvs_utils]: No value set for: clientId
I (10245) [va_nvs_utils]: No value set for: refreshToken
I (10245) [alexa]: Checking for companion app based authentication...

[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens

Nalininalu, 2021-09-02:
@chiragatal  FYI I previously configured simple WiFi program.. is that messing up with this..?? Bcz it get automatically connected with the phone ... But in doc they mentioned first we have to connect the board via Bluetooth to ESP app and then we should connect with WiFi

chiragatal, 2021-09-02:
Can you erase_flash, and the flash AIA and try again?

Nalininalu, 2021-09-02:
Issue solved thank you

closed 2021-09-02 by Nalininalu