espressif/esp-va-sdk #152: NVS/AIA issues

danielhendrickson, 2021-09-13:
Stuck on flashing firmware step. I got through the linking issues and commented out the 2nd definition of set_xpd_sar in the app_wifi file. Now when I run idf.py flash monitor I am getting some NVS issues as well as and AIA issue:

E (2245) [app_main]: NVS open failed with error 4354
E (2245) [app_main]: NVS open failed with error 4354
E (2255) [app_main]: NVS open failed with error 4354
E (2255) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS aws_endpoint: XXX
E (2265) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS aws_account_id: XXX
E (2275) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS client_id: 240ac4e6ce80
E (2295) [app_main]: AIA device certificates not found. Please flash the certificates at the correct location and reboot the device to proceed.

Any ideas on what the problem is? I am using macOS 11.4 for reference

Full logs are below:

dhendrickson@syn0098 amazon_aia % idf.py flash monitor
Checking Python dependencies...
Python requirements from /Users/dhendrickson/esp-idf/requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory /Users/dhendrickson/esp-va-sdk/examples/amazon_aia/build
Executing "ninja all"...
[1/4] cd /Users/dhendrickson/esp-va-sdk/examples/amazon...******************************************************
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
fctry_aia,data,nvs,0x10000,24K,
fctry,data,nvs,0x16000,24K,
otadata,data,ota,0x1c000,8K,
ota_0,app,ota_0,0x20000,4032K,
ota_1,app,ota_1,0x410000,4032K,
*******************************************************************************
[2/4] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Running esptool.py in directory /Users/dhendrickson/esp-va-sdk/examples/amazon_aia/build
Executing "/Users/dhendrickson/.espressif/python_env/idf4.0_py3.9_env/bin/python /Users/dhendrickson/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/cu.usbserial-110 -b 460800 --before default_reset --after hard_reset write_flash @flash_project_args"...
esptool.py -p /dev/cu.usbserial-110 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB 0x8000 partition_table/partition-table.bin 0x1c000 ota_data_initial.bin 0x1000 bootloader/bootloader.bin 0x20000 aia.bin
esptool.py v2.9-dev
Serial port /dev/cu.usbserial-110
Connecting........_
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:e6:ce:80
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 149...
Wrote 3072 bytes (149 compressed) at 0x00008000 in 0.1 seconds (effective 423.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0001c000 in 0.1 seconds (effective 1192.5 kbit/s)...
Hash of data verified.
Compressed 24576 bytes to 15090...
Wrote 24576 bytes (15090 compressed) at 0x00001000 in 0.4 seconds (effective 502.9 kbit/s)...
Hash of data verified.
Compressed 3417664 bytes to 2472117...
Wrote 3417664 bytes (2472117 compressed) at 0x00020000 in 63.6 seconds (effective 430.2 kbit/s)...
Hash of data verified.

[... 101 lines cut ...]
I (2175) wifi:Init static rx buffer size: 1600
I (2175) wifi:Init static rx buffer num: 10
I (2185) wifi:Init dynamic rx buffer num: 32
I (2185) wifi_init: rx ba win: 16
I (2185) wifi_init: tcpip mbox: 32
I (2195) wifi_init: udp mbox: 6
I (2195) wifi_init: tcp mbox: 12
I (2205) wifi_init: tcp tx win: 14360
I (2205) wifi_init: tcp rx win: 14360
I (2205) wifi_init: tcp mss: 1440
I (2215) wifi_init: WiFi/LWIP prefer SPIRAM
I (2215) [app_smart_home]: Registering command: smart-home
[app_smart_home]: *************** Light's Power turned ON ***************
[app_smart_home]: *************** Light's Brightness changed to 100 ***************
I (2235) wifi_prov_scheme_ble: BT memory released
E (2245) [app_main]: NVS open failed with error 4354
E (2245) [app_main]: NVS open failed with error 4354
E (2255) [app_main]: NVS open failed with error 4354
E (2255) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS aws_endpoint: XXX
E (2265) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS aws_account_id: XXX
E (2275) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS client_id: 240ac4e6ce80
E (2295) [app_main]: AIA device certificates not found. Please flash the certificates at the correct location and reboot the device to proceed.

danielhendrickson, 2021-09-13:
Nevermind I think I have it figured out

closed 2021-09-13 by danielhendrickson

PaulAnurag, 2022-07-20:
@danielhendrickson  What was the issue? do you remember?

PaulAnurag, 2022-07-21:
@chiragatal Got it. Thanks. After flashing the certificates, I encounter a new issue in the firmware. Can you please help me out in this regard.
[image]
.

chiragatal, 2022-07-21:
@PaulAnurag Could you please open a new issue for this, and share the complete device logs from the start?