espressif/esp-va-sdk #155: Does this method only support Lyra board?

Nalininalu, 2021-09-23:
https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md

chiragatal, 2021-09-23:
Which board do you have and which example are you trying to build?

Nalininalu, 2021-09-27:
I'm trying to do Alexa example ( with music and playback support ) by using vaquita board)

chiragatal, 2021-09-29:
In the release yesterday, we have added the supported boards along with all the supported voice assistants on the master branch. The branches of other repos and the build commands have also been updated. Check the new READMEs for more info.

Nalininalu, 2021-09-29:
Could u share the link plz?

chiragatal, 2021-09-29:
https://github.com/espressif/esp-va-sdk/tree/master
Check the readme here.

Nalininalu, 2021-09-29:
Hi Thank you so much..
    And another doubt.. by using espressif's aws account, music and playback features can be enabled right? Or should I have to whitelist an account..

Nalininalu, 2021-09-29:
Hi thank you so much.. and another one this IDF only support Hungama music ?

Nalininalu, 2021-09-30:
Hi.. this SDK  does not support Spotify ?

chiragatal, 2021-09-30:
The default AVS product here does not support the music services which require approval. You can create your own AVS product, enable the music services that you want and use that AVS product instead of the default one. (You will also have to modify the phone app to use your AVS product and build it again.)

Apart from Spotify, all the music services supported by Alexa should work.

Nalininalu, 2021-09-30:
Hi @chiragatal ...thanks.. actually I flashed amazon_aia and it supported music except Spotify, pandora and some others