espressif/esp-va-sdk #161: Voice Assistance - Failed to verify peer certificate!

yusufk, 2021-10-19:
Getting the following error on console:

`E (202374) esp-tls: Failed to open new connection
E (202384) [sh2lib]: [sh2-connect] esp-tls connection failed
E (202384) [grpc_http_transport]: Failed to connect
I (204384) [grpc_http_transport]: Connecting to server: https://embeddedassistant.googleapis.com
E (204724) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (204724) esp-tls-mbedtls: Failed to verify peer certificate!
I (204724) esp-tls-mbedtls: verification info:   ! The certificate is not correctly signed by the trusted CA
`

s95i, 2021-12-14:
I have the same problem. Did you find the solution?

yusufk, 2021-12-14:
Unfortunately not.

avsheth, 2021-12-15:
Hi @yusufk could you please tell which app are you trying and provide complete logs for the same?
It appears that we might need to update some certificate in our library.

yusufk, 2021-12-15:
Hi,

I believe it was the alexa app I was trying at the time.

Regards
Yusuf

s95i, 2021-12-15:
Hi @avsheth, I had the same issue with the GVA app.

avsheth, 2021-12-16:
I believe this is GVA app and not the Alexa app. We will check and get back.

s95i, 2022-01-11:
Hi, @avsheth, Is there any way to add a new certificate to solve this issue?

andrewwinter, 2022-01-18:
I'm still having this issue does anyone have a solution? I believe that the problem is that the CA certifications are hardcoded into the lib files (found at esp-va-sdk/voice_assistant/*.a), I found an older google CA certification in them, however i believe that they're expired. Could you please recompile these libraries with the new CA certificates? @avsheth 
Thanks

avsheth, 2022-02-10:
Hi @andrewwinter 
Sorry for the delay. We will provide updated library by next week.

s95i, 2022-03-01:
Hi @avsheth ,
When can we expect the updated library?
Thanks