espressif/esp-va-sdk #162: Compile and flash for esp32 LyraT board

Marcoz08, 2021-12-07:
Hi, I am a beginner in IDF esp programming, I have been using the Lyrat board with the ADF. Now I want to use the va-sdk, specifically test amazon alexa.
Could someone help me with compiling the code? and help me with the Development Setup?

Best regards.

I use the windows terminal.

avsheth, 2021-12-14:
Hi @Marcoz08 
Which IDF branch are you on?

chaitanya-smartrotamac, 2021-12-14:
I got a similar issue when I used IDF branch 4.3.3 so I reverted to 4.2 and build everything now working try it.

Marcoz08, 2021-12-14:
> Hi @Marcoz08 Which IDF branch are you on?

ESP-IDF 4.3

Marcoz08, 2021-12-14:
> I got a similar issue when I used IDF branch 4.3.3 so I reverted to 4.2 and build everything now working try it.

Hi, i'm using IDF 4.3, how can i go back to version 4.2?

chaitanya-smartrotamac, 2021-12-15:
How you installed IDF 4.3 - Windows or linux?
If Windows I used "https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#get-started-windows-tools-installer" - IDF Tools installer ,
So I uninstalled it and while installing again it asked 4.3 or 4.2.2 , This time I selected 4.2.2. Done.
If linux no idea, I think you need to checkout 4.2 version branch from git. 
Check getting repositories section in this link "https://github.com/espressif/esp-va-sdk"
Try and let me know.

Marcoz08, 2021-12-15:
> ¿Cómo instaló IDF 4.3 - Windows o Linux? Si utilicé Windows " https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#get-started-windows-tools-installer " - Herramientas IDF instalador, así que lo desinstalé y mientras lo instalaba nuevamente me preguntó 4.3 o 4.2.2, esta vez seleccioné 4.2.2. Hecho. Si linux no tiene idea, creo que debe verificar la rama de la versión 4.2 de git. Consulte la sección de obtención de repositorios en este enlace " https://github.com/espressif/esp-va-sdk " Intente y avíseme.

Thanks, i'm going to try.

Marcoz08, 2021-12-18:
The problem was solved using IDF 4.2, now the code compiles and flashes successfully.  If someone has the same problem and is using IDF branch 4.3, uninstall it and install version 4.2, to do so use the idf tool.

closed 2021-12-18 by Marcoz08