espressif/esp-va-sdk #163: Music and radio playback are not supported on this device

chaitanya-smartrotamac, 2021-12-14:
Good evening,
I am using ESP32 LYRAT 4.3 board
ESP IDF = 4.2.2
ESPVASDK 
After all the efforts my made I am able to flash amazon SDK.
But now when I ask " Alexa can you play some music"
it says "Music and Radio playback are not supported on this device" why?
Am i missing something??

chaitanya-smartrotamac, 2021-12-14:
I used "https://espressif.github.io/esp-va-sdk/#/" to generate certificated and generate mfg partition and uploaded it to Lyrat
still no improvement " Music and Radio playback are not supported on this device"

avsheth, 2021-12-15:
Music playback is subjected to whitelisting of product ID used on the device. For which, an organisation/user needs to register their products on developer.amazon.com and fill out all product specific details, following which Amazon would enable music playback on devices using that product ID.
Product ID in this repo is for trial purpose and hence couldn't be whitelisted.

chaitanya-smartrotamac, 2021-12-15:
Thank you @avsheth 
Will try to Whitelist my productID and do once again process.

zhaozhaozhaozheng, 2022-12-09:
> Good evening, I am using ESP32 LYRAT 4.3 board ESP IDF = 4.2.2 ESPVASDK After all the efforts my made I am able to flash amazon SDK. But now when I ask " Alexa can you play some music" it says "Music and Radio playback are not supported on this device" why? Am i missing something??

> Thank you @avsheth Will try to Whitelist my productID and do once again process.

Hello@chaitanya-smartrotamac , did you solve the problem?   What is the solution?   I have the same problem now.   Looking forward to your reply.   Thank you very much.

chaitanya-smartrotamac, 2022-12-09:
> Hello@chaitanya-smartrotamac , did you solve the problem? What is the solution? I have the same problem now. Looking forward to your reply. Thank you very much.

HI @zhaozhaozhaozheng 
Actually, I got to know that the AMAZON services like music must be linked to an account which has a membership with AMAZON which is the first step to check, 

Next is "Whitelist my productID"
refer to this issue chain
https://github.com/espressif/esp-va-sdk/issues/61

But as far as I know this is kind of Proof of concept solution but not works great for production so we dropped this project.

zhaozhaozhaozheng, 2022-12-12:
> 
Thank you very much for your reply. I will try it.