espressif/esp-va-sdk #164: spi_flash: Detected size(8192k) smaller than the size in the binary image header(16384k). Probe failed

chaitanya-smartrotamac, 2021-12-15:
amazon_alexa with Set ALEXA_BT=1 enabled.
But with LyraT board i am getting this error.

E (1955) spi_flash: Detected size(8192k) smaller than the size in the binary image header(16384k). Probe failed.
assertion "flash_ret == ESP_OK" failed: file "C:/Users/LENOVO/Desktop/esp-idf/components/esp32/cpu_start.c", line 472, function: start_cpu0_default

abort() was called at PC 0x40201500 on core 0
0x40201500: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

I think we cant enable bluetooth with alexa in LyraT board is it true?? due to memory..

Can I use memory card someway for this?

avsheth, 2021-12-15:
Hi @chaitanya-smartrotamac 
If you want to have OTA support as well, you might need a board with 16MB flash.
If not, you can remove ota_1 partition from partition.csv in examples directory and set flash size to 8MB via menuconfig->Serial flash config option.

Marcoz08, 2021-12-16:
> Hi @chaitanya-smartrotamac If you want to have OTA support as well, you might need a board with 16MB flash. If not, you can remove ota_1 partition from partition.csv in examples directory and set flash size to 8MB via menuconfig->Serial flash config option.

I have the same error. I did what you said, and removed ota_1 but now it doesn't compile the code.

avsheth, 2021-12-17:
can you provide your updated partitions.csv and the error log?

chaitanya-smartrotamac, 2021-12-17:
Thanks @avsheth 
But with this, I lost the Alexa wake word feature.
Board is not responding for wakeword. But responding to button.
Bluetooth working now.

Log :
entry 0x40080710
I (29) boot: ESP-IDF v4.2.2-dirty 2nd stage bootloader
I (29) boot: compile time 10:06:55
I (29) boot: chip revision: 3
I (32) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32: SPI Speed      : 80MHz
I (42) boot.esp32: SPI Mode       : QIO
I (47) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (90) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (98) boot:  4 ota_0            OTA app          00 10 00020000 00480000
I (105) boot: End of partition table
I (109) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1cde7c (1891964) map
I (676) esp_image: segment 1: paddr=0x001edea4 vaddr=0x3ffbdb60 size=0x02174 (  8564) load
I (679) esp_image: segment 2: paddr=0x001f0020 vaddr=0x400d0020 size=0x141b88 (1317768) map
0x400d0020: _stext at ??:?

I (1070) esp_image: segment 3: paddr=0x00331bb0 vaddr=0x3ffbfcd4 size=0x036f0 ( 14064) load
I (1075) esp_image: segment 4: paddr=0x003352a8 vaddr=0x40080000 size=0x18a50 (100944) load
0x40080000: _WindowOverflow4 at C:/Users/LENOVO/Desktop/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (1126) boot: Loaded app from partition at offset 0x20000
I (1127) boot: Disabling RNG early entropy source...
I (1127) psram: This chip is ESP32-D0WD
I (1132) spiram: Found 64MBit SPI RAM device
I (1137) spiram: SPI RAM mode: flash 80m sram 80m
I (1142) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1149) cpu_start: Pro cpu up.
I (1153) cpu_start: Application information:
I (1158) cpu_start: Project name:     avs
I (1163) cpu_start: App version:      1.0
I (1167) cpu_start: Compile time:     Dec 17 2021 10:06:19
I (1174) cpu_start: ELF file SHA256:  e9ffccbdbd4e6b33...
I (1180) cpu_start: ESP-IDF:          v4.2.2-dirty
I (1185) cpu_start: Starting app cpu, entry point is 0x40081dc8
0x40081dc8: call_start_cpu1 at C:/Users/LENOVO/Desktop/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (1683) spiram: SPI SRAM memory test OK
I (1684) heap_init: Initializing. RAM available for dynamic allocation:
I (1684) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1689) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1696) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1702) heap_init: At 3FFC7878 len 00018788 (97 KiB): DRAM
I (1708) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1715) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1721) heap_init: At 40098A50 len 000075B0 (29 KiB): IRAM
I (1727) cpu_start: Pro cpu start user code
I (1732) spiram: Adding pool of 4077K of external SPI memory to heap allocator
I (1753) spi_flash: detected chip: generic
I (1753) spi_flash: flash io: qio
I (1753) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1762) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1772) [app_main]: ==== Voice Assistant SDK version: v1.2.2 ====
I (1782) [time_utils]: Initializing SNTP.
I (1782) [time_utils]: Waiting for time to be synchronized. This may take time.
I (1822) wifi:wifi driver task: 3ffd5e34, prio:23, stack:6656, core=0
I (1822) system_api: Base MAC address is not set
[... 140 lines cut ...]
[22 seconds]: [http_stream]: [stream_delete: 3] Internal: 19324, External: 980496, min ever internal: 17880, largest free block: 17844
[http_stream]: [stream_new]: Internal: 19484, External: 980704
[http_transport]: New stream event: {"context":[{"header":{"namespace":"Speaker","name":"VolumeState"},"payload":{"volume":59,"muted":false}},{"header":{"namespace":"Bluetooth","name":"BluetoothState"},"payload":{"alexaDevice":{"friendlyName":"ESP32-0E0"},"pairedDevices": [{"uniqueDeviceId":"d246b1df-2144-42b1-a408-192eebe497b7","friendlyName":"ChaituK20Pro","supportedProfiles":[{"name":"A2DP-SOURCE","version":""}]}]}},{"header":{"namespace":"Speaker","name":"VolumeState"},"payload":{"volume":59,"muted":false}},{"header":{"namespace":"SpeechSynthesizer","name":"SpeechState"},"payload":{"token":"","offsetInMilliseconds":0,"playerActivity":"FINISHED"}},{"header":{"namespace":"AudioPlayer","name":"PlaybackState"},"payload":{"token":"","offsetInMilliseconds":0,"playerActivity":"IDLE"}},{"header":{"namespace":"AudioActivityTracker","name":"ActivityState"},"payload":{}},{"header":{"namespace":"Alerts","name":"AlertsState"},"payload":{"allAlerts":[],"activeAlerts":[]}},{"header":{"namespace":"Notifications","name":"IndicatorState"},"payload":{"isEnabled":false,"isVisualIndicatorPersisted":false}}],"event":{"header":{"namespace":"Alexa.ApiGateway","name":"VerifyGateway","messageId":"e21a08ae-f65a-d521-e6ce-4c111c31c485"},"payload":{}}}
[http_stream]: [stream_post: 5]: /v20160207/events
[http_stream]: [sid: 5] Response code: 204
[alexa_discovery]: Capabilities unchanged
I (25052) [esp_dsp]: Created I2S audio stream
I (25052) audio_stream: Starting i2s_reader stream
I (25062) audio_stream: Starting audio stream i2s_reader
[... cut ...]
I (25362) [va_nvs_utils]: No value set for: alert_0_data
I (25362) [va_nvs_utils]: No value set for: alert_1_data
I (25362) [va_nvs_utils]: No value set for: alert_2_data
I (25362) [va_nvs_utils]: No value set for: alert_3_data
I (25372) [alerts_nvs]: Alerts loaded from NVS
############## Alexa is ready ##############
I (26152) [http_transport]: [sid: 5] stream close
[24 seconds]: [http_stream]: [stream_delete: 5] Internal: 7392, External: 867036, min ever internal: 92, largest free block: 7016
I (61882) [network_diagnostics]: Network rssi: -41

chaitanya-smartrotamac, 2021-12-17:
How to use my own certificates for this example amazon_alexa ?

Marcoz08, 2021-12-18:
> can you provide your updated partitions.csv and the error log?

Thanks, I resolved the partitions.csv, code build and flash successfully but i can't connect to app.

Log:

`entry 0x40080710
I (29) boot: ESP-IDF v4.2.2-dirty 2nd stage bootloader
I (29) boot: compile time 17:53:31
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 8MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (97) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (105) boot:  4 ota_0            OTA app          00 10 00020000 00480000
I (112) boot: End of partition table
I (117) boot: No factory image, trying OTA 0
I (121) boot_comm: chip revision: 3, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1cfb04 (1899268) map
I (766) esp_image: segment 1: paddr=0x001efb2c vaddr=0x3ffbdb60 size=0x004ec (  1260) load
I (767) esp_image: segment 2: paddr=0x001f0020 vaddr=0x400d0020 size=0x150f18 (1380120) map
0x400d0020: _stext at ??:?

I (1229) esp_image: segment 3: paddr=0x00340f40 vaddr=0x3ffbe04c size=0x0590c ( 22796) load
I (1238) esp_image: segment 4: paddr=0x00346854 vaddr=0x40080000 size=0x1dde0 (122336) load
0x40080000: _WindowOverflow4 at C:/Users/MARCOZ/Desktop/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (1304) boot: Loaded app from partition at offset 0x20000
I (1304) boot: Set actual ota_seq=1 in otadata[0]
I (1304) boot: Disabling RNG early entropy source...
I (1310) psram: This chip is ESP32-D0WD
I (1314) spiram: Found 64MBit SPI RAM device
I (1319) spiram: SPI RAM mode: flash 80m sram 80m
I (1324) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1332) cpu_start: Pro cpu up.
I (1336) cpu_start: Application information:
I (1340) cpu_start: Project name:     avs
I (1345) cpu_start: App version:      1.0
I (1350) cpu_start: Compile time:     Dec 17 2021 17:53:00
I (1356) cpu_start: ELF file SHA256:  da111424e87f7aac...
I (1362) cpu_start: ESP-IDF:          v4.2.2-dirty
I (1368) cpu_start: Starting app cpu, entry point is 0x40082010
0x40082010: call_start_cpu1 at C:/Users/MARCOZ/Desktop/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (1876) spiram: SPI SRAM memory test OK
I (1878) heap_init: Initializing. RAM available for dynamic allocation:
I (1878) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1883) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1889) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1896) heap_init: At 3FFC7E40 len 000181C0 (96 KiB): DRAM
I (1902) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1908) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1915) heap_init: At 4009DDE0 len 00002220 (8 KiB): IRAM
I (1921) cpu_start: Pro cpu start user code
I (1926) spiram: Adding pool of 4077K of external SPI memory to heap allocator
I (1946) spi_flash: detected chip: generic
I (1947) spi_flash: flash io: qio
I (1947) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1955) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1965) [app_main]: ==== Voice Assistant SDK version: v1.2.2 ====
[... 88 lines cut ...]
I (32595) [va_nvs_utils]: No value set for: friendlyname
I (32595) SSDP: Network Interface List (1):
I (32595) SSDP: 1. wifi  : 0.0.0.0
I (32595) [time_utils]: SNTP already initialized.
I (32595) LSSDP: create SSDP socket 57
I (32615) [tls_certification]: Done setting global CA store
I (32625) [va_nvs_utils]: No value set for: clientId
I (32625) [va_nvs_utils]: No value set for: refreshToken
I (32625) [alexa]: Checking for companion app based authentication...

[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[app_wifi]: Disconnect event: 5, reason: 201
E (36105) LSSDP: sendto wifi (0.0.0.0) failed, errno = Host is unreachable (118)
E (36105) LSSDP: sendto wifi (0.0.0.0) failed, errno = Host is unreachable (118)
E (36105) LSSDP: sendto wifi (0.0.0.0) failed, errno = Host is unreachable (118)
E (36115) LSSDP: sendto wifi (0.0.0.0) failed, errno = Host is unreachable (118)
E (36125) LSSDP: sendto wifi (0.0.0.0) failed, errno = Host is unreachable (118)
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[app_wifi]: Disconnect event: 5, reason: 201
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[app_wifi]: Disconnect event: 5, reason: 201`