espressif/esp-va-sdk #165: Device Not connectable after building and flashing AIA example to Vaquita board

effigy, 2022-01-13:
Followed getting started steps indicated in https://github.com/espressif/esp-va-sdk:

using the windows ESP-IDF install tool I cloned the release/v4.2 branch of the repository to C:\Dev\esp-idf and insalled the tools to C:\Dev\.espressif/esp-va-sdk

Installation fails with Error Code 1, offers to display a log which fails upon selection and just opens the System32 folder.  Error in dialog indicated ESP32c3 was an invalid target.

Repeat installation with esp32-c3 deselected as a target, finished succesfully

launched ESP-IDF command prompt environment and navigated to c:\Dev\esp-idf

ran "git submodule init; git submodule update --init --recursive;"

ran install.bat, finished with only a warning about PIP version

ran export.bat as indicated by prompt, finished succesfully

navigated up one level to c:\Dev\

ran "git clone https://github.com/espressif/esp-va-sdk.git", finished witout error

ran cd esp-va-sdk/examples/amazon_aia/ to navigate 

ran "set ESPPORT=COM11" to set port and ran "%IDF_PATH%" to confirm that IDF_PATH was properly set by installer, it is set to "C:\Dev\esp-idf"

skipped running export.xx script as it was run previously

ran set AUDIO_BOARD_PATH=c:\Dev\esp-va-sdk/components/audio_hal/audio_board/audio_board_vaquita_dspg

ran idf.py menuconfig, succesfully entered config menu

made change indicated for wrover-e modules in guide:  -> Component config -> ESP32-specific -> Minimum Supported ESP32 Revision -> change Rev_0 to Rev_3

other two changes mentioned in guide refer you to "(Refer to the $AUDIO_BOARD_PATH/audio_board.cmake file" to determine if they are neccasry but this cmake file does not seem to contain any relevant information, perhaps because my board doesnt need those changes? hard to tell... I did not make them

quit config menu and saved changes

ran idf.py flash monitor

received Error: E (2011) [app_main]: NVS open failed with error 4354

I (1961) wifi_prov_scheme_ble: BT memory released
E (2011) [app_main]: NVS open failed with error 4354
E (2011) [app_main]: NVS open failed with error 4354
E (2011) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS aws_endpoint: XXX
E (2021) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS aws_account_id: XXX
E (2031) [app_main]: NVS open failed with error 4354
[app_main]: Using the non-NVS client_id: 240ac4e68058
E (2041) [app_main]: AIA device certificates not found. Please flash the certificates at the correct location and reboot the device to proceed.

The additional setup indicates to look to the readme in the example folder. Readme indicates that I must flash certs to the board and can use temporary certs from Espressif to evaluate.

Navigate to the /esp-va-sdk github page as indicated in the readme to locate the [Generating Certificates] section indicated in readme

Section does not exist?

Assuming Section 4, device provisioning is intended to replace manual cert deployment for AIA example I continue to step 4 aand immediately notice that there is no orange led pattern as guide says there should be.

Open esp alexa app on android phone, allow bluetooth, and search to add new device

No devices found.

chiragatal, 2022-01-13:
The step for flashing the AIA device certificates is required. This is the link to generate the certificates: https://espressif.github.io/esp-va-sdk/#/

effigy, 2022-01-13:
that sure worked, perhaps the documentation could be upgraded to provide a working procedure and correct links.  Ambiguity abounds in the instructions and make it very difficult to figure out WHAT the problem is.  Not sure how I was supposed to work around the broken link....

effigy, 2022-01-13:
Now the ESP Alexa app is asking me for a proof of posession PIN number, a step which is never mentioned in ANY of the documentation.  how could this be?

effigy, 2022-01-13:
ESP Alexa app is indicated to automatically login to amazon if shopping app is installed, this functionality is not working.

closed 2022-01-13 by effigy

PaulAnurag, 2022-07-20:
@effigy I am also trying this example in windows, i could build the project and flash it but then after flashing I got some errors, I could solve some of them but I am getting an issue now which I am not being able to solve. Can you please help me out.

The snippet of the Issue

[image]
e