espressif/esp-va-sdk #166: error to compile amazon_aia for DSPG board

stone-tong, 2022-03-10:
```
Executing action: all (aliases: build)
Running ninja in directory /home/stone/workspace/esp-va-sdk/examples/amazon_aia/build
Executing "ninja all"...
[1/6] Performing build step for 'bootloader'
[1/1] cd /home/stone/workspace/esp-va-sdk/examples/amazon_aia/build/bootloader/esp-idf/esptool_py && /home/stone/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/stone/workspace/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/stone/workspace/esp-va-sdk/examples/amazon_aia/build/bootloader/bootloader.bin
Bootloader binary size 0x6990 bytes. 0x670 bytes (6%) free.
[2/4] Linking CXX executable aia.elf
FAILED: aia.elf 
: && /home
......................................................

home/stone/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/stone/workspace/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a(lssdp.c.obj):(.literal.trim_spaces+0x0): undefined reference to `__locale_ctype_ptr'
/home/stone/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/stone/workspace/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a(lssdp.c.obj):(.text.trim_spaces+0x20): undefined reference to `__locale_ctype_ptr'
/home/stone/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/stone/workspace/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a(lssdp.c.obj):(.text.trim_spaces+0x38): undefined reference to `__locale_ctype_ptr'
/home/stone/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/stone/workspace/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a(lssdp.c.obj):(.text.trim_spaces+0x51): undefined reference to `__locale_ctype_ptr'
/home/stone/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/stone/workspace/esp-va-sdk/components/esp-ssdp/lib/libesp-ssdp.a(lssdp.c.obj):(.text.trim_spaces+0x6a): undefined reference to `__locale_ctype_ptr'
/home/stone/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/stone/workspace/esp-va-sdk/components/voice_assistant/lib/libaia.a(dialog.c.obj):(.literal.dialog_sm_thread+0x30): undefined reference to `xTaskNotify'
/home/stone/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/stone/workspace/esp-va-sdk/components/voice_assistant/lib/libaia.a(dialog.c.obj):(.text.dialog_sm_thread+0xae): undefined reference to `xTaskNotify'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

chiragatal, 2022-03-11:
Can you use IDF version release/v4.2 and try again?

stone-tong, 2022-03-11:
> Can you use IDF version release/v4.2 and try again?

change esp-idf version to 4.2 and after compile, getting the following error

`esp-va-sdk/examples/common/app_wifi.c:184:5: error: implicit declaration of function 'adc_power_acquire'; did you mean 'adc_power_off'? [-Werror=implicit-function-declaration]
     adc_power_acquire();
     ^~~~~~~~~~~~~~~~~
     adc_power_off
`
so this function is not defined.

arunrkumaran, 2022-06-09:
undefined reference to `xTaskNotify'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

prabhu012, 2023-06-13:
*  Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B e:\esp_idf\esp-idf\components\esp-va-sdk\examples\google_voice_assistant\build -S e:\esp_idf\esp-idf\components\esp-va-sdk\examples\google_voice_assistant 

-- AUDIO_BOARD_PATH not defined
-- Configuring done
CMake Error:
  The detected version of Ninja () is less than the version of Ninja required
  by CMake (1.3).

CMake Warning:
  Manually-specified variables were not used by the project:

    ESP_PLATFORM
    PYTHON_DEPS_CHECKED

CMake Generate step failed.  Build files cannot be regenerated correctly.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B e:\esp_idf\esp-idf\components\esp-va-sdk\examples\google_voice_assistant\build -S e:\esp_idf\esp-idf\components\esp-va-sdk\examples\google_voice_assistant" terminated with exit code: 1. 
Tell me how to solve