espressif/esp-va-sdk #167: ESP32-LyraTD-DSPG board execution error

stone-tong, 2022-03-12:
**The followings are the error, pls help me. Thanks.**
**After sucessing pair the Wi-Fi, the log "dspg_init() ...." repeat print until system failure.**

`I (79161) [dspg_init]:   dspg_init() 
I (80321) [dspg_init]: Enabling D4P FW debug logs
I (80671) [dspg_init]: Enabling D2 UART DEBUG 
I (81051) [dspg_utility]: 5a 02 29 08 00 00 00 00 00 00 
I (81161) [dspg_init]:   dspg_init() 
I (82321) [dspg_init]: Enabling D4P FW debug logs
I (82671) [dspg_init]: Enabling D2 UART DEBUG 
I (83051) [dspg_utility]: 5a 02 29 08 00 00 00 00 00 00 
I (83161) [dspg_init]:   dspg_init() 
I (84331) [dspg_init]: Enabling D4P FW debug logs
I (84681) [dspg_init]: Enabling D2 UART DEBUG 
I (85061) [dspg_utility]: 5a 02 29 08 00 00 00 00 00 00 
I (85171) [dspg_init]:   dspg_init() 
I (86341) [dspg_init]: Enabling D4P FW debug logs
I (86681) [dspg_init]: Enabling D2 UART DEBUG 
I (87061) [dspg_utility]: 5a 02 29 08 00 00 00 00 00 00 
I (87181) [dspg_init]:   dspg_init() 
I (88341) [dspg_init]: Enabling D4P FW debug logs
I (88691) [dspg_init]: Enabling D2 UART DEBUG 
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (main) 
Core  0 register dump:
PC      : 0x40097862  PS      : 0x00060136  A0      : 0x8009a852  A1      : 0x3ffba8e0  
A2      : 0x3ffbd30c  A3      : 0x0000cdcd  A4      : 0x00060123  A5      : 0x00060120  
A6      : 0x0000abab  A7      : 0xb33fffff  A8      : 0xb33fffff  A9      : 0xb33fffff  
A10     : 0x00000003  A11     : 0x00060123  A12     : 0x00060120  A13     : 0x00000044  
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40094731  LEND    : 0x40094741  LCOUNT  : 0xfffffff8  

Backtrace:0x4009785f:0x3ffba8e0 0x4009a84f:0x3ffba910 0x4009a8e5:0x3ffba950 0x4008451a:0x3ffba970 0x400845f5:0x3ffba9b0 0x401555de:0x3ffba9d0 0x400d3b2d:0x3ffba9f0 0x400936aa:0x3ffbaa10 0x40091d65:0x3ffbaa30 0x40091df4:0x3ffbaa50 0x4009249f:0x3ffbaa80 0x401e6501:0x3ffbaab0 0x401ec172:0x3ffbaae0 0x401ec303:0x3ffbadf0 0x401fbb1f:0x3ffbae20 0x4009b51b:0x3ffbae50 0x4017287f:0x3ffbaea0 0x400e627e:0x3ffbaf20 0x400e5260:0x3ffbaf50 0x400e51dc:0x3ffbaf90 0x400e62ae:0x3ffbafb0 0x400e5260:0x3ffbafe0 0x400e51dc:0x3ffbb020 0x400e62ae:0x3ffbb040 0x400e5260:0x3ffbb070 0x400e51dc:0x3ffbb0b0 0x400e62ae:0x3ffbb0d0 0x400e5260:0x3ffbb100 0x400e51dc:0x3ffbb140 0x400e62ae:0x3ffbb160 0x400e5260:0x3ffbb190 0x400e51dc:0x3ffbb1d0 0x400e62ae:0x3ffbb1f0 0x400e5260:0x3ffbb220 0x400e51dc:0x3ffbb260 0x400e62ae:0x3ffbb280 0x400e5260:0x3ffbb2b0 0x400e51dc:0x3ffbb2f0 0x400e62ae:0x3ffbb310 0x400e5260:0x3ffbb340 0x400e51dc:0x3ffbb380 0x400e62ae:0x3ffbb3a0 0x400e5260:0x3ffbb3d0 0x400e51dc:0x3ffbb410 0x400e62ae:0x3ffbb430 0x400e5260:0x3ffbb460 0x400e51dc:0x3ffbb4a0 0x400e62ae:0x3ffbb4c0 0x400e5260:0x3ffbb4f0 0x400e51dc:0x3ffbb530 0x400e62ae:0x3ffbb550 0x400e5260:0x3ffbb580 0x400e51dc:0x3ffbb5c0 0x400e62ae:0x3ffbb5e0 0x400e5260:0x3ffbb610 0x400e51dc:0x3ffbb650 0x400e62ae:0x3ffbb670 0x400e5260:0x3ffbb6a0 0x400e51dc:0x3ffbb6e0 0x400e62ae:0x3ffbb700 0x400e5260:0x3ffbb730 0x400e51dc:0x3ffbb770 0x400e62ae:0x3ffbb790 0x400e5260:0x3ffbb7c0 0x400e51dc:0x3ffbb800 0x400e62ae:0x3ffbb820 0x400e5260:0x3ffbb850 0x400e51dc:0x3ffbb890 0x400e62ae:0x3ffbb8b0 0x400e5260:0x3ffbb8e0 0x400e51dc:0x3ffbb920 0x400e62ae:0x3ffbb940 0x400e5260:0x3ffbb970 0x400e51dc:0x3ffbb9b0 0x400e62ae:0x3ffbb9d0 0x400e5260:0x3ffbba00 0x400e51dc:0x3ffbba40 0x400e62ae:0x3ffbba60 0x400e5260:0x3ffbba90 0x400e51dc:0x3ffbbad0 0x400e62ae:0x3ffbbaf0 0x400e5260:0x3ffbbb20 0x400e51dc:0x3ffbbb60 0x400e62ae:0x3ffbbb80 0x400e5260:0x3ffbbbb0 0x400e51dc:0x3ffbbbf0 0x400e62ae:0x3ffbbc10 0x400e5260:0x3ffbbc40 0x400e51dc:0x3ffbbc80 0x400e62ae:0x3ffbbca0 0x400e5260:0x3ffbbcd0 0x400e51dc:0x3ffbbd10 0x400e62ae:0x3ffbbd30 0x400e5260:0x3ffbbd60 0x400e51dc:0x3ffbbda0 0x400e62ae:0x3ffbbdc0 0x400e5260:0x3ffbbdf0 0x400e51dc:0x3ffbbe30 0x400e62ae:0x3ffbbe50 0x400e5260:0x3ffbbe80 |<-CONTINUES

ELF file SHA256: 8e9188a78cdc1b7a

Entering gdb stub now.
$T1f#eb`

MohammedMer, 2023-01-06:
Hi @stone-tong did you solve your problem ? Unfortunatly I have the same one...

avsheth, 2023-01-06:
Hi @MohammedMer what is your AUDIO_BOARD_PATH set to and which board are you using?

MohammedMer, 2023-01-06:
My AUDIO_BOARD_PATH  is "/home/esp-va-sdk/components/audio_hal/audio_board/audio_board_lyratd_dspg" and I am using the ESP32-LyraTD-DSPG board.

avsheth, 2023-01-06:
did the hardware use to work before? One of the possible reasons could be cable issue between the main board and the mic board.

0-Siddhant-0, 2025-07-28:
I'm facing this exact same dspg_init() reboot loop, which results in a Stack canary watchpoint triggered error.
I’ve checked all my connections; the LEDs and buttons on the sub-board (DSPG) are working perfectly fine.
so the failure is specific to the microphone array initialization ( I have checked that my config is correctly set as 3-mic array for ESP32-LyraTD-DSPG)
It looks like there might be an issue with the DBMD5P config or firmware files provided in the esp-va-sdk.
Would really appreciate a fix if possible or source code used to create the object file for the firmware would be massive help! (specifically for the dspg_driver)
Thanks!