espressif/esp-va-sdk #171: lyrat alexa

tcpipchip, 2022-07-08:
I am back after 3 years...

Do you know what means this error ?

I am using the LYRAT WROVER-B

==============================

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7328
load:0x40078000,len:13892
ho 0 tail 12 room 4
load:0x40080400,len:5312
entry 0x40080710
I (29) boot: ESP-IDF v4.2.3-112-ga3374b2d2f 2nd stage bootloader
I (29) boot: compile time 16:16:44
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) qio_mode: Enabling default flash chip QIO
I (46) boot.esp32: SPI Speed      : 80MHz
I (50) boot.esp32: SPI Mode       : QIO
I (55) boot.esp32: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00390000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 1, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3dd90 (253328) map
E (193) esp_image: invalid segment length 0xc37c9de5
E (193) boot: Factory app partition is not bootable
E (193) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57

tcpipchip, 2022-07-08:
i forgot to erase all flash :(

Now....getting progress :)

tcpipchip, 2022-07-09:
working
[image]
Brazil

tcpipchip, 2022-07-11:
Still working nice!

tcpipchip, 2022-07-11:
and in portguese :)

avsheth, 2022-07-13:
That's great to hear :)
Thanks

closed 2022-07-13 by avsheth

tcpipchip, 2022-07-14:
Hi, how can i reset the Lyrat to make a new provisioning ? (on Alexa running on Lyrat)
I want to reset to FACTORY state
Thanks!

tcpipchip, 2022-07-15:
Hi @avsheth , but my board is this model... :(
[image]

vikramdattu, 2022-07-15:
@tcpipchip please press `Rec` and `Mode` buttons simultaneously for 10 seconds.

tcpipchip, 2022-07-15:
hi @vikramdattu ok, i will do this afternoon and give you a feedback!

I remember that if did this got GDB message....

Maybe i didnt press "simultaneously"

tcpipchip, 2022-07-15:
i tested
[image]
not solved the problem :(