espressif/esp-va-sdk #172: Build error Windows

PaulAnurag, 2022-07-18:
Hi 
I am getting error while building in windows for example AUDIO path error and ninja version error.
Has anyone been able to achieve the same using esp idf v4.4 in windows?

vikramdattu, 2022-07-18:
Did you export/setup `AUDIO_BOARD_PATH` as mentioned in README?
For Windows, adding this to environment variables should work.

PaulAnurag, 2022-07-18:
Yes I have added this path inside Environment Variables to Set inside properties of the imported project file.

PaulAnurag, 2022-07-18:
Building in: C:\Espressif\releases\esp-idf-v4.2.3\esp-idf-v4.2.3\Workspace\esp_va_sdk\build
Configuring in: C:\Espressif\releases\esp-idf-v4.2.3\esp-idf-v4.2.3\Workspace\esp_va_sdk\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\Phynart\esp-idf\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 C:\Espressif\releases\esp-idf-v4.2.3\esp-idf-v4.2.3\Workspace\esp_va_sdk
-- AUDIO_BOARD_PATH not defined
-- Configuring done
CMake Error:
  The detected version of Ninja () is less than the version of Ninja required
  by CMake (1.3).

CMake Warning:
  Manually-specified variables were not used by the project:

    CCACHE_ENABLE
    CMAKE_EXPORT_COMPILE_COMMANDS
    CMAKE_TOOLCHAIN_FILE
    IDF_TARGET

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake --build . -- -v
Error: could not find CMAKE_PROJECT_NAME in Cache
Build complete (1 errors, 0 warnings): C:\Espressif\releases\esp-idf-v4.2.3\esp-idf-v4.2.3\Workspace\esp_va_sdk\build
Total time taken to build the project: 712 ms

This is the generated error.

PaulAnurag, 2022-07-18:
-- AUDIO_BOARD_PATH not defined
-- Configuring done
CMake Error:
  The detected version of Ninja () is less than the version of Ninja required
  by CMake (1.3).

CMake Warning:
  Manually-specified variables were not used by the project:

    CCACHE_ENABLE
    CMAKE_EXPORT_COMPILE_COMMANDS
    CMAKE_TOOLCHAIN_FILE
    IDF_TARGET

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake --build . -- -v
Error: could not find CMAKE_PROJECT_NAME in Cache
Build complete (1 errors, 0 warnings): C:\Espressif\releases\esp-idf-v4.2.3\esp-idf-v4.2.3\Workspace\esp_va_sdk\build
Total time taken to build the project: 712 ms

This the error I am getting.

avsheth, 2022-07-18:
@PaulAnurag can you check if you've followed all steps mentioned [here](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/windows-setup.html)

PaulAnurag, 2022-07-19:
Can You help me with the command for defining the AUDIO_BOARD_PATH?

PaulAnurag, 2022-07-19:
This has been solved. Thank You.

closed 2022-07-19 by PaulAnurag