espressif/esp-va-sdk #174: Response: {"code": "INVALID_AWS_ACCOUNT", "description": "The AWS IoT account provided does not support AIS"}

PaulAnurag, 2022-07-22:
Hi I am trying to build and run the amazon_aia project in MAC but I get this AWS account registration error.
I wanted to know from where is this authentication token and clientId : amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9 being generated from. Is this related to Device Product ID: esp-alexa-open?

ERROR:
E (126659) [auth-delegate]: Error! response code: 403, receive ret: 99
[auth-delegate]: Response: {"code": "INVALID_AWS_ACCOUNT", "description": "The AWS IoT account provided does not support AIS"}
E (127169) [alexa_auth_delegate]: Registration attempt failed...retrying !
E (127169) [alexa_auth_delegate]: Please check connectivity and/or credentials
E (127169) [alexa_auth_delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning
[alexa_auth_delegate]: Registration Request Buffer: {"authentication":{"token":"Atzr|IwEBIOv6y_fifEJ3ByAxLT67GH0DU9hwlL6wwvKtorMf3b7xjcOBhUBfYTiZrigxf_GjM7YeXgLxBrO3ctW9-hE3fi2Q6k9tBuXRFv3zGU1eTxeJG4UW1kRto43xz5ZBDKEwiBmVRSr8B5MfciIi9iD0kksyx2OexYG-jqpaGYyxMH5hmD8TYbx5k-bzQ-iAcNHn3Vt3bQj7_lY9KAQu99LVtmmMwd17i5uN86wiLkR8g1zoOwADUPfwY61Po7hSiZVegfHWU2ZvR8XorAkQnvS-IEkaPpQ1Zr6QPE1eBNEAWSVyV5KSP89DYQPyB4JHpFLp1DMQR0VtQ2n8M0QQEqHhjptYsYe-x6FMe5rxXej3fTGc6g", "clientId":"amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9"},"encryption":{"algorithm":"ECDH_CURVE_25519_32_BYTE", "publicKey":"8pSLbh0/12jp1azc+lCPHGh7iZne7sMLQF5Ro6FCrgI="},"iot":{"awsAccountId":"189413457560", "clientId":"0000000014A70848","endpoint":"aa5kb0c48boxm-ats.iot.ap-south-1.amazonaws.com","topicRoot":"$aws/alexa"}}
E (128499) [auth-delegate]: Error! response code: 403, receive ret: 99
[auth-delegate]: Response: {"code": "INVALID_AWS_ACCOUNT", "description": "The AWS IoT account provided does not support AIS"}
E (129009) [alexa_auth_delegate]: Registration attempt failed...retrying !
E (129009) [alexa_auth_delegate]: Please check connectivity and/or credentials
E (129009) [alexa_auth_delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning
[alexa_auth_delegate]: Registration Request Buffer: {"authentication":{"token":"Atzr|IwEBIOv6y_fifEJ3ByAxLT67GH0DU9hwlL6wwvKtorMf3b7xjcOBhUBfYTiZrigxf_GjM7YeXgLxBrO3ctW9-hE3fi2Q6k9tBuXRFv3zGU1eTxeJG4UW1kRto43xz5ZBDKEwiBmVRSr8B5MfciIi9iD0kksyx2OexYG-jqpaGYyxMH5hmD8TYbx5k-bzQ-iAcNHn3Vt3bQj7_lY9KAQu99LVtmmMwd17i5uN86wiLkR8g1zoOwADUPfwY61Po7hSiZVegfHWU2ZvR8XorAkQnvS-IEkaPpQ1Zr6QPE1eBNEAWSVyV5KSP89DYQPyB4JHpFLp1DMQR0VtQ2n8M0QQEqHhjptYsYe-x6FMe5rxXej3fTGc6g", "clientId":"amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9"},"encryption":{"algorithm":"ECDH_CURVE_25519_32_BYTE", "publicKey":"8pSLbh0/12jp1azc+lCPHGh7iZne7sMLQF5Ro6FCrgI="},"iot":{"awsAccountId":"189413457560", "clientId":"0000000014A70848","endpoint":"aa5kb0c48boxm-ats.iot.ap-south-1.amazonaws.com","topicRoot":"$aws/alexa"}}
E (130339) [auth-delegate]: Error! response code: 403, receive ret: 99

chiragatal, 2022-07-22:
Since you are using your own AWS account, make sure you have followed this: https://github.com/espressif/esp-va-sdk/blob/master/examples/amazon_aia/README.md#112-using-your-aws-account. You will also need to build your own phone apps for using with your own AWS account.

You can also use temporary certificates from Espressif's AWS account for a quick evaluation: https://github.com/espressif/esp-va-sdk/blob/master/examples/amazon_aia/README.md#111-using-espressifs-aws-account.

PaulAnurag, 2022-07-29:
@chiragatal 

I think using our own AWS account will be a long road. Let me see if I can complete this using Espressif's AWS account.