espressif/esp-va-sdk #175: Specify core2forAWS path

IcingTomato, 2022-07-26:
I am using esp-idf 4.2 and this repo's amazon-alexa example. OS is WSL2-Ubuntu20.04 with cmake 3.16.3, gcc  9.40.

I followed the README.md and `idf.py menuconfig`, an error occured:

```shell                                                                                                                                                           
 ➜  amazon_alexa git:(master) ✗ ls                                                                                                                                                                                  CMakeLists.txt  Makefile  README.md  main  partitions.csv  partitions_4mb_flash.csv  partitions_spiffs.csv  sdkconfig.bt.defaults  sdkconfig.defaults                                                              
➜  amazon_alexa git:(master) ✗ export ESPPORT=/dev/ttyUSB0                                                                                                                                                         ➜  amazon_alexa git:(master) ✗ export IDF_PATH=/home/icing/esp-idf4.2/esp-idf                                                                                                                                      ➜  amazon_alexa git:(master) ✗ . $IDF_PATH/export.sh                                                                                                                                                               Detecting the Python interpreter                                                                                                                                                                                   Checking "python" ...                                                                                                                                                                                              Checking "python3" ...                                                                                                                                                                                             Python 3.8.10                                                                                                                                                                                                      "python3" has been detected                                                                                                                                                                                        Adding ESP-IDF tools to PATH...                                                                                                                                                                                    Not using an unsupported version of tool openocd-esp32 found in PATH: 0.10.0.                                                                                                                                      Using Python interpreter in /home/icing/.espressif/python_env/idf4.2_py3.8_env/bin/python                                                                                                                          Checking if Python packages are up to date...                                                                                                                                                                      Python requirements from /home/icing/esp-idf4.2/esp-idf/requirements.txt are satisfied.                                                                                                                            Updated PATH variable:                                                                                                                                                                                               /home/icing/esp-idf4.2/esp-idf/components/esptool_py/esptool:/home/icing/esp-idf4.2/esp-idf/components/espcoredump:/home/icing/esp-idf4.2/esp-idf/components/partition_table:/home/icing/esp-idf4.2/esp-idf/components/app_update:/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/home/icing/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/home/icing/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/icing/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/icing/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/icing/.espressif/python_env/idf4.2_py3.8_env/bin:/home/icing/esp-idf4.2/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files/Python310/Scripts/:/mnt/c/Program Files/Python310/:/mnt/c/Program Files/Eclipse Adoptium/jdk-17.0.4.8-hotspot/bin:/mnt/c/Windows/system32:/mnt/c/Windows:/mnt/c/Windows/System32/Wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0/:/mnt/c/Windows/System32/OpenSSH/:/mnt/c/Program Files/Microsoft VS Code/bin:/mnt/c/Program Files/dotnet/:/mnt/c/Program Files/Git/cmd:/mnt/c/Program Files/usbipd-win/:/mnt/c/Program Files/Go/bin:/mnt/c/Users/zhife/scoop/shims:/mnt/c/Users/zhife/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/zhife/.dotnet/tools:/mnt/c/Users/zhife/go/bin:/usr/local/bin:/usr/local/go/bin                                                                                                                                                                                                         Done! You can now compile ESP-IDF projects.                                                                                                                                                                        Go to the project directory and run:                                                                                                                                                                                                                                                                                                                                                                                                    

idf.py build 

➜  amazon_alexa git:(master) ✗ idf.py menuconfig                                                                                                                                                                   Executing action: menuconfig                                                                                                                                                                                       Running cmake in directory /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build                                                                                                                           Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa"...                                                        -- Found Git: /usr/bin/git (found version "2.25.1")                                                                                                                                                                -- Component directory /home/icing/esp-idf4.2/esp-idf/components/cmock does not contain a CMakeLists.txt file. No component will be added                                                                          -- Component directory /home/icing/esp-idf4.2/esp-idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added                                                                        -- Component directory /home/icing/esp-idf4.2/esp-idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added                                                                     -- Component directory /home/icing/esp-idf4.2/esp-idf/components/openthread does not contain a CMakeLists.txt file. No component will be added                                                                     -- IDF_TARGET not set, using default target: esp32                                                                                                                                                                 -- Please specify core2forAWS path via AWS_EDUKIT_PATH                                                                                                                                                             -- The C compiler identification is GNU 8.4.0                                                                     [... cut ...]                                             -- Configuring incomplete, errors occurred!                                                                                                                                                                        See also "/home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeOutput.log".                                                                                                               cmake failed with exit code 1 
```

IcingTomato, 2022-07-26:
Here is the cmake log:

```shell
➜  amazon_alexa git:(master) ✗ cat /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeOutput.log                                                                                        The target system is: Generic -  -                                                                                                                                                                                 The host system is: Linux - 5.10.102.1-microsoft-standard-WSL2+ - x86_64                                                                                                                                           Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.                                                                                                                                Compiler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc                                                                                                 Build flags: -mlongcalls;-Wno-frame-address                                                                                                                                                                        Id flags:                                                                                                                                                                                                                                                                                                                                                                                                                             The output was:                                                                                                                                                                                                    0                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"                                                                                                                                                                                                                                                                                                                                             The C compiler identification is GNU, found in "/home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/3.16.3/CompilerIdC/a.out"                                                                                                                                                                                                                                                                                    Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.                                                                                                                          Compiler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++                                                                                                 Build flags: -mlongcalls;-Wno-frame-address                                                                                                                                                                        Id flags:                                                                                                                                                                                                                                                                                                                                                                                                                             The output was:                                                                                                                                                                                                    0                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"                                                                                                  [... cut ...]                                                                         implicit fwks: []              
                                                                                                                                                                                                                                                                                                                                                                                                        
```

IcingTomato, 2022-07-26:
```shell
➜  amazon_alexa git:(master) ✗ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/icing/esp-idf4.2/esp-idf/components/cmock does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/icing/esp-idf4.2/esp-idf/components/esp_phy does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/icing/esp-idf4.2/esp-idf/components/ieee802154 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/icing/esp-idf4.2/esp-idf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
-- Please specify core2forAWS path via AWS_EDUKIT_PATH
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /home/icing/esp-idf4.2/esp-idf/tools/cmake/build.cmake:188 (message):
  Failed to resolve component 'esp_codec'.
Call Stack (most recent call first):
  /home/icing/esp-idf4.2/esp-idf/tools/cmake/build.cmake:219 (__build_resolve_and_add_req)
  /home/icing/esp-idf4.2/esp-idf/tools/cmake/build.cmake:220 (__build_expand_requirements)
  /home/icing/esp-idf4.2/esp-idf/tools/cmake/build.cmake:220 (__build_expand_requirements)
  /home/icing/esp-idf4.2/esp-idf/tools/cmake/build.cmake:215 (__build_expand_requirements)
  /home/icing/esp-idf4.2/esp-idf/tools/cmake/build.cmake:432 (__build_expand_requirements)
  /home/icing/esp-idf4.2/esp-idf/tools/cmake/project.cmake:399 (idf_build_process)
  CMakeLists.txt:42 (project)

-- Configuring incomplete, errors occurred!
See also "/home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

IcingTomato, 2022-07-26:
```shell
➜  amazon_alexa git:(master) ✗ cat /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeOutput.log
The target system is: Generic -  - 
The host system is: Linux - 5.10.102.1-microsoft-standard-WSL2+ - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/3.16.3/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/3.16.3/CompilerIdCXX/a.out"

Checking whether the ASM compiler is GNU using "--version" matched "(GNU assembler)|(GCC)|(Free Software Foundation)":
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Determining if the C compiler works passed with the following output:
Change Dir: /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_de14a/fast && /usr/bin/make -f CMakeFiles/cmTC_de14a.dir/build.make CMakeFiles/cmTC_de14a.dir/build
make[1]: Entering directory '/home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_de14a.dir/testCCompiler.c.obj
/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address    -o CMakeFiles/cmTC_de14a.dir/testCCompiler.c.obj   -c /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_de14a
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_de14a.dir/link.txt --verbose=1
/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address     CMakeFiles/cmTC_de14a.dir/testCCompiler.c.obj  -o cmTC_de14a 
make[1]: Leaving directory '/home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_2a5c8/fast && /usr/bin/make -f CMakeFiles/cmTC_2a5c8.dir/build.make CMakeFiles/cmTC_2a5c8.dir/build
make[1]: Entering directory '/home/icing/esp-idf4.2/esp-va-sdk/examples/amazon_alexa/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_2a5c8.dir/CMakeCCompilerABI.c.obj
/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address    -v -o CMakeFiles/cmTC_2a5c8.dir/CMakeCCompilerABI.c.obj   -c /usr/share/cmake-3.16/Modules/CMakeCCompilerABI.c
Using built-in specs.
COLLECT_GCC=/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2020r3' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.4.0 (crosstool-NG esp-2020r3) 
COLLECT_GCC_OPTIONS='-mlongcalls' '-Wno-frame-address' '-v' '-o' 'CMakeFiles/cmTC_2a5c8.dir/CMakeCCompilerABI.c.obj' '-c'
 /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/cc1 -quiet -v -iprefix /home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/ /usr/share/cmake-3.16/Modules/CMakeCCompilerABI.c -quiet -dumpbase CMakeCCompilerA [... cut ...]
  collapse library dir [/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib] ==> [/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/lib]
  implicit libs: [stdc++;m;gcc;c;nosys;c;gcc]
  implicit dirs: [/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.4.0;/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/lib/gcc;/home/icing/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/lib]
  implicit fwks: []

```

IcingTomato, 2022-07-26:
Although I compiled amazon_aia, it occurred `Please specify core2forAWS path via AWS_EDUKIT_PATH` and `Failed to resolve component 'esp_codec'` when I choose  `export AUDIO_BOARD_PATH=~/esp-idf4.2/esp-va-sdk/components/audio_hal/audio_board/audio_board_m5_core2_aws`

chiragatal, 2022-07-26:
For m5_core2 board, can you try cloning this repo: https://github.com/m5stack/Core2-for-AWS-IoT-EduKit/ and pointing the AWS_EDUKIT_PATH to Core2-for-AWS-IoT-EduKit/Alexa_for_IoT-Intro/components/core2forAWS and see if this works?

IcingTomato, 2022-07-27:
Thank you very much. I can use Alexa on my M5Core2. But there is a question. I cannot play music on Core2, Alexa said the device doesn't support it.

avsheth, 2022-07-27:
Hi @IcingTomato Unfortunately, Amazon music functionality needs whitelisted product. Only the commercial products are whitelisted by Amazon. Let us know if you are going to commercialise it.

IcingTomato, 2022-07-27:
Wow, that's cool. I will talk to my boss for commercialization. Thank you. I will close this issue.

closed 2022-07-27 by IcingTomato