espressif/esp-va-sdk #178: Getting started with ESP32-LYRATD-SYNA development kit

ChandraDheeraj, 2022-11-01:
I bought ESP32-Lyratd-Syna development board to explore audio functions. I tried looking for some documentation to work with tis board but couldn't find much info. 

Whatever info I came across is suggesting to work with ESP-VA-SDK but there is no mention about this board in the library. Even in the boards supported page there is no mention about Syna board. I am very confused about the usage of this board.

Please help me getting started with Lyratd-Syna board. I just need simple functions of recording the audio an playing back the audio. I am not looking for Alexa or Google assistant kind of advanced functions. 

There are some Arduino libraries supporting all the Lyra family devices except Lyratd Syna board. Please point me in the right direction.

bboissonneault, 2023-01-24:
Did you ever get a response?  I find myself in exactly the same position, although I'm using vscode and the idf.

shushenghong, 2025-01-24:
same for me