espressif/esp-va-sdk #180: Alexa can not operate. Error log > esp-tls: 'couldn't get hostname for :api.amazon.com:' and '[alexa]: Waiting for authentication tokens.' (IDFGH-8752)

luvinland, 2022-11-23:


avsheth, 2022-11-23:
Hi @luvinland In the meantime, can you try disabling config option CONFIG_MBEDTLS_DATE_TIME and restart provisioning and see if it works alright?

luvinland, 2022-11-23:
> Hi @luvinland In the meantime, can you try disabling config option CONFIG_MBEDTLS_DATE_TIME and restart provisioning and see if it works alright?

Oh my god~
In the last few days, I've been struggling with a full feature merge based new IDF.
After about 10 hours, I can check your mention(disabling config option CONFIG_MBEDTLS_DATE_TIME). I will leave a comment.
I'm really looking forward to it.

luvinland, 2022-11-24:
Alexa normally works.
menuconfig > 'Component config' > 'mbedTLS' > **Uncheck** 'Enable mbedtls time'
Don't I need valid certificates anymore?
Anyway, Thanks.

avsheth, 2023-01-05:
Hi @luvinland
Good to know that it worked. 
We are trying to get the release out but some issues are still being debugged.
It's relatively safe to have the time check enabled. But implications of disabling it aren't too severe as well.
You can enabled that option again with new release.

closed 2023-01-05 by avsheth