espressif/esp-va-sdk #184: cmake error

prabhu012, 2023-06-13:
C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>idf.py build
Executing action: all (aliases: build)
Running cmake in directory C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant"...
-- AUDIO_BOARD_PATH not defined
-- Configuring done
CMake Error:
  The detected version of Ninja () is less than the version of Ninja required
  by CMake (1.3).

CMake Warning:
  Manually-specified variables were not used by the project:

    CCACHE_ENABLE
    ESP_PLATFORM
    PYTHON_DEPS_CHECKED

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1

C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>cmake --version
cmake version 3.23.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>ninja --version
1.10.2

C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>

sudeep-mohanty, 2023-06-28:
Hi @prabhu012 Please try the solution suggested [here](https://github.com/espressif/esp-idf/issues/11730#issuecomment-1610946303)

Whiteliondnd8, 2023-08-01:
Hi @sudeep-mohanty , I have success build and flash the code but I don't know what is next step this is the log of my terminal
can you explain further what next

[log.txt](https://github.com/espressif/esp-va-sdk/files/12229186/log.txt)

prabhu012, 2023-08-02:
Hi @sudeep-mohanty , I have successfully build and flashed the code but I don't recognize what is next step this is the log of my terminal
can you explain further what next
Thankyou 
[log.txt](https://github.com/espressif/esp-va-sdk/files/12236506/log.txt)

vikramdattu, 2023-08-02:
@Whiteliondnd8 @prabhu012 
Can you share the devkit or module details you are using? ESP-VA-SDK requires external RAM of at least 4MB. It appears, your module doesn't have the same.

Whiteliondnd8, 2023-08-02:
I am working with ESP32 Development Board - DEVKIT V1 which have 4MB flash and 320kb RAM and externally I have connected 
INMP441 microphone for audio input

DEVKIT V1   https://grobotronics.com/esp32-development-board-devkit-v1.html?sl=en

INMP441 microphone https://robu.in/product/inmp441-mems-high-precision-omnidirectional-microphone-module-i2s/

If this module is not supportable please suggest to me which development module i use

vikramdattu, 2023-08-02:
I see, the above particular board has WROOM-ESP32 module and does not have external RAM (SPI-RAM) installed.
You should use a dev kit with WROVER module instead. For evaluation purpose I would suggest you to go for a simple audio board which has SPI-RAM, speaker and microphone arrangement as well. (e.g, https://www.espressif.com/en/products/devkits/esp32-lyrat)

If you are planning for a production with `esp-va-sdk` however, you should use the board with the external DSP for reliable performance. Few boards we support are `lyrat_dspg` and `vaquita_dspg`. Do drop an email to sales@espressif.com for more information.

cc @avsheth

prabhu012, 2023-08-22:
Hi @vikramdattu @sudeep-mohanty I have bought the esp32 lyraT mini v1.2(https://www.digikey.in/en/products/detail/espressif-systems/ESP32-LYRAT-MINI/11310550)
Now I am trying to flash the code but I am getting some errors that are given in the .txt file 
Can you suggest to me what is the proper way to successfully configure the amazon_aia  example code
[amazon_aia_log.txt](https://github.com/espressif/esp-va-sdk/files/12407208/amazon_aia_log.txt)

Thankyou